gergelytakacs/AutomationShield #5: Naming of the "ball blower thingy"

gergelytakacs, 2018-03-28:
Find a good name for the ball blower thingy.

mgulan, 2018-03-28:
Something like BlowJobShield... ?

Naaah, I am kidding, I meant **FloatBallShield** / FloatShield / ...

gergelytakacs, 2018-03-28:
Hmm. FloatShield is not bad. Any other ideas from @Gabor7697  and co. ?:)

gergelytakacs, 2018-03-28:
WindShield :)

Gabor7697, 2018-03-28:
BongShield

mgulan, 2018-03-28:
yeah, right..:D

I think FloatShield (familiarly Floaty) sounds good.

gergelytakacs, 2018-03-28:
Ok, unless the group comes up with a better name, I can live with FloatShield. @Gabor7697 can you please liaison with the group and see if you can come up w/ something better? Or if this is okay for you guys?

Gabor7697, 2018-03-28:
FloatShield is okay for us.

gergelytakacs, 2018-03-28:
All right, FloatShield it is! Agreed.

:)

g

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

closed 2018-03-28 by gergelytakacs