gergelytakacs/AutomationShield #13: PID function

mgulan, 2018-04-05:
Here is the version of the PID controller (general incremental form) that you could implement, see attachment.
If you needed help/ideas with the code, let me know.

[image]

Think of what will be the inputs/outputs of the function, whether it shall internally contain anti wind-up, saturation, pre-filtering, etc.

gergelytakacs, 2018-04-05:
Thanks for your contribution and greetings from the train...:) Here is my take:
- I think it should-eventually-be both incremental and absolute formulation. For now any one of them is fine, have to start somewhere.
- Anti-wind up should be part of the PID method as it concerns the internal variables of the PID algorithm. However, it is worth cosidering using a stand-alone method for it, which could aid maintainability of the code.
- Do not forget, this is mainly an educational tool, so we don't want to make it _too_ self-closed and possibly complex. So, imho pre-filtering and saturation should be outside to increase didactic value (and improve code maintainability).

mgulan, 2018-04-05:
OK, I agree:-)
To correct myself, the form I sent is in fact the absolute one, yet can be easily used as incremental when only the input increment is used.

gergelytakacs, 2018-04-05:
Haven't seen the picture on my cell, yes - this is the one. Absolute is a bit more intuitive, incremental could be an alternate option...

Gabor7697, 2018-04-07:
https://github.com/gergelytakacs/AutomationShield/commit/d0948aa0bc14187cf8d3d358bf72a27bee78856b

gergelytakacs, 2018-04-07:
- I don't see any reason why PID needs its own class, do you @Gabor7697 ? Why can't it be `AutomationShield.PID()` ???
- I don't really understand the logic behind your new class with its own `comp` method etc. Very-very confusing, obscure and not sure about your motivation.
- Ts should be considered as a globally accessible constant, defined somewhere outisde of PID. Just initialize it right at the beginning in the header and assume it is there `long int Ts`. PID should just assume it is available, or if not, default to 1 or throw an error. There is no good reason to have it entered inside the PID method again.

Gabor7697, 2018-04-08:
pid1() is based on @mgulan s difference equation method.

gergelytakacs, 2018-04-08:
Ok, I see! In that case it needs to be properly tetsed and thought out how it will be included in the library.

kulho, 2018-04-08:
I've been trying to commit my code for PID for at least two hours, but I cannot seem to figure out what's wrong. Do I need a permission to do that or I am still doing something wrong?

gergelytakacs, 2018-04-09:
@kulho You haven't accepted the invite yet. Sent a mail about it.

gergelytakacs, 2018-04-24:
Just a quick idea: pick a way to input tuning to PID and stick with it. But then, you could do a function that translates between the two...

gergelytakacs, 2018-05-02:
There has been lots of improvement from the original PID function and the redesign helped a lot as well, see #46

closed 2018-05-02 by gergelytakacs