gergelytakacs/AutomationShield #14: OptoShield LED too bright / major hardware issue

gergelytakacs, 2018-04-06:
@Gabor7697  has found out that the LED in the OptoShield is too bright, causing resolution and calibratin issues. At 5/255 analogWrite levels the sensor ouput is 900/1023; reducing resolution. Moreover according to him:
- The 10 k resistor for the LDR divider is not optimal, he calculated 2.3 kOhm (see below).
- A different colored LED can also help to reduce output.
- Larger LED resistor could reduce power. 

All of the solutions above should be possibly tested. 

@Gabor7697's calculations on a possible change in the LDR voltage divider.
[image]

The feedback process (because of the bad ranging) is not very nice, looks something like this:
[image]

gergelytakacs, 2018-04-07:
@Gabor7697 pls. talk to @TiborKonkoly on possible HW changes when you can.

Gabor7697, 2018-04-07:
Ok.

gergelytakacs, 2018-04-20:
@Gabor7697 and @TiborKonkoly  have patched the shield successfully. @TiborKonkoly  should (after his thesis) modify the remaining boards. Do not forget to update any documentation, e.g. the wiki!

closed 2018-10-23 by gergelytakacs