gergelytakacs/AutomationShield #15: Adafruit_VL53L0X library

Gabor7697, 2018-04-08:
It's impossible to reverse engineer this : https://github.com/adafruit/Adafruit_VL53L0X/tree/master/src

gergelytakacs, 2018-04-08:
Well, not impossible but quite complex and probably does not worth the trouble.

As a far as I understand, this takes a library from STM and just wraps it for the arduino. Maybe this could actually be done, but for now just put it on hold, there are more pressing issues to solve.

gergelytakacs, 2018-04-08:
Lets leave this issue on the sidelines and re-visit (possibly) later. Assume the user has thid library installed and we go from there.

gergelytakacs, 2018-05-02:
The library is not really worth re-doing. It has been now submoduled to the this project, this way it is accessible for the user without external installation.

closed 2018-05-02 by gergelytakacs