gergelytakacs/AutomationShield #44: FloatShield: The next one

gergelytakacs, 2018-04-26:
Please include all hardware design ideas and "bugs" that can be improved upon in the next version.

gergelytakacs, 2018-04-26:
Smaller potentiometer (#19) a bit farther away from the fan for easier accesbility.

kulho, 2018-04-28:
sensor and vent connectors too close together (possible solution - smaller vent connector size fits, for future PCBs might be redesigned as well)
short-circuited resistor (due to contacts touching usb port, has to be moved on the PCB in future design, for now, duct tape will do)

gergelytakacs, 2018-05-15:
Notes from project presentation:
- Resistor over the USB connector, bad placement for shorts.
- Turbulent flow, use some sort of grid to make airflow more laminar. Grid available for download and print
- Use something else instead of the ball, the part could be custom designed for stability.
- Fan connector too close.
- Linking the Adafruit library.

gergelytakacs, 2019-06-17:
@mgulan Just FYI, Peter Chmurciak a talented student from UAMM will take up work with FloatShield. He may do a thesis on it.

mgulan, 2019-06-17:
@gergelytakacs Yep, I know him, that would be great.. unless he gets discouraged by JH.

gergelytakacs, 2019-06-17:
@mgulan They already tried really hard:)

mgulan, 2019-06-17:
@gergelytakacs :-)) and only one succeeded?

gergelytakacs, 2019-06-17:
No, I mean, the "dark side" tried to discourage him, but so far the young Padavan has won the battles. ;)