gergelytakacs/AutomationShield #55: MagnetoShield: The next one

gergelytakacs, 2018-05-10:
Ideas to be implemented in the next iteration of the MagnetoShield

KuboJakub, 2018-05-12:
now uses for I2C comunication analog pins A4 and A5. Nex time change for common SDA and SCL pins so it will be compatible with MEGA too.  Necessary to mark them on ArduinoShield boards - now without mark.

gergelytakacs, 2018-11-14:
- Only one distance column is sufficient on each side, two is an overkill.
- The bottom plastic screws are too thick. They prevent the shield from full insertion, because they hit the MCU and the barrel jack.

gergelytakacs, 2018-11-14:
The AOUT LED just takes off ~0.8 V from the DAC. This can have an impact on the performance of the entire system, as it can't push the MOSFET into saturation.

gergelytakacs, 2018-11-14:
The IRF520 was a horrible choice for MOSFET because it is not a logic level device. That's why it opens around 3 V at the Gate-Source... That leaves a very limited room for actual control. A logic level MOSFET is needed, like the IRF540 or similar. Hypothetically the MOSFETs can be swapped on the boards (even the old ones).

gergelytakacs, 2018-11-14:
There is no protection resistor for the AOUT pin of the DAC (series resistor) and there is also no resistor between ground and the gate lead of the MOSFET (to define floating states). This should be added in future designs.

gergelytakacs, 2018-11-15:
How about a diode for back-emf protection... ? Is there a diode in the MOSFET?

g

gergelytakacs, 2018-11-16:
- Increase input resolution.
- Increase output resolution. 
- How to measure the voltage on the magnet?
- How to measure current on the magnet?

KuboJakub, 2018-11-18:
> The AOUT LED just takes off ~0.8 V from the DAC. This can have an impact on the performance of the entire system, as it can't push the MOSFET into saturation. Removing the LED can help to get a G-S voltage on the current that goes to 5 V.

Yes i should i think i mentioned it in my work too. without led it gives almost 5V but there is that saturation level when is mosfet opened

KuboJakub, 2018-11-18:
I had also an idea about DAC. PWM on arduino has a oportunity to create analog voltage with higher frequences (PWM modes) so DAC wouldnt be needed. But it leeds to lower resolution of input voltage

KuboJakub, 2018-11-18:
> The voltage of the Hall sensor goes to no more than 2.5 V by definition of the sensor. (That's 0 G). So to increase resolution, either an OP-AMP solution or a change in the reference voltage would be suitable. This is not really critical but a nice addition.

Yes i used bipolar hall sensor...it create +2,5 -2,5V. If we used unipolar, resolution increase and also should totaly disappear influance of pernament magnet on hall sensor data.

gergelytakacs, 2018-11-19:
@KuboJakub 

> I had also an idea about DAC. PWM on arduino has a oportunity to create analog voltage with higher frequences (PWM modes) so DAC wouldnt be needed. But it leeds to lower resolution of input voltage

Could be and should be tested, it could be good to get rid of the DAC alltogether. However, these are pulses (at high frequency)  which are likely to create funky electromagnetic effects. I'm guessing it is not a good idea, maybe to "rectify" and stabilize it with a capacitor. At the moment I don't have the means to test it though...

>Yes i used bipolar hall sensor...it create +2,5 -2,5V. If we used unipolar, resolution increase and also should totaly disappear influance of pernament magnet on hall sensor data.

Can you pick a very similar but unipolar one? I am thinking on making small changes and ordering a new PCB...

gergelytakacs, 2018-11-19:
IRF520 output characteristics
[image]

gergelytakacs, 2018-11-19:
[image]

KuboJakub, 2018-11-19:
> @KuboJakub
>
>
> Could be and should be tested, it could be good to get rid of the DAC alltogether. However, these are pulses (at high frequency) which are likely to create funky electromagnetic [...]

i found this one. There is allways output rangne 2,5-3V. This one is preatty simmilar:
https://uk.farnell.com/texas-instruments/drv5056a3qdbzt/hall-effect-sensor-linear-sot/dp/2899436RL

datasheet to original one:
http://www.farnell.com/datasheets/1934044.pdf?_ga=2.141078448.346893941.1542627791-266032973.1523966527&_gac=1.250924338.1541516015.EAIaIQobChMI_Ir0_4HA3gIVCbDtCh1TxwqVEAAYASAAEgLUwvD_BwE

gergelytakacs, 2018-11-21:
@erik1392 

Variable string comparisons are not possible, numbers are okay though:
´´´
#define FOO 1
// Caveat: variable string expressions are not possible
void setup() {
  Serial.begin(9600);

#ifdef FOO
  Serial.println("FOO is a defined token.");
#endif

#if FOO==1
  Serial.println("FOO is 1");
#elif FOO==2
  Serial.println("FOO is 2");
#else
  Serial.println("FOO is something else.");
#endif
}

void loop() {
}
´´´

gergelytakacs, 2018-11-21:
@KuboJakub 
A marking of the polarity of the magnet would have been nice:)))

gergelytakacs, 2018-11-21:
@KuboJakub @mgulan @erik1392 
Here is an upgrade to the MagnetoShield. All the changes can be "patched" to the old version, so we have a working set of 10 units. You have until 14:00 today (21.11.2018) to say what you want regarding the design.
[image]

[image]
(Not routed yet.)

gergelytakacs, 2018-11-21:
@KuboJakub @mgulan @erik1392 
Further update. Please review ASAP.
[image]

[image]

erik1392, 2018-11-21:
@gergelytakacs 
I suggest switching positions of the pot and the logo because it will be to close to that mounting hole for the coil. Maybe also make the logo a bit smaller.

mgulan, 2018-11-21:
Agree with Erik here. Placing the pot in the other corner, near the converter, may be handier, unless there some special reason (routing) to leave it as is. If you checked it does not matter then do not waste time on it. Otherwise, all the modifications to R1 make sense (given your findings), quite an upgrade! Awesome job!

gergelytakacs, 2018-11-22:
Hello! Thanks for the input. The pot on the other side may touch the USB
connector, which is a GND. I'll try to think about how to do that.

gergelytakacs, 2018-11-22:
I couldn't sleep and spotted another (serious) mistake in my mind. Can't
use A4, A5, since that is I2C on the Uno!

gergelytakacs, 2018-11-22:
Changes from last time
- MOSFET gate resistor changed to smaller 220R (from 1k)

gergelytakacs, 2018-11-22:
Also @erik1392 the holes are 40 mm apart and 
[image]

erik1392, 2018-11-22:
Looks OK, nice work. The pot will probably collide with usb, you can try rotating it by 90 deg takt only gnd pin can touch usb

gergelytakacs, 2018-11-22:
@erik1392 If I rotate it 90 deg counter-clockwise, then the wiper will touch ground. If I rotate it 90 deg clockwise then both the GND and the 3V3 pins are likely to touch USB. This is  a situation where GND may touch it but the others not. Unsure though...

gergelytakacs, 2018-11-23:
Here is what went to production:
[image]
[image]

KuboJakub, 2018-11-23:
Did you already sent it? It looks good but today i realised there was a problem by design that some parts of arduino are magnetic. USB, crystal and button. That was reason why i pleaceg electromagnet in the middle of the board.

KuboJakub, 2018-11-23:
My mistake. The place should be okey.

gergelytakacs, 2018-11-26:
@KuboJakub The electromagnet and the Hall sensor are at the same place.

gergelytakacs, 2018-11-26:
@KuboJakub @erik1392 @mgulan
A quick update:
- PCB should be with me Thursday (28.11)
- I already have some new parts from TME
- I will try to assemble the R2 prototype on the next Tuesday (4.12), for this I will need the 3D printed superstructure from @erik1392 and I may need some assistance from @KuboJakub. Do we have any of those insulation rings left? And also what sort of glue have you used?

KuboJakub, 2018-11-26:
I have just 3 more insulations of the right size and the glue was:
https://www.google.sk/url?sa=i&source=images&cd=&ved=2ahUKEwjF4_bQqfLeAhUCEVAKHZzTAY4QjRx6BAgBEAU&url=%2Furl%3Fsa%3Di%26source%3Dimages%26cd%3D%26ved%3D2ahUKEwjF4_bQqfLeAhUCEVAKHZzTAY4QjRx6BAgBEAU%26url%3Dhttps%253A%252F%252Fceys.cz%252Fsk%252Fprodukty%252Flepidla%252Fspecialni-lepidla%252Fspecial-plastik-na-obtizne-plasty%252F%26psig%3DAOvVaw3-fomba6Fqbf4dpjBChRoZ%26ust%3D1543330863650558&psig=AOvVaw3-fomba6Fqbf4dpjBChRoZ&ust=1543330863650558

gergelytakacs, 2018-11-28:
[image]
[image]

mgulan, 2018-11-28:
omg they are fast. looks amazing too!

gergelytakacs, 2019-01-14:
The MagnetoShield R2 is done.

closed 2019-01-14 by gergelytakacs