gergelytakacs/AutomationShield #61: VibroShield: Clamp

gergelytakacs, 2018-10-23:
Design the clamp. The clamp holds the beam.

- It must be cheap, simple. 
- Prefer off-the shelf clamps and mechanisms.

KuboJakub, 2018-11-18:
So idea with 2 metal pillars and screws with washers binding vibring stripe up is not good?
Between each washer and stripe would be isolation gum circle to eliminate crossing vibration from stripe on arduino
https://www.tme.eu/sk/details/tff-m8x30_dr1213/distancne-prvky-kovove/dremec/1213x30/

gergelytakacs, 2018-11-19:
Actually, we I was thinking of 3D printing the superstructure. One feature could be possibly added - a screw-type height adjustment of the magnet.

gergelytakacs, 2018-11-19:
It's quite hard for me to imagine what you mean. Maybe attach a drawing?

KuboJakub, 2018-11-23:
[image]

gergelytakacs, 2018-11-23:
It's a beginning, but keep thiking.
How about sideways placement?

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

closed 2022-05-19 by gergelytakacs