gergelytakacs/AutomationShield #66: VibroShield: High voltage

gergelytakacs, 2018-10-29:
How to get high voltage on the board? Look for ways to step-up the 12 V source. DC-DC converters? Would that be possible?

gergelytakacs, 2018-11-10:
Check this out: https://www.gearbest.com/other-tools/pp_009726936177.html?wid=1433363
Also other DC-DC boost converters.

gergelytakacs, 2019-01-09:
At the moment I have 2 breakouts in my hand:
- One uses the XLSEMI XL6009E1 component, this is rated for 32 V max. Not quite high enough but should do at least something. Would ideally want more. [XL6009.pdf](https://github.com/gergelytakacs/AutomationShield/files/2741490/XL6009.pdf) This board seem to rhoughly follow the typical application chart in the datasheet.

- The other contains a LM2577 and a LM2596 IC. The first, according to its datasheet has a maximum rating of 65 V. The second IC is a step-down regulator and is rated for 37 V. The second breakout is more complicated and has no marking. What is the role of the 2 IC?

Ideally we'd need similar components but with a higher output rating in a similar configuration. For example the TI TPS40210DGQR is rated up to 260 V... Or check [this out.](https://www.analog.com/en/parametricsearch/11389#/p5362=Boost|Buck-Boost&p5573=min|10&p5574=12|max&p5357=100|max&p5358=8.3|60&p5349=0.5|5&qsfv=vinmin|10_vinmax|12_vout|100_iout|500m) For now, @KuboJakub  should perform some inital get-to know testing with the available boards.

gergelytakacs, 2019-01-09:
@KuboJakub Also check out [this](https://en.wikipedia.org/wiki/Boost_converter) as a beginning.

KuboJakub, 2019-08-29:
My first design of hopefully working Step-UP. Just create a high voltage with very small load current. By potentiometer R1 is possible to manipulate with the Vout.
Attached Simulink scheme - Simscape electrical library is needed.
[Steu_UP_convertor2_FixFreqency_FixR.zip](https://github.com/gergelytakacs/AutomationShield/files/3556573/Steu_UP_convertor2_FixFreqency_FixR.zip)