gergelytakacs/AutomationShield #67: VibroShield: Sensors

gergelytakacs, 2018-10-31:
- A short (2.2") resistive flex sensor should be built in.
- An accelerometer should also be mounted to the end of the beam. Mounting with thin cables? Flexible PCB?
- What's the minimum sensing distance of TOF sensors? What's the sampling rate? Could it be also used?

[image]

KuboJakub, 2018-12-08:
There is a problem that flex sensor is measuring deflexion only in one direction. We can combine 2 of them as it is showed on a picture.
[image]

gergelytakacs, 2018-12-10:
Have you tested this? It could hypothetically work, but I'm unsure... It
should be tested!

KuboJakub, 2018-12-10:
just with potentiometers so far and it worked. I would like to with 2 flex sensors

gergelytakacs, 2018-12-11:
Allright then. Any alternative idead to the pair of flex sensors? ToF?

KuboJakub, 2018-12-20:
So i tried 2 other possibilities for measuring of deflection. Firts one was a Tof sensor vl5310x with Pololu library https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Single/Single.ino 
With this library i can sample with the highest frequency 50Hz. Acctualy it is useable but a noise is preatty big and faster sampling would be better. Also resolution is not so good and there has to by some matt stick on the beam because sensor has issues with reflection from the beam.
It is possible after rework of the library we can sample faster but i saw also in others librarys maximum sampling rate 50Hz.
[image]

KuboJakub, 2018-12-20:
Second option is to use 2 hall effect sensors. By this option there is "no sampling limit". Also resolution is  really sensitive but a magnet is needed on the beam. I used preaty strong but not so big neodymium magnet and results were much better than by Tof sensor i think. On the picture is measuring with hall sensors by approximately the same sampling fequency 50Hz.
I used 2 of them (beam is located between them) because a power of a magnetic field strongly depends on a distance from the sensor.
[image]

KuboJakub, 2018-12-20:
There is one more issue. because of the magnet at the end of the beam a moment of inertia is much higher and the beam is oscillateing longer.
[image]

gergelytakacs, 2019-02-18:
That is not really an issue. We can design the beam as we wish, we just
assume that the magnet is part of the structure then.

g.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

closed 2022-05-19 by gergelytakacs