gergelytakacs/AutomationShield #75: MagnetoShield: A first principles model

gergelytakacs, 2018-11-22:
A first principles model of the MagnetoShield shall be developed. [This](http://ctms.engin.umich.edu/CTMS/index.php?example=Introduction&section=ControlStateSpace) could be used as a starting point, but remember, that we have an additional attraction between the two permanent magnets! Also, There is a parallel capacitor added there to stabilize the voltage signal, but that could be possibly neglected.

gergelytakacs, 2018-11-27:
- The weight of the magnet is 0.76 g
- Resistance of the electromagnet is 192 Ohm

gergelytakacs, 2018-11-28:
Amateur hour: essential questions
Q: Can an input to output (u(k)->y(k)) measurement be used in closed loop identification? Does the presence of the controller affect the results? How and why?

gergelytakacs, 2018-11-28:
@mgulan 
The usual model, with simplified electrical response is:
[image]
After linearisation leading to:
[image]

gergelytakacs, 2018-11-28:
@mgulan If you wish to try grey box (or any other type)
[MagnetoShield.zip](https://github.com/gergelytakacs/AutomationShield/files/2625664/MagnetoShield.zip)
 identification:

The columns go: u(k) - input [V], w(k) - injected noise [V], y(k) - position [mm], sampled at Ts=0.004 s.

gergelytakacs, 2018-11-30:
The linearized state-space models are third-order. A third-order black box model yields fairly sensible results:
[image]
Given one uses detrended data ("delta") formulation and
[image]
that is 
```
 Options = ssestOptions;                              
 Options.Display = 'on';                              
 Options.WeightingFilter = 'prediction';              
 Options.InitialState = 'estimate';                   
 Options.N4Horizon = [15 39 39];                                                                          
 ss27 = ssest(Magnetic Levitationd, 3, Options)  
```

gergelytakacs, 2018-11-30:
Here is the usual nonlinear and linearized model for magnetic levitation:
[image]

gergelytakacs, 2018-11-30:
Amateur hour: essential questions
Q: What is the capacitance of the solenoid? How does the extra capacitor parallel to the solenoid affect the dynamics?

gergelytakacs, 2018-12-03:
Accounting for the permanent magnet shall yield the same linearized model:
[image]

gergelytakacs, 2018-12-03:
Also, according to (Barie and Chiasson)[https://www.tandfonline.com/doi/abs/10.1080/00207729608929322] and their references, inductance is distance dependent, so we get a slightly more complex model as follows:
[image]

gergelytakacs, 2018-12-03:
@mgulan 
As of 3rd December the 3rd order black-box model seems to be decent, however, I had no luck with linearized and nonlinear first principle models yet. Neither of the latter two converge to anything useful. If you are up for the challenge, yo may try to make them work - the current "skeletons" for the identification (are here):[https://github.com/gergelytakacs/AutomationShield/tree/master/matlab/examples/MagnetoShield]. I'd probably try the lienarized grey-box at first, the nonlinear is even worse.

gergelytakacs, 2018-12-05:
@mgulan  A sensible black box TF model (1st order!) added to the collection. Maybe it would be fun to use the data (or even the hardware) in the Identification classes:
[image]
```
model =
 
  From input "Solenoid Voltage" to output "Position":
   0.02382
  ---------
  s + 46.93
 
Continuous-time identified transfer function.

Parameterization:
   Number of poles: 1   Number of zeros: 0
   Number of free coefficients: 2
   Use "tfdata", "getpvec", "getcov" for parameters and their uncertainties.

Status:                                                         
Estimated using TFEST on time domain data "Magnetic Levitation".
Fit to estimation data: 74.69% (data prefiltered)               
FPE: 7.842e-09, MSE: 7.665e-09  
```

gergelytakacs, 2018-12-05:
@mgulan @erik1392 
Amateur hour:
**Q: If you have an unstable model, testing its time-domain response to input data will necessarily cause an unstable response?**
**A: ?**
My guess is yes,

gergelytakacs, 2018-12-05:
@mgulan 
Successful grey-box linearized TF model:
[image]
```
  From input "Solenoid Voltage" to output "Position":
    -1.215
  ----------
  s^2 - 2218
 
Continuous-time identified transfer function.

Parameterization:
   Number of poles: 2   Number of zeros: 0
   Number of free coefficients: 2
   Use "tfdata", "getpvec", "getcov" for parameters and their uncertainties.

Status:                                                
Estimated using TFEST on frequency domain data "dataf".
Fit to estimation data: 88.95%                         
FPE: 1.465e-09, MSE: 1.46e-09 
```
This conforms to the "classical" model, where the mechanics of the ball are augmented with an U=Ki*I approach

gergelytakacs, 2018-12-07:
Amateur hour:
**Q: Can MATLAB's `nlgreyest()' handle unstable models?**

gergelytakacs, 2018-12-07:
[image]
[image]
[image]

mgulan, 2018-12-07:
Hard to find out. Ver. 2018b of Scientific Toolbox provides only few useless information on p. 17-129/130.

gergelytakacs, 2019-01-18:
There is something strange with the TF sys ID...
- K is by definition positive
- The data is OK: the more voltage (current) the closer it gets to the magnet, that is, the less distance.
- This logic should be mirrored by the transfer function. From a reference position (linearization point) the ball should "fall" to negative values, not run to positive (towards the magnet). This means, that the gain, and thus b0 is negative.
- Yet, the system identification toolbox consistently tells, that the gain is positive.

Could this problem be caused by:
- the feedback
- linearization

Don't know where my logic fails.

gergelytakacs, 2019-02-18:
The first principles linearized models are quite cool actually, but there is still no directly identified (or compared) nonlinear model. I will make a new issue out of this,

closed 2019-02-18 by gergelytakacs