gergelytakacs/AutomationShield #80: MagnetoShield: The next one (R2->R3)

gergelytakacs, 2019-01-14:
For the next release of the MagnetoShield:

- The voltage sensing voltage divider is on the high side on the R2, thus can only sense supply voltage. This should be moved to the low side of the electromagnet, so it can be utilized to sense the magnet voltage.
- The ring holding the magnet cage covers the logo partially. The logo can fit below the potentiometer.

gergelytakacs, 2019-02-14:
The shield is ready, so...

gergelytakacs, 2019-11-13:
@erik1392 @KuboJakub 

We have decided to go ahead and re-design the power part of the shield, thus making R3 necessary.

Here is an idea for @KuboJakub - look at some basic tutorials to learn [LTSpice](https://www.analog.com/en/design-center/design-tools-and-calculators/ltspice-simulator.html) and then try to simulate our problem. I'd model the analog part starting from the signal leaving from the DAC (the 0-5V) signal. You should include our electromagnet, the diode on it, the capacitor on it and then attempt to test a couple of possible solutions (JFET?) to upgrade the control of the magnet. 

I have just tried LTSpice with another problem (related to opamps of @EvaVargova ) and it was really-really helpful.

erik1392, 2019-11-19:
The solution they suggested there with oamp and NPN transistor looks quite good. I was also thinking about something like that in case the Jfet won't work. 
[image]
 
But I suggest to simulate it with circuit we have ( inductor, cap, diode...) not just some arbitrary resistance.

erik1392, 2019-11-19:
I measured the inductance of the electromagnet with method 1 from [here](https://daycounter.com/Articles/How-To-Measure-Inductance.phtml?fbclid=IwAR2pqiXBKQIU5Vya9NhzuWJnSVIgqQbx_L_6DSDRIJMI5zIAo4o93lvt-uw).
I tried electromagnets on 4 different shields and the results were in the range of 240-260 uH.

gergelytakacs, 2019-11-20:
@KuboJakub Thanks for the question on StackExchange. I have no beef with the opamp-transistor configuration, actually - so that is a good possibility and the author of the original comment is kind of right about building feedback into the loop like this. This is a good idea, thank the man:)

Also, you have mentioned that you had no success with the JFETs in simulation. It would be also good to put a couple of screenshots here that illustrate us what you meant, since it is a bit hard to imagine that by words only.

@erik1392 Good job with the inductance. I have failed miserably then before, since did not know the correct method:) However, the multimeter (benchtop one) in the lab as far as I remember has direct inductance measuring options. Also there is an LCR meter in my office. Can you confront your measurements with those ones? It's not that I don't believe your results are fine but I am thinking what are we doing wrong with the instruments.

KuboJakub, 2019-11-22:
So i checked the possibility to use JFET a little bit more (sorry for the previous status) and based on the video https://www.allaboutcircuits.com/video-lectures/jfet-amplifiers/ I made a simulation which was quite satisfying (common source amplifier). The circuit is not totally the same because the JFET amplifier in the video is arranged for signal amplification with minimum current but we need the current (minimally this is my conclusion... there is possibility i understood it wrong).

Some notes how does it works:
- N-JFET needs an opposite sign of the voltage on the gate, so we are regulating "big circuit " with voltage 0 to -5V. For this conversion is responsible capacitor C2 (not sure how does it work) right behind Arduino voltage source (DAC). So if we send 5V from Arduino on the gate appears 0V, if we send 0V on the gate appears -5V.
- Now if there is a 0V on the gate = JFET is opened and we can measure on the coil maximal current that can flow through JFET (in datasheet like Zero-Gate Voltage Drain Current)
- If it is -5V on the gate = JFET is closed and there is no current flowing through electromagnet (concretely JFET used in the simulation is totally closed by voltage around -5V...in datasheet like Gate-Source Cutoff Voltage).

From the simulation i get the result: If we would use N-JFET, we are able to get 0V on the electromagnet quite easy. The small problem is, if we want to have JFET fully opened (means that we would require 12V on electromagnet). Probably because of the internal resistance of the FET and limitation of the maximum current which can flow through the JFET.
I got the best result with JFET 2N4391 and I got around 10V on the electromagnet. I was able to increase the voltage on the electromagnet when i added a small resistance between the 12V source and the coil. But the result was higher voltage on the electromagnet but with smaller current and for electromagnetic force more important thing is the current.

Here is a description of the graph:
V(ref) = voltage from arduino
V(in_shifted) = somehow capacitor C2 is shifting the voltage into range (0 to -5V) what we need for the closing of the N-JFET.
V(12-Vmag) = 12V - voltage on the electromagnet
I(L1) = current on the electromagnet

[image]

I found the JFET 2N4391 on Mouser webside...it is not so cheap, but if you also think, this could be the solution i can make an order and we can try it https://www.mouser.sk/Search/Refine?Keyword=2N4391
The idea which proposed the guy from the StackExchange i have to try. So far I didn't do that.

KuboJakub, 2019-11-23:
I tried the idea from StackExchange. As the first thing, i tried the circuit just with suitable op-amp (so opamp which is rail to rail, can be supplied by 12V and can provide output current around, or more than 50 mA).  The results were very nice (see below) but as the gentleman said for exact 12V we would need a supply bit bigger than 12V, but thanks to rail to rail we are able to set exact 0V and to be very close to 12V... it is around 11.7 V.
+ We don't need so many components and results are very good in simulation - probably the most accurate option
- (maybe is contra) We are feeding the coil from the opamp. It should be able to produce 70mA on output but i don't know exactly how much current the coil needs (I made measurements just through the current observer, which is on the shield).
Opamp I used is AD8614 https://www.analog.com/en/products/ad8614.html
[image]

The second possibility is to take some opamp which can be supplied by 12V but the output current is smaller than needed one. The output from opamp is connected to NPN transistor. I this way i was able to produce an output voltage around 11.2 V. It is a bit smaller than in the previous example, probably because of the resistance on the transistor.
+ we dont have to care about how much current the electromagnet needs
- more components so in the conclusion, this way can be a little bit more expensive
- The accuracy is not as good as previous (but the difference is really small)
Opamp LTC6240 https://www.mouser.sk/ProductDetail/Analog-Devices/LTC6240CS5TRMPBF?qs=sGAEpiMZZMvi6wO7nhr1LxXEqu5%252BusKChi5O6%252B5rnVI%3D
Transistor e.g. 2N2222 https://www.tme.eu/sk/details/2n2222-cdi/tranzistory-npn-tht/cdil/2n2222/
[image]

If i compare these 2 solutions with the previous one with the JFET, here is the problem that "the control part" is supplying the electromagnet on the + sign. The version of MagnetoShield is constructed that we regulate the voltage of electromagnet on the - sign. So this 2 solutions require new design of the circuit.

KuboJakub, 2019-11-23:
I am personally for the JFET solution because we don't need to change the whole circuit so radically and it is still providing quite satisfying results (we don't have to design whole PCB). On the other hand, our maximum voltage on the electromagnet will be around 10V (i hope).

 Here are my circuits which i used for the simulations.
[VolrageRegulation.zip](https://github.com/gergelytakacs/AutomationShield/files/3882823/VolrageRegulation.zip)

gergelytakacs, 2019-11-26:
Great job @KuboJakub , really useful stuff. First of all I am sorry for the slow response time, I was out of the office from Fri-Mon (inlcuding). I am going to respond to your latest post in single comment.

> N-JFET needs an opposite sign of the voltage on the gate, so we are regulating "big circuit " with voltage 0 to -5V. For this conversion is responsible capacitor C2 (not sure how does it work) right [...]

Yes, but don't forget this is the same thing as you've done with the MOSFET. Ground is relative, so essentially you've been feeding your MOSFET negative voltages (compared to 12 V). So there is nothing new. The capacitor is not the one responsible for this, I'm not sure about its role actually.

> I got the best result with JFET 2N4391 [...]

This looks like somehting from a 1950's movie:) 

> We are feeding the coil from the opamp. It should be able to produce 70mA on output but i don't know exactly how much current the coil needs (I made measurements just through the current observer, which is on the shield).

I think it should be under 100 mA, I have used the current sensor onboard as well. Shall we decide to only use an opamp, we should have some headroom because there might be some voltage peaks, and we don't want to fry our opamp (and board) at the first trial. This is why I'm thinking its a safer choice to go with opamp+transistor. The one you have suggested has it only to 70 mA. That's only so and so, I don't trust it. Are there any more "powerful" rail-to-rail opamps that can be powered directly by 12 V?

> more components so in the conclusion, this way can be a little bit more expensive

Yes, but still cheaper than using a weird / specific JFET or OPAMP. I think, purely from the viewpoint of price, opamp+transistor should be the cheapest option. And I think there is plenty of space to put that on the PCB! (Don't forget, we can put components on the bottom too!:)

> So this 2 solutions require new design of the circuit. I do not think that should be such an issue. 

The redesign could be done in a couple of hours...

>I am personally for the JFET solution because we don't need to change the whole circuit so radically and it is still providing quite satisfying results (we don't have to design whole PCB). 

We already have it design, we only need to change it. As I said, it is not a big deal.

* My preference is to go with the opamp-transitor combination * It also touches the point what the guy on StackExchange said. This is closed loop, not open-loop! What is your opinion @erik1392 ?

Here is a couple of facts / questions I'd like you to consider and try.
- We have a [TCL2272](https://www.ti.com/product/TLC2272?utm_source=google&utm_medium=cpc&utm_campaign=asc-null-null-GPN_EN-cpc-pf-google-wwe&utm_content=TLC2272&ds_k=TLC2272&DCM=yes&gclid=CjwKCAiAlO7uBRANEiwA_vXQ-_tsfI7jWBA80bEiFZl29MqDitjxJWj0w1iW2M9h_TYPPONK7qyrVRoC-xAQAvD_BwE&gclsrc=aw.ds) general purpose opamps available both DIP and SMD versions. I have ordered some too, so it would be a good idea to use those. They are rail-to-rail. I have attached the LTSpice simulation model for it. Can you please try your circuit with that one?
- Do you really need to feed the opamp with 12 V? What are the output voltages of the opamp? The reason why I'm asking this is, that the 5 V rail on the Arduino is regulated, while the 12 V is not. So each slight change in supply type (the adapter) could possibly reflect on the opamps amplification. Can you please check what sort of voltages the opamp outputs?
- We have TCL2272 opamps here and a bunch of 2N2222 transistors. Can you try the simulations with those? Can you try to find an "equivalent" of the 2N2222 but in a SMD version? You can also try this combination for real, since we have the parts.

gergelytakacs, 2019-11-27:
@erik1392 @KuboJakub 

I am (trying to:) order some parts and I have included these two SMD transistors, that are more or less equivalent to the 2N2222:
- [BC817,215](https://sk.farnell.com/nexperia/bc817-215/transistor-npn-45v-sot-23/dp/1081223), and
- [BCW65CLT1G](https://sk.farnell.com/on-semiconductor/bcw65clt1g/transistor-bipol-npn-32v/dp/2535613).

You could (and should) use these parts in simulation, if you find a model for them. If you do a quick lab test with the TCL2272, you can safely use the 2N2222 as a replacementl

erik1392, 2019-11-27:
@gergelytakacs I have already tried the BC817 but the ON Semi .. version i sent you.  So far, it was the same as 2N2222.

 @KuboJakub i sugest to tweek the oamp ratio pull the voltage is not clipped that much.  And it would be nice to use precise resistors for that feedback loop so amp ratio between different shields will be the same.

gergelytakacs, 2019-11-27:
Meeting 27.11.2019
- Try a physical experiment with this setup.
- Try if the TLC2272 tolerates 12 V, otherwise we must select a different OPAMP.
- If you experiment use TLC2272 + one of the transistors we are buying.
- Experiment with the resistor ratios so the output is not clipped, tell which resistors exactly we need.
- Use the NAS please:)
- Start to draw the new PCB.
- Try to work with the Simulink API

erik1392, 2019-11-28:
I augmented the simulation scheme that it is closer to our real scheme. There is also an additional almost 1V voltage drop on the 10-ohm resistor for current measurement.
[image]

gergelytakacs, 2019-11-28:
@erik1392 

Awesome job. Love it.

- Can you please upload the scheme and the parts to the NAS so @KuboJakub may have a look at it?
- Are these resistor values available easily?
- I think maybe it would be better to tune the gain a bit more conseratively, but these values should be close enough.
- @KuboJakub can you confirm that this works in experiment as well?
- BTW: I've ordered SMD versions of the TLC2272, thus if it works and the PCB is done (and ordered) we will have those as well.

g.

erik1392, 2019-11-28:
@gergelytakacs 

- The files are already there. I also edited the Diptrace scheme and tomorrow / during the weekend I will look at the PCB. 
- Resistor values are taken from the table I found. I haven't checked yet if they are in stock.

gergelytakacs, 2019-11-29:
@erik1392 @KuboJakub 
The TLC2272's have arrived:

[image]

KuboJakub, 2019-11-30:
I appologyze i did not answer earlier because i am abroad. I already made the experiment but without the current measurements. 
I will try to complete it till monday but so far it looks quite nice. Supply voltage from Vin in arduino was around 11.4V and i get on electromagnet something above 10.5V

KuboJakub, 2019-12-01:
So i made the measurements.
The measurement i made is based on a schematic below (instead of coil I have used a resistor)
I have considered:
- supply voltage form Vin = 11.5V
- Vmax form DAC = 4.8V
- The resistance of the electromagnet is around 200 Ohms. I had just 230 Ohm resistor, so i used that...+30ohms could simulate the resistance for current measurement and maybe some additional loses caused by capacitor, diode, resistors for voltage measurement.
[image]

I think the best ratio between resistors is R1/R2 = 6/5 kOhm. This results into the gain which amplifies the DAC voltage on maximum 10.3-10.4V (the measured one, theoretically we should get 10.56V from 4.8V). Ratio 6.2/5 (or bigger) is from calculation more suitable, but because of the voltage drops on the transistor the top of the signal is cut off.
[image]

I would maybe prefer to use 5 kOhms for R2 resistor as a fixed one and as R1 transistor i would use some accurate potentiometer. The reason is, as @gergelytakacs discovered by improving the library
for MagnetoShield, every board provides a bit different supply from Vin, which could cause cutting off the top of the signal.
I am also not sure about the resistance of electromagnets if they are the same or different. Also in the library, if we describe the curve characterizing the voltage on the electromagnet some boards can have the top cut off and  maybe the accuracy of the regulation will be worse (maybe not...still it is just top values which are not used so often).
If we use the potentiometer we can tune the gain as we need in ratio var./5 kOhm.
Otherwise, the best results were with 6 and 5 kOhm resistors... on the load, I got around 10.3 V as the maximum value.

KuboJakub, 2019-12-01:
5 kOhm resistor: https: //www.tme.eu/sk/details/ar0805-5k-0.1%25/presne-rezistory-smd-0805/viking/ar05btcw5001/
6 kOhm resistor: https://www.tme.eu/sk/details/era6aeb6041v/presne-rezistory-smd-0805/panasonic/
6.2 kOhm resistor: https://www.tme.eu/sk/details/era6aeb622v/presne-rezistory-smd-0805/panasonic/
potentiometer -This one is +-10% I could find no more accurate potentiometer on TME in SMD package (there are +-3% but are super expencive and big). Maybe we could use the one we have or this one:
https://www.tme.eu/sk/details/1806wsmd-10k/potenciometre-viacotackove-smd/sr-passives/

gergelytakacs, 2019-12-02:
@erik1392 Please also review @KuboJakub 's latest results, they seem to be good judging by the first look...

First of all, great job @KuboJakub ! Thank you. Before we proceed, I have a couple of notes / questions.

- Using a potetntiometer as one of the gain resistors is not a bad idea, but that would necessitate hand - calibrating each magneto after manufacture. I mean, that's not really a big deal, just saying. Also a re-calibration of input-output voltages can be now done automatically, as we should be able to measure the solenoid voltage!  So I could not decide on the potetiometer issue as of yet, I'd like to hear @erik1392 's opinion too.
- Where have you exactly got the figure 11.5 V for the supply? Although this seems, correct - just curious if you measured or assumed this.
- I might have forgotten, but why is the input voltage going only to 4.8 V? Is that happening in reality as well? Why?
- In your charts, you are showing the voltages going out of the opamp vs. going into the load. Can you please show the voltages going into the opamp (out of the DAC) and the load too?
- You don't need to simulate the coil (your assumptions were correct though) by a resistor, there is a free standing extra solenoid - just ask Erik. Either he has it in his office or it's in the lab.

erik1392, 2019-12-02:
- I quite like the idea of a potentiometer. I as alternative I found this [one](https://www.tme.eu/sk/details/3214w-1-103e/potenciometre-viacotackove-smd/bourns/) which is a bit smaller but more expensive. 
- the free standing solenoid is in my office, so I can give it to you. I think for the accuracy of the experiment it is necessary to use exact resistances. Important information is the voltage on the solenoid after the 10ohm resistor for current measurement.

gergelytakacs, 2019-12-02:
@erik1392 @KuboJakub 

I made measurements with a new, more precise LCR meter, according to which at
**f=100 Hz**,
**L=263.1 mH** (so the cheap Chinese meter was allright), and
**Q=0.72** -
where is the ratio of the inductance L to the resistance R of a coil at a given frequency and this defines the efficiency of inductor, e.g. 
[image]
the higher Q factor of the inductor, the closer the inductor to be ideal without losses. [source](https://www.quora.com/What-is-the-Q-factor-of-a-coil) Also, look at [WikiPedia](https://en.wikipedia.org/wiki/Q_factor)

gergelytakacs, 2019-12-02:
@erik1392 @KuboJakub 

Ok, let's go with the potentiometer. The small one by erik seems to be fine.

gergelytakacs, 2019-12-02:
@erik1392 @KuboJakub 

I would also like to change the DAC to mcp4725 - I have a couple of them bought. This shouldn't be too hard.

gergelytakacs, 2019-12-02:
@erik1392 @KuboJakub 

Xmas has come!

[image]
[image]

KuboJakub, 2019-12-03:
> Where have you exactly got the figure 11.5 V for the supply? Although this seems, correct - just curious if you measured or assumed this.

Supply voltage from arduino is around 11.4-11.5V. I have measured it.
> I might have forgotten, but why is the input voltage going only to 4.8 V? Is that happening in reality as well? Why?

Honestly I also dont know but in my opinion, it has something to do with that the supply of DAC is 5V and intern circuit consumes some voltage too so probably that is the reason for voltage drop on 4.8V. The value is also measured.

> * You don't need to simulate the coil (your assumptions were correct though) by a resistor, there is a free standing extra solenoid - just ask Erik. Either he has it in his office or it's in the lab.

So i made an experiment with real electromagnet and 10-ohm resistor. The voltage drop is luckily bit smaller than 1V. I got 9.96V directly on the electromagnet. So I think this works quite well. The resistors i used were as priviously - 5k for R2 and potentiometer as R1.

gergelytakacs, 2019-12-04:
@KuboJakub thanks for the experiments. We will discuss the next steps in person today, I hope @erik1392 can be also present.

As I said, I'd prefer to exchange the 8-bit DAC to a 12-bit more advanced one too. As far as I know @erik1392 is already on the process of re-designing the PCB - this should be our first and foremost priority.

erik1392, 2019-12-04:
@gergelytakacs @KuboJakub 
Here is almost finished PCB (some text and logos are missing). Check it, if you can find any mistakes.
[image]
[image]
[image]

erik1392, 2019-12-04:
As we spoke with @KuboJakub here is also the second version of PCB where the 3D printed part should have legs only from one side and would enable a more clean view of the magnet.

[image]
[image]

gergelytakacs, 2019-12-05:
Excellent job @erik1392 I will start reviewing it right away.

KuboJakub, 2019-12-06:
There is a mistake in gain resistors. They are switched, we want to have R3 like potentiometer and R10 fixed one. Gain is like 1+Rpot/Rfix = 1+0to10kOhm/5.04kOhm. Theoretically, it
 can be like this but i think to have the denominator fixed is much better for tuning of the gain.
[image]

KuboJakub, 2019-12-06:
May i ask what for a device is the current sensor? I can't find it under the name INA149. Maybe if you have a link for it i would appreciate it.

gergelytakacs, 2019-12-09:
@KuboJakub  Sorry for the late reply. Yes, the current sensor is INA169, that was  misspell in the paper. I haven't had the chance to go through the PCB as of yet. I am hoping I will have the time to do that today.

gergelytakacs, 2019-12-09:
@erik1392 Can you address an review Jakub's note on the potentiometer issue?

erik1392, 2019-12-09:
@gergelytakacs sure, I'll change it. 

@KuboJakub maybe you have written ti wrongly

> The resistors I used were as previously - 5k for R2 and potentiometer as R1.

[image]

gergelytakacs, 2019-12-09:
@KuboJakub The new PCB has been ordered, the fixed / variable resistors for the opamp gain have been flipped. I think (and hope) that the PCB could be here before x-mas but we'll see.

gergelytakacs, 2019-12-09:
@KuboJakub The parts have been ordered. We should have enough material to complete about 5 pcs of R3 MagnetoShields.

gergelytakacs, 2019-12-11:
Update: The components have arrived, I can bring them to the lab tomorrow.

gergelytakacs, 2019-12-11:
Update: PCB is on its way.

[image]

gergelytakacs, 2019-12-11:
@erik1392 @KuboJakub 

Santa has came early. All will be in your drawer in your lab. Note that some components are shared between Magneto and Link, you might need to check that too. (I don't want to split the contents, because it would be hard to identify some components without the original bag.)

[image]

[image]

gergelytakacs, 2020-01-13:
@KuboJakub I have ordered the 5 kOhm resistors just now. They've cost a lot more than they should've :)

KuboJakub, 2020-01-15:
I have some unpleasent news:
- As i tried new a R3shield some part started to burn...couldnt identifie whitch one so far. 
- Because of it i started to chack PCB and all components. DAC behaves like i made a soldering mistake and connected 2 pins together, but visualy cannot see the mistake. I will chacke it in lab but i am afraid of something just burned the DAC.
- Pins for Vdd and GND on PCB for OPamp looks switched :/ I know we are using differential voltage +-8V so it should not destroy OPamp but than amplification is in a different direction and probably this is the reason of the burning smell when i plugged R3 in.
[image]
[image]

erik1392, 2020-01-15:
Unfortunately, you are right... there was a stupid mistake in the scheme with overlapping lines....

[image]
[image]

For the purpose of testing, we can correct it by cutting the incorrect traces and running thin wires from traces to correct pins.

KuboJakub, 2020-01-15:
But still it is switched also on 2. fig. Pin number 8 on the scheme is in the reality Vdd and pin number 4 on the scheme is in reality GND. Like it can be switched but in that case we get oposite amplification.

KuboJakub, 2020-01-15:
Ou of course now i get it what u meant. Sorry, yes it switching the pins should work.

gergelytakacs, 2020-01-15:
@KuboJakub @erik1392 

Thanks for the news, even though they were less than pleasant. In any case, please try to "patch" the problem to see if the shield is otherwise healthy. As soon as you get it working, we'll order a new PCB. But in any case, we should catch the rest of the mistakes. 

I can't believe that neither of the three of us caught this in time:)

KuboJakub, 2020-01-15:
Okay i repaired the mistake with GND and Vdd, unfortunately i cut one way i should not. But i repaired that one too.
But something is wrong with the transistor. When i pluged in 12V the collector start to burn... Right now i have no idea why that pin.
[image]
DAC works. When i send 4095 output from it is approximately 4.9V. Output from OPamp is in than 4V. This dropp could be caused because of my mystake (maybe not) but still cant see the reason of burning transistor. It is so also when i send 0V from DAC, so Transistor has some intern problem.

gergelytakacs, 2020-01-16:
@erik1392 @KuboJakub 

Okay, so as we have discussed here in the lab, you have soldered the protection diode the wrong way, so shorted the high voltage part... (Besides the issue with the OP-amp).

KuboJakub, 2020-01-16:
Now i have just finished the new shield. It is without the part of circuit
for measurements of the current and voltage and so far it works. So i hope
the only issues were that one we discussed. Next week i will solder the
rest.

gergelytakacs, 2020-01-17:
@KuboJakub 

All right, awesome. Let's hope the rest works as expected, then we'll order the corrected board and you can continue working on software/examples using the "patched" prototype.