gergelytakacs/AutomationShield #94: BoBShield: Week 8 (2/13)

gergelytakacs, 2019-02-18:
This issue details tasks to do and problems to solve for Week 8 (2/13 of the academic year).

- Determine the name of the device.
- Team leaders (HW, SW, Wiki). Team members? There can and should be overlap.
- Research any new ideas for sensing.
- Order and research sensors. What sort of hall sensor? Where to get linear pots? Any new ideas or concepts? Any new types of position sensors, direct or indirect?
- Order and research actuators. I have ordered a BLDC motor for trying. We have plenty of servos, these are easy to get. Try steppers? What about the stepper controllers?
- Where to get the tubes from? GT has ordered some, but until they arrive we should have something to work with.
- Where to get magnets from? GT has ordered some, but until they arrive we should have something to work with.
- Try actuators. Make calculations if the moment will be enough for e.g. a micro-sized servo.
- Everyone should register for Git.
- Do a Github tutorial. Learn Git.
- Learn Markdown.
- Learn DIP trace.

TiborKonkoly, 2019-03-13:
13.3.2019

This weeks results of the Hardware team:

- We have tried both continuous rotation servo and basic servo, our choice is the basic servo, because we can set the exact position in angles, which can make the future control easier. 

- We have also tried Hall sensors, the ordered ones, unfortunately they are NOK, they react for both poles of the magnet, that makes it improper for our application.

- We have also tried the cheap versions of the VL6180X TOF sensor, they are NOK, but we had good results with the VL53L0X, we will perform further tests with this one.

gergelytakacs, 2019-03-14:
Week 7
- Try what a magnetic cylinder, could it be used instead of a ball and combined with a Hall sensor?
- Finish trying the sensors. What is the FOV for the one and the other? How does that relate to the diameter and length of the tube? The TOF sensor is the biggesst issue.
- Improved mechanical design.
- Prototype with some level of functionality.
- Tube endings for the TOF sensor.
- How to handle the  wire going in and out of the I2C device.
- Which surface is the best for the TOF? What size should be?
- Change sampling to handle the `Servo` library. Recommended: make an #ifdef case for Servo, and change to Timer2 in that case.
- Test BLDC trial.

Conclusions
- Steppers should be ignored for this test.

gergelytakacs, 2019-03-21:
Week 8
- Magnetic cylinder - does the Hall sensor work if it is looking at the side of the cylinder. The cylinder must be able to roll, otherwise, if it slides, control and actuation is becoming really complicated. The cylinder must roll.
- Software team: demonstrate and test the changes in the sampling subsystem: show that sampling and servo work together nicely, at this point with the AVR architecture.
- TOF sensor: test the new chinese VL6x copies if and when they arrive, assume that the Adafruilt VL6x will be used.
- Create the ending(s) for the tube which will hold the TOF sensors.
- Test the Adafruit TOF VL6X with the endings and the tube.
- FOV for Adafruit VL6xxx and the Vishay one is still not know. Find it in the datasheet and provide me with hard data.
- How to handle the wire going in and out of the I2C device.
- Alternatives for the ball. Find some other types of balls that can go into the tube.
- Mechanical design improved.

TiborKonkoly, 2019-04-02:
Todays results:

- The sensor holders were printed, the dimensions are too accurate, it was hard to insert the sensors - the holder for VL6X was redimensioned, hopefully now it will fit

- We made a test with the VL6X sensor in the holder, unfortunately the results were not accurate. Small distances up to 55mm were ok, larger (from 55 to 100mm) values had some offset, which was not linear, mabe the inaccuracy is caused by the holder ? 

- Some FTC cables were selected, see G Sheet, but there might be an issue about how to connect the FTC to the sensor leads

* Ball alternatives
- Wooden balls: 
https://www.ebay.com/itm/Natural-Wooden-Craft-Wood-Balls-Sphere-Round-Craft-Supplies-6mm-to-75mm-Diameter/172754237776?hash=item2838f47950:m:ml45bA-qxB93iPQPHnm-m-g

- Cork balls:
https://www.ebay.com/itm/Gardner-Cork-balls-8mm-10pk-Fishing-tackle/152611949896?hash=item2388618d48:g:ldYAAMXQkl9Ra5ir

- Steel balls
https://www.ebay.com/itm/Lot-Dia-Bearing-Balls-High-Quality-Stainless-Steel-Precision-2-16mm-10-5000-x/253270729246?var=552409527955&hash=item3af81ca21e:m:mUigDETWeMHOzUlVwpIJxSw

- Plastic balls
https://www.ebay.com/itm/8mm-Polypropylene-PP-Solid-Plastic-Bearing-Balls-Precision-Sphere/323631916946?hash=item4b59f72792:m:mIFExhedg2JRb0o7lyU78xw

- Glass balls
https://www.ebay.com/itm/100-pcs-Glass-Ball-Outdoor-Hunting-Slingshot-Shot-Pinball-Catapult-Target-Ammo/302881701896?hash=item4685280008:m:miA9wxRylWVluTypzwjlTJA

gergelytakacs, 2019-04-03:
>    Ⓜ️ml45bA-qxB93iPQPHnm-m-g
>    Ⓜ️mUigDETWeMHOzUlVwpIJxSw
>    Ⓜ️mIFExhedg2JRb0o7lyU78xw
>    Ⓜ️miA9wxRylWVluTypzwjlTJA

Thanks for the update.

I think your biggest problem and issue are still the sensors. If the sensor
works otherwise, the issue is surely the holder or the tube itself. You
have no choice but to continue experimenting further. See if a different
holder ( without the inner ring) possibly helps. One thing would be to use
a larger diameter tube. Also, has the alternative sensor been tested? Do
not forget that you should properly test the sensor, before we can order
stuff off Mouser - this can cause further delay...

You are right, connecting the FCC to the breakout can be a problem.
Altough, I do not require FCC to be used in this case, it was just an idea.

Thanks for the list of alternative balls. This is helpful.

gergelytakacs, 2019-04-05:
@TiborKonkoly @Cortan 
Tips for creating the PCB:
https://www.seeedstudio.com/blog/2019/04/04/the-subtle-art-of-not-making-mistakes-with-pcb-design-for-manufacturing/?utm_source=Mailchimp&utm_medium=edm&utm_campaign=20190404&utm_source=Seeed+Fusion+Customers&utm_campaign=0334a06f1c-Mid-Autumn_festival_notice_COPY_01&utm_medium=email&utm_term=0_ec22c50d38-0334a06f1c-49585653&mc_cid=0334a06f1c&mc_eid=cbe130e8ab

gergelytakacs, 2019-04-05:
@TiborKonkoly I have bought 10 of each type of 8 mm ball, except cork...

TiborKonkoly, 2019-04-09:
Todays results:

- We assembled our first prototype which was tested, I´ve sent you a video about the test. In the future we can do more tests but I think this prototype is suitable for the SW team. 

- Mechanical design was not improved yet, we will consult it with Martin. 

- VCNL sampling was tested,  we used the basic example from the library, contains light and proximity measurement.  For the sampling the delay() function was used, it worked perfectly even with 3ms. 

- VCNL was also tested with te tube, but it is not suitable for our purposes. It provides non linear "unitless" data about the proximity, See picture *. 

- VL6180X was also tested with the tube, it measures the distance well. The sampling was also tested with the delay() function and we used the basic example. It worked well with 5ms. 

- According to our results, we have chosen the Adafruit VL6180X sensor for our project. It is more user friendly even for the SW team, they do not need to make calibration for the non linear values :)

- The part order list is completed for the project, I have to add my stuff, it will be done till  Wednesday. 

Non linear proximity values for the VCNL sensor
[image]

*
[image]

TiborKonkoly, 2019-04-11:
Outpot of the sensor VL6180X

[image]

gergelytakacs, 2019-04-12:
@TiborKonkoly Thanks for the writeup, this is at least the level of information I need to see your progress. Please do keep up the commenting.

About the "VCNL" thing - all righty. I did not know that it was unitless - that's actually not a great issue, but the nonlinearity is worse. You were correct about rejecting this possibility. So VL6x it is, the choice is clear!

I can look at the list of parts on monday at earliest. Until then, you are welcome to edit and update it.

However, at this point I really need to see a **working** prototype. I am still a bit worried wheter the sensor will work nicely... Anyways, keep up the good job and show me a working version...

gergelytakacs, 2019-04-12:
@TiborKonkoly On the output of the sensor: I don't really understand the image attached. Is this a single stationary position? What do I see on the picture? Ideally I'd like to see a reading with a "see-saw" motion, that could tell me something about the sensor...

mgulan, 2019-04-12:
Give him a minute:-)
We just put it there to back it up as it was logged on the lab PC at the last minute.

gergelytakacs, 2019-04-16:
@Cortan please double check prices. It is fine to leave it without VAT ("DPH"), we can just recompute at the end. But double check if they are correct.

lukasvadovic2, 2019-04-25:
My sampling library, there is a problem  with including .cpp file.
[sampling3.zip](https://github.com/gergelytakacs/AutomationShield/files/3116714/sampling3.zip)

gergelytakacs, 2019-04-25:
**Week 18**
Hardware team: Marko, Tibor, Pato - 10 bodov
- [ ] Functional PID control 10% + 6 b.
- [ ] Schematic and PCB Design finished - 10%
- [ ] Make the parts order w/ Mrs. Ikhradtova  - 10 %
- [ ] Improve mechanical design - 10%
- [ ] 3D design of the entire system 60 % 
+1 b for teambuilding beer w/ photo

Software team: Biro, Mlady, Vadovic - 10 bodov
- [ ] Solve the #include / header misery 30 %
- [ ] Sampling module 30%
- [ ] Functional PID control, full software functionality 20% + 6 b
- [ ] Method calibration() 5%. 
- [ ] Method sensorRead() 5 %. 
- [ ] Method actuatorWrite() 5 %. 
- [ ] Method referenceRead() 5 %
+1 b for teambuilding beer w/ photo

Documentation team: Marek, Rasto - 10 bodov
Example documentation is the HeatShield Wiki
- [ ] Basic description 10 %
- [ ] Electrical scheme image and description, downloadable file 20 %
- [ ] PCB layout image and description, downloadable file 20 %
- [ ] Table of components, w/ links 20 %
- [ ] 3D model, with downloadable sources 10 %
- [ ] Description of basic Arduino API Methods 20% 
+1 b for teambuilding beer w/ photo

gergelytakacs, 2019-04-25:
Thanks. The earliest I can look at it is Monday. Please do explore the
issue until then.

It bothers me that the Servo works and the others do not. Can you look at
this? Maybe inspect the rest of the libraries, how architecture-specific
code is segmented?

g.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

lukasvadovic2, 2019-04-26:
[image]

Thursdays team building :)

lukasvadovic2, 2019-04-26:
Hi there, so little update for our little-big cross-library/architecture library problem 👎 

I figured out why I wasn't  able to compile .cpp files from Sampling3 library - It was beacause in newer IDE versions  1.5.x+, the source code resides in the src folder and it all has to have certain structure, witch I was not following back than.

For further understanding of topic some infos on how does compilation works:

   -The headers (.h) are compiled all together and exchange informarion if included in one another, than they are compiled into an object.
   -All .cpp files in library folder are compiled separately, and all are compiled  into functional objects!!!
 See pic.1 - I added into 3 of .cpps in Servo library warning sentences - all of them are compiled:
[image]
pic.1

IDE has developed  preprocessor macros (ARDUINO_ARCH_XXX). Those are different from tokens and are possible to recognize even from .cpp files.
 That is how Servo library is able to activate only one .cpp file out of many in the folder.
thanks to ifdef condition (see pic.2)
[image]
pic.2

The problem is with recognizing diferent library tokens(Servo_h), in library Sampling. 
Step one:
.Ino file compiles (Servo.h included) and calls Sampling.h, witch recognizes insider tokens of Servo and sets to timmer 2 (just simplifing).
Step two:
.cpp file (All of them in library folder)  compiles and each one runs on different timer.... see on  pic.3
[image]
pic.3
![ewrfrefrwdf](https://user-images.githubusercontent.com/48005372/56809276-c472

So tokens are quiet useless in what we want to achieve.
I am going to put this theme on different forums and see if some (most likely Indian) guy is able to help us.

gergelytakacs, 2019-04-28:
@lukasvadovic2 #corleone68

My suggestion is as follows. Ignore the entire problem arising from token-dependent includes and create a copy of the `Sampling.cpp` and `Sampling.h` as (e.g.) `SamplingServo.cpp` and `SamplingServo.h`. We will try to look at this issue later, but this will allow you to test the actual hardware and not kill time with this important but not directly connected problem. 

Also, use #101 for communication regarding this issue and the entire sampling idea.

Cortan, 2019-05-01:
There is the PID regulation. 

[image]

- The setpiont for regulation is 45 mm, the starting position is with servo turned to 70 degrees. 
- There was a problem because it doesn't work with potentiometer, I couldn't set the setpoint with it. If I turned potentiometer everithing stopped.

gergelytakacs, 2019-05-02:
Oh, that looks promising!!! Nice job!:)

We'll analyze the results today, but I tell you in advance, that the pot is
not that important. If you set a nice profile of reference im software and
the PID compensates nicely, its already a win!

mgulan, 2019-05-02:
True that. So the green profile is.... input?

gergelytakacs, 2019-05-02:
I'd guess so. It's only a question of reference where to choose 0 degrees
to the servo...

g.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-05-02:
Week 19

Project head: Tibor

Hardware team: Tibor, Pato, Marko - 10 pts
- [ ] PID with reference tracking 25%
- [ ] Get the order from Mrs. Ikhardtova, send it to MOUSER 25%
- [ ] Improve mechanical design (servo is sliding in the socket) 25%
- [ ] Improve 3D design of the entire system  25%  
+1 pts. for teambuilding beer/wine w/ photo

Software team: Vadovic, Biro, Mlady - 10 pts
- [ ]  Solve the #include / header misery 30 % + 3 b.
- [ ]  Sampling module 30% + 3 b.
- [ ]  Functional PID control, full software functionality 20% + 8 b
- [ ]  Method calibration() 5 + 1 p%
- [ ]  Method sensorRead() 5 + 1 p%
- [ ]  Method actuatorWrite() 5 + 1 p%
- [ ]  Method referenceRead() 5 + 1 p%
+1 pts. for teambuilding beer/wine w/ photo

Documentation team: Marek, Rasto - 10 pts

Example documentation is the HeatShield Wiki.
The text must have flawless English.
- [ ]  Imrpove  basic description 10 %
- [ ]  Move all files and images into Wiki, do not use the Issues or the main repo. 20% 
- [ ]  Electrical scheme image and description, downloadable file 10 %
- [ ]  Improve PCB layout image and description, downloadable file 10 %
- [ ]  Table of components, w/ links 20 %
- [ ]  3D model, with downloadable sources 15 %
- [ ]  Description of basic Arduino API Methods 15%
+1 pts. for teambuilding beer/wine w/ photo

gergelytakacs, 2019-05-02:
@corleone68 @Cortan @TiborKonkoly  @lukasvadovic2  @raleh ...etc

Until I forget it: absolutely no superglue on the hardware!

(Leaves a nasty stain because of the vapors...)

Cortan, 2019-05-07:
Some measurments with PID regulation.

[image]

[image]

[image]

[image]

gergelytakacs, 2019-05-08:
@Cortan 

This, my dear sir, looks very promising. Nice job, kudos to you guys, I'm happy and satisfied! :) Especially the later ones (lower in the post) look very good.

[image]

Have you used the PID from the AutomationShield? What were the tuning parameters? What sampling rate?

Cortan, 2019-05-08:
Hi, we used the PID library made by Arduino. The PID library from AutomationShield didn't work. There was  lot af mistakes, we fixed all of them except one mistake  we couldn't fix.

gergelytakacs, 2019-05-09:
@Cortan The PID library included worked just perfectly before! How come it did not work now? Please do specify the errors and issues in more detail the next time. Nobody can do real work with vague statements like "it did not work" ...

lukasvadovic2, 2019-05-11:
[image]
[image]
just woke up from dead... so uploading our last tb :)

gergelytakacs, 2019-05-12:
@lukasvadovic2  See, you are very good at this!:)

lukasvadovic2, 2019-05-12:
@gergelytakacs Drinking beer you mean? :D

gergelytakacs, 2019-05-12:
@lukasvadovic2 Exactly!:)

gergelytakacs, 2019-05-16:
*Week 20*

TENTATIVE ONLY THIS IS A PLAN

At the meeting
- Register Git: Mlady, Marek
- Show PCB to guys
- 3D printing. Erik is gone?!?!
- Soldering - one by one. Start with 2 only. Headers to speed thing up?

Project head: Tibor

**Hardware team: Tibor, Pato, Marko** - 10 pts
- [ ] PID with reference tracking 25%
- [ ] Get the order from Mrs. Ikhardtova, send it to MOUSER 25%
- [ ] Improve mechanical design (servo is sliding in the socket) 25%
- [ ] Improve 3D design of the entire system  25%  
+1 pts. for teambuilding beer/wine w/ photo

**Software team: Vadovic, Mlady** - 10 pts
- [ ]  Solve the #include / header misery 30 % + 3 b.
- [ ]  Sampling module 30% + 3 b.
- [ ]  Functional PID control, full software functionality 20% + 8 b
- [ ]  Method calibration() 5 + 1 p%
- [ ]  Method sensorRead() 5 + 1 p%
- [ ]  Method actuatorWrite() 5 + 1 p%
- [ ]  Method referenceRead() 5 + 1 p%
- [ ] Which timers are used by the Zero and the Due? Will there be conflict with the current `Sampling`? 20% 
- [ ] How are custom I2C device libraries implemented in MATLAB for Arduino? 20% 
- [ ] How are custom I2C device libraries implemented in Simulink for Arduino? 20% 
- [ ] What is the highest theoretical Serial output on the Arduino Due? Why is the Serial not functional with certain speeds? (see #85).  20% 
- [ ] Does the avr-gcc compiler optimize out the `const` qualifiers or does it move these to RAM? In other words, does using `#DEFINE` directives instead of `const` produce differences? 20%
+1 pts. for teambuilding beer/wine w/ photo

**Documentation team: Marek, Rasto** - 10 pts
Example documentation is the HeatShield Wiki.
The text must have flawless English.
- [ ]  Imrpove  basic description 10 %
- [ ]  Move all files and images into Wiki, do not use the Issues or the main repo. 20% 
- [ ]  Electrical scheme image and description, downloadable file 10 %
- [ ]  Improve PCB layout image and description, downloadable file 10 %
- [ ]  Table of components, w/ links 20 %
- [ ]  3D model, with downloadable sources 15 %
- [ ]  Description of basic Arduino API Methods 15%
+1 pts. for teambuilding beer/wine w/ photo

lukasvadovic2, 2019-05-18:
[image]
Our todays PID results, using automationshield library.

TiborKonkoly, 2019-05-19:
Hello all,

Yesterday we worked on the tasks for the SW team, and, finally, I am glad to tell you, that all of the tasks are (at least partly) done. The bobshield branch is up to date, so it contains the actual version of the examples and functions.  We would like to demonstrate the functions and the examples on Thursday, using  the prototype.  After that, we should set final goals, how to improve the examples or the functions in case, if they are not in the desired form. 

Good work guys, we will finish soon! ;)

gergelytakacs, 2019-05-20:
@TiborKonkoly @lukasvadovic2 
Awesome work guys, I'm glad to hear the good news. Try to look at the set tasks for this week and finish those by Thursday, then we'll see where exactly we stand. 

[image]

However, these are very good news!

gergelytakacs, 2019-05-20:
@TiborKonkoly Fire and brimstone on any developer using nonsensical commits like `new` !
[image]

gergelytakacs, 2019-05-21:
@TiborKonkoly  and team. Please note that the usual time, Thursday 13:00 I will not be available. Can we meet tomorrow for cca. 30 min? Can you give me a time that is good for you in the AM?

Cortan, 2019-05-21:
Update about the order. The miracle happend, the invoice was paid.

gergelytakacs, 2019-05-22:
@Cortan Glory to the skies above, let the angels sing!:) So when should we have the package?

Cortan, 2019-05-22:
Hi! The package will be in Bratislava on Monday so I think we will have it on Tuesday, but there is a little problem because they sent us just 4 not 10 servo which we wanted for our BoBShield. Other parts are in the required numbers.

gergelytakacs, 2019-05-22:
@Cortan I'm guessing that's next Tuesday, not yesterday. Oh snap. It's good that @EvaVargova ordered independently some parts to make at least 1 build.

Have we ordered 10 servos? Have we paid for 10 servos? I'm just trying to see if we actually payed for 10, and if they will send it later or not....

Cortan, 2019-05-22:
We could have it next Tuesday. We paid for 10 servos, I will call to Mouser if they will send the remaining servos to us then I will write what they will say.

gergelytakacs, 2019-05-22:
*Week 20*
 Next meeting Wed, 11.

Project head: Tibor

**Hardware team: Tibor, Pato, Marko** - 10 pts
- [ ] Get the package 25%
- [ ] If you have the parts, manufacture 3 pcs. 50 %
- [ ] 3D print the parts, improve if necessary 25%
+1 pts. for teambuilding beer/wine w/ photo

**Software team: Vadovic, Mlady** - 10 pts
- [ ] Housekeeping in PID example 10 %
- [ ] Self test example + housekeeping (see Magneto) 20%
- [ ] Fully functional identification example (see e.g. Heat) 30 %
- [ ] Housekeeping in calibration() 10 % 
- [ ] Housekeeping in sensorRead() 10 %
- [ ] Housekeeping in actuatorWrite() 10  %
- [ ] Housekeeping in referenceRead() 10 % 

Iba Mlady:
- [ ] Which timers are used by the Zero and the Due? Will there be conflict with the current `Sampling`? 2 b 
- [ ] How are custom I2C device libraries implemented in MATLAB for Arduino? 2 b 
- [ ] How are custom I2C device libraries implemented in Simulink for Arduino? 2 b
- [ ] What is the highest theoretical Serial output on the Arduino Due? Why is the Serial not functional with certain speeds? (see #85).  2 b
- [ ] Does the avr-gcc compiler optimize out the `const` qualifiers or does it move these to RAM? In other words, does using `#DEFINE` directives instead of `const` produce differences? 2 b
+1 pts. for teambuilding beer/wine w/ photo

**Documentation team: Marek, Rasto** - 10 pts
Example documentation is the HeatShield Wiki.
The text must have flawless English.
- [ ] 3D model, with downloadable sources 1.5 b
- [ ] A bit smaller images for 3D print 0.5 B
- [ ] Description of basic Arduino API Methods 10 %, 2.5 b
- [ ] Improve the wiki a bit 30 %
- [ ] Prototype video (how the hardware looks, PID control with screen grab from Arduino) 60 %
+1 pts. for teambuilding beer/wine w/ photo

- Wiki better logic and indentation at top. Thighter 3D figures. Links for PARTS

gergelytakacs, 2019-05-22:
@Cortan Cool. I was worried about the paying part, since it is so complicated / slow. In that case they should send it in a separate package.

lukasvadovic2, 2019-05-23:
We have made the changes but as soon as we want to commit changes to BoB branch this error occures:
[image]
. Any idea what is going on?

Cortan, 2019-05-23:
Hello my friends! We have Christmas today because I got the package. Eva will take it to school tomorrow. Have a nice day!

gergelytakacs, 2019-05-23:
@TiborKonkoly 

You are trying to commit to master. *never* commit to master directly. Switch to your own branch, then commit.
[image]

gergelytakacs, 2019-05-23:
@Cortan

Joy oh joy!:) Thanks for the great job wrangling with the dark arts of ordering anything at STU! :) Awesome news.

lukasvadovic2, 2019-05-23:
> @TiborKonkoly @lukasvadovic2
> 
> You are trying to commit to master. _never_ commit to master directly. Switch to your own branch, then commit.
> [image]

That error ocures when I try to change branch :/

gergelytakacs, 2019-05-24:
@lukasvadovic2 

Can't really diagnose what's going on. Try to google the message, you also might need to resort to git command line to resolve the issue.

Cortan, 2019-05-28:
@gergelytakacs 

Hi! Updates about order. Estimated shipment date to deliver the remaining servos is on next Tuesday (JUN 04, 2019). See you tomorrow.

Have a nice day.

gergelytakacs, 2019-05-29:
@Cortan Thanks for the update. The rest of the parts are in L8 ready for assembly. They are located in the top drawer next to the door.

gergelytakacs, 2019-05-29:
*Week 20*
 Next meeting Wed, 11.

Project head: Tibor

**Hardware team: Tibor, Pato, Marko** - 10 pts
- [ ] Get the package 25%
- [ ] If you have the parts, manufacture 3 pcs. 50 %
- [ ] 3D print the parts, improve if necessary 25%
+1 pts. for teambuilding beer/wine w/ photo

**Software team: Vadovic, Mlady** - 10 pts
- [ ] Housekeeping in PID example 10 %
- [ ] Self test example + housekeeping (see Magneto) 20%
- [ ] Fully functional identification example (see e.g. Heat) 30 %
- [ ] Housekeeping in calibration() 10 % 
- [ ] Housekeeping in sensorRead() 10 %
- [ ] Housekeeping in actuatorWrite() 10  %
- [ ] Housekeeping in referenceRead() 10 % 

Iba Mlady:
- [ ] Which timers are used by the Zero and the Due? Will there be conflict with the current `Sampling`? 2 b 
- [ ] How are custom I2C device libraries implemented in MATLAB for Arduino? 2 b 
- [ ] How are custom I2C device libraries implemented in Simulink for Arduino? 2 b
- [ ] What is the highest theoretical Serial output on the Arduino Due? Why is the Serial not functional with certain speeds? (see #85).  2 b
- [ ] Does the avr-gcc compiler optimize out the `const` qualifiers or does it move these to RAM? In other words, does using `#DEFINE` directives instead of `const` produce differences? 2 b
+1 pts. for teambuilding beer/wine w/ photo

**Documentation team: Marek, Rasto** - 10 pts
Example documentation is the HeatShield Wiki.
The text must have flawless English.
- [ ] 3D model, with downloadable sources 1.5 b
- [ ] Improve the wiki a bit (better English) 30 % + 3 b
- [ ] Description of basic Arduino API Methods 10 %, 3.5 b
- [ ] Early prototype video (how the hardware looks, PID control with screen grab from Arduino) 60+6b %
+1 pts. for teambuilding beer/wine w/ photo

- Wiki better logic and indentation at top. Thighter 3D figures. Links for PARTS

Cortan, 2019-05-29:
@gergelytakacs 

I found the balls with diameter 7 mm. Here is the link:

[https://www.evrox.sk/e-shoploziska/eshop/1-1-Loziska/12-2-Gulky-valceky-ihlicky/5/7940-Nerezova-gulka-7-mm](url)

gergelytakacs, 2019-05-29:
@Cortan

Awesome, good job! Can you order these? I'll pay for the order personally
...

gergelytakacs, 2019-06-06:
roject head: Tibor

**Hardware team: Tibor, Pato, Marko** - 10 pts
- [ ]  Manufacture 3 pcs. 50 + 5 b%, tested and functional 
- [ ] Change the 3D part to better suit the motor 25 %
- [ ] Find other pipe or ball, so we can finish them all 25%
+1 pts. for teambuilding beer/wine w/ photo

**Software team: Vadovic, Mlady** - 10 pts
- [ ] Fully functional identification example (see e.g. Heat) 30 % + 3 b + screenshot or other sort of demonstration to Git. Try sampling 10 ms, same as PID.
- [ ] Housekeeping in PID example, more testing + screenshot or other sort of demonstration to Git., 30 %
- [ ] Self test example + housekeeping (see Magneto) 10 % 2 b + screenshot or other sort of demonstration to Git.
- [ ] Housekeeping with comment indents everywhere 10 % 
- [ ] Create only one BOB Header, no cpp. 10%
- [ ] Pull in changes from Master and try with new sampling 10 %.

Iba Mlady:
- [ ] Which timers are used by the Zero and the Due? Will there be conflict with the current `Sampling`? 2 b 
- [ ] How are custom I2C device libraries implemented in MATLAB for Arduino? 2 b 
- [ ] How are custom I2C device libraries implemented in Simulink for Arduino? 2 b
- [ ] What is the highest theoretical Serial output on the Arduino Due? Why is the Serial not functional with certain speeds? (see #85).  2 b
- [ ] Does the avr-gcc compiler optimize out the `const` qualifiers or does it move these to RAM? In other words, does using `#DEFINE` directives instead of `const` produce differences? 2 b
+1 pts. for teambuilding beer/wine w/ photo

**Documentation team: Marek, Rasto** - 10 pts
Example documentation is the HeatShield Wiki.
The text must have flawless English.
- [ ] 3D model, with downloadable sources 1.5 b
- [ ] Improve the wiki a bit (better English) 30 % + 3 b
- [ ] Description of basic Arduino API Methods 10 %, 3.5 b
- [ ] Early prototype video (how the hardware looks, PID control with screen grab from Arduino) 60+6b%
+1 pts. for teambuilding beer/wine w/ photo

gergelytakacs, 2019-06-18:
In case the BoBShield device is published in a paper or article, the BoBShield team has agreed in the following order of authors:

~ [...] Tibor, Pato, Vadovic, Biro, Marko, [...] ~

As the Wiki and documentation team did not have (much) direct impact on the project they will be mentioned in the Acknowledgement section:

~ [...] Rasto, Marek [...] ~

As the contribution of Samo is questionable, it will be later decided if he will be even in the Acknowledgements.