gergelytakacs/AutomationShield #96: BOBShield: ToF sensor

gergelytakacs, 2019-02-28:
Patrik is right, the VL63... ToF sensor measures from 5 mm! This is great news, as it adds the sensor as a possibility to the project! Therefore,
- Test the VL63x, possibly build it into an early prototype
- Test the maximal sampling frequency possible with the sensor. Check datasheets, but test it yourself too!

gergelytakacs, 2019-02-28:
@Cortan and @TiborKonkoly Please be aware that the ToF sensor we gave you may not be 5V tolerant. Just check it before use...

markomichal, 2019-03-07:
https://www.sparkfun.com/products/15177  Find out whether is this sensor suitable for our purpose.

gergelytakacs, 2019-03-07:
Thanks @markomichal will try this one out. Have not ran into this one
before.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-03-07:
@markomichal It is certainly a good find. Seems to be quite new, as I don't find it on ebay, that means I should try to get it directly from SparkFun... Its a cool device...

gergelytakacs, 2019-03-16:
@Cortan @TiborKonkoly 
How about (this one)[https://eu.mouser.com/ProductDetail/Adafruit/466?qs=sGAEpiMZZMsMyYRRhGMFNqmrFsVqHqPjS%252B%252BQdEqowdo%3D]?

They say it operates from 1 mm, see (here)[https://cdn-shop.adafruit.com/datasheets/vcnl4000.pdf]

gergelytakacs, 2019-03-16:
@Cortan @TiborKonkoly  PS. see the description [here](https://learn.adafruit.com/using-vcnl4010-proximity-sensor)..

TiborKonkoly, 2019-03-17:
It does look good, but what about this figure?  According to the figure I think it is the same problem as in some previous cases, it measures at an angle, and it would sense the tube.   

[image]

gergelytakacs, 2019-03-18:
Hmmm. I am not entirely sure that this shows the shape of the "cone". Look
at the caption, and look at the bottom axis - it says wavelength, the left
Y axis says spectral responsivity... Probably you guys should dig deeper
(in the datasheet) to find out. Anyways, a unit is on its way to try out.

Also what about the VL6xxx "cone shape"?

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-03-25:
@Cortan @TiborKonkoly  Be advised that the Adafruit VCNL4010 has already arrived. This should be thoroughly tested to determine if it is suitable for the application...

gergelytakacs, 2019-04-23:
The VL6180X was selected, because
- The VL5... is for long distances and is unreliable for this applicaiton
- The VCNL4010 is unitless and nonlinear
- The Chinese knockoffs are questionable (if they can be used) since the HW team was not very successful with it. Could be their 5 V compatibility... (or the lack of)

closed 2019-04-23 by gergelytakacs