gergelytakacs/AutomationShield #100: Rotational link shield

gergelytakacs, 2019-03-18:
@TiborKonkoly 

1. We need a name for your thesis project
2. The flat cables have arrived and they look very good for the application. The only question remains to find some suitable connectors for both ends.

gergelytakacs, 2019-05-02:
@TiborKonkoly 

Should it be LinkShield?

mgulan, 2019-05-02:
Either that or something like DampShield(?) unless it may get confused with Vibroshield.

gergelytakacs, 2019-05-02:
I'd say LinkShield simply...

g.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2022-05-19:
Naming has been decided as LinkShield

closed 2022-05-19 by gergelytakacs