gergelytakacs/AutomationShield #101: BOBShield: Timers and Libraries

corleone68, 2019-04-01:


corleone68, 2019-04-01:
Found a library using Timer2 for servo. In this case, we don't have to change Sampling library. Just need to test it

gergelytakacs, 2019-04-01:
Absolutely not. We are using our own library.

corleone68, 2019-04-01:
Understand, but we are including library Servo.h . Instead of that we can include this new library and leave Sampling.h as it was. That would be easy and simple.

gergelytakacs, 2019-04-01:
The sampling functionality must be under our control completely. By the
way, who said I want it to work only on AVR architecture? Current sampling
functionality in AutomationShield  handles other hardware as well. As I
said before - the current goal of the software team is to solve the case
when the Arduino servo library is used.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

corleone68, 2019-04-03:
Found out that suggested solution doesn't work. Sampling library couldn't recognize if library Servo is used. Including Servo.h into Sampling library is not a solution cos other architectures won't be supported.

Possible solution : additional Sampling library, only for BOBShield. 
Drawback :  Can't be used for other shields or architectures

gergelytakacs, 2019-04-03:
Well, the "sampling library" will not recognize if the servo one is used,
you have to program that by yourself. Worst case scenario, the user can
still define a flag in his main program - this still makes the
suggested/required solution possible.

g

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-04-18:
@corleone68 Can you please provide me with a minimal example of how the define checks fail in cpp due to complilation to objects? Also the page that explained how the checks are performed at compilation time?

gergelytakacs, 2019-04-18:
@corleone68 You guys (e.g.) were essentially right with this ifdef misery. Just look at the way how the Servo library itself handles #ifdefs: it uses them in the header itself! So, I don't think the key is to move the code to the .h file and get rid of *.cpp. Instead break up the sampling.h file to more parts and include the one you need based on the ifdefs:

So say you have:

Sampling.h
#ifdef SERVO_TOKEN
    #include Sampling_servo.h
#endif

#ifndef SERVO_TOKEN
     #include sampling_noservo.h
#endif

TLDR: leave the *.cpp alone, but break it up to sub-units and compile based on the preprocessor in the header, but avoid these tricks in the cpp, since we get o files.

@tiborkonkoly @cortan

gergelytakacs, 2019-04-23:
[image]

gergelytakacs, 2019-04-23:
[image]

gergelytakacs, 2019-04-23:
So a rhough recap of what could be done:
- Create the `/src/Sampling` folder.
- Leave the `Sampling.h` in the original `/src/` folder.
- The `Sampling.h` should first evaluate architecture by `#ifdef` macros based on tokens such as `ARDUINO_ARCH_AVR` and then use the correct version of the header and cpp file.
- Thus, in `/src/Sampling` create the same folders as in e.g. the `Servo` library: `avr`, `sam` `samd` for now.
- The original `Sampling.h` should then point to the correct folder with the given header and cpp, that could be called e.g. `SamplingTimer.h` and `SamplingTimer.cpp` - these are hidden from the user anyways and the object instance still remains `Sampling`.
- The `Sampling.h` should check for the presence of the `Servo` library after this. If on an AVR architecture, this can either call a different version of the header that is located e.g. in the `/Servo/` subfolder within `avr`, or use a different header name such as `SamplingTimerServo.h` etc.

gergelytakacs, 2019-04-25:
lukasvadovic2 and Corleone68

Could you maybe copy the `Servo` library to the `libraries` folder, then remove everything unnecessary from it to identify the problem w/ inclusion/compilation.

Also, does your header define the function/class prototype? You could just try a `void foo()` sort of prototyope, then a ` void foo(){}` sort of test function in the prototype?

Also, maybe installing a library through the GUI can add some sort of search path to the compiler. Can you try "installing" your test library to see what's wrong?

FYI: TiborKonkoly You should help the guys, since you will need the same functionality in your Master's project. So do not forget about that.

gergelytakacs, 2019-04-28:
@lukasvadovic2 You can accept my invite to the repo, you should have it your mail...

lukasvadovic2, 2019-04-28:
There is a guy on the forum who seemed to know what is he talking about : 
https://forum.arduino.cc/index.php?topic=612783.0
according to him, it is possible but really hard..

gergelytakacs, 2019-04-29:
Allrighty.

As I said before, let's sideline this issue for now, we'll get to it later.
It is really holding back progress.

Just copy (same style) the current `Sampling.*` and create a
`Sampling_Servo.*` And use that for now...

Also you can think of any other possible and reasonable solution.

gergelytakacs, 2019-04-29:
Things we (including me) all need to understand a lot better to resolve the conditional comply misery is:
- The scope of defines see [here](https://www.learncpp.com/cpp-tutorial/introduction-to-the-preprocessor/)
- That defines are understood sequentially,
- That defines cant go inside fuctions
- in what order are cpp files compiled
- what particular way does the Arduino IDE preprocess files
- where and when are hardware specific tokens defined

gergelytakacs, 2019-04-29:
@lukasvadovic2 

Okay: read your forum discussion. I've done my own reading and its getting clearer. So what we learned from this:
- #defines are only seen in individual translation units, as if they were local.
- the architecture identifiers are "global" since they are called first at compile time, in the command line, which we will not mess with.

The take away is, that now that we understand it a bit better, what I wanted was not possible. So the next best thing is truly your original idea: try to see how the Sampling functionality would look like if you did just that (within the *.h).

FYI: @corleone68 @TiborKonkoly @EvaVargova

gergelytakacs, 2019-04-30:
Here is the test for the object / class inclusion.
[main.zip](https://github.com/gergelytakacs/AutomationShield/files/3130286/main.zip)

lukasvadovic2, 2019-04-30:
Good (exhausting) evening :)
So there is a little problem, we did some adjustments for the library, ones that we have been talking during todays class (included in .zip file).

Our current problem is with the sketch. It is including all the libriaries, instead of one, for example MotoShield.h  we really #included in sketch . (see picture)
This should not be happening.
In fact a curious thing happend, at first it was compiling just one library that we included in the sketch.
But all of the sudden it has started to compile all of them...
[image]
[AutomationShield.zip](https://github.com/gergelytakacs/AutomationShield/files/3132185/AutomationShield.zip)

We also checked all includes in the path of MotoShield.h and non of them is connected to BoBshield or MagnetoShield..

from https://github.com/arduino/Arduino/wiki/Build-Process, is clear that it should not work like that :
The .c and .cpp files of the target are compi
led and output with .o extensions to this directory, as is the main sketch file and any other .c or .cpp files in the sketch and any .c or .cpp files in any libraries which are #included in the sketch.

Any taugths?

gergelytakacs, 2019-05-01:
Hello there!:) This is exactly what I told you yesterday. It does not
matter if `MotoShield.h` and `MotoShield.cpp` is not connected to the rest
of the code. The compiler will compile everything to objects, then the
linker will try to link it but discard what is not used. Thus, this is
normal. Say you have an error in `BoBShield.cpp` - this will affect the
compilation of even `MotoShield.cpp'.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

lukasvadovic2, 2019-05-01:
Thank you for your answer, our heads just didnt work properly after the whole day :D 
We continue in our work than.

gergelytakacs, 2019-05-01:
No worries;)

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-05-03:
Quite good read on the build process:
https://github.com/arduino/Arduino/wiki/Build-Process

gergelytakacs, 2019-05-03:
Please note that using timer 2 for servo (ie the suggested alternate library) will create further problems, as it has lower resolution. This may be a problem  with precision.

gergelytakacs, 2019-05-03:
Good read on extern in C https://www.google.com/amp/s/www.geeksforgeeks.org/understanding-extern-keyword-in-c/amp/
@corleone68 @lukasvadovic2

gergelytakacs, 2019-05-03:
@corleone68 @lukasvadovic2 

Directly including `Sampling` into the header (and not in CPP) is yet again not something that will work. In that case the header will be included in the individually compiled object files, thus creating multiple definitions of the same thing. Since all the objects (for various hardware) are compiled independently, all of these will use the AutomationShield.h header, and all of them will therefore try to implement their own classes for the same name, thus creating a conflict.

gergelytakacs, 2019-05-03:
@corleone68 @lukasvadovic2 

If you try to create separate classes for servo noservo, the compilation still fails saying `(.text+0x0): multiple definition of `__vector_11'`. This is because you try to assign the interrupt to an interrupt vector twice. This is caused by the last part of the *.cpp, where the interrupts are actually assigned.

gergelytakacs, 2019-05-03:
@corleone68 @lukasvadovic2 

Okay, here is what I learned so far
- `*.cpp` that is placed in a subfolder of `src` gets also built.
- Can't place the entire Sampling functionality in just the header, since this will be pulled in through `AutomationShield.h` to all hardware libraries (etc.), creating multiple definitions of  `Sampling` even without splitting it to two version.
- If you split `Sampling` and `SamplingServo` to two completely separate classes, there is still a build error, and that is caused by the ISR (re-definition).

gergelytakacs, 2019-05-03:
@corleone68 @lukasvadovic2 @TiborKonkoly 

The fact that we could not solve this still bugs me. Any one of you offering a viable solution gets an instant "A" for the course. So I'm still interested in a possible solution, if you investigate out of pure interest. There are a couple possible workarounds: we could try to make the Sampling library compatible with Servo by default. This would be one solution. Another solution could be that the user must include sampling alone, however, I'm not a big fan of this possibility.

After much contemplation, this problem seems to be actually much bigger than I originally anticipated. Therefore I now agree with you that `Servo2` should be used in this case. So for now forget about the Sampling library entirely and try to make use of the `BoBShield` using the alternate servo library.

Regarding this, I have a couple of questions though:
- There are lots of alternatives for the Servo2 library online. Which one do you plan to use?
- Will you lose on precision by using the 8-bit timer vs. the 16-bit timer of the normal Servo library?

The plan B is to take the Sampling library and make it work with Servo, so that it uses Timer2 by default... Actually this is only a problem on the UNO... So maybe this is also a way to go. Use Timer 2 on the UNO (AVR) use Timer 5 (16 bit) on the Mega and that way it should be fine...

gergelytakacs, 2019-05-03:
Would appreciate any ideas on how to do this: @mgulan @Gabor7697 @rkoplinger ?

[image]

gergelytakacs, 2019-05-03:
@corleone68 @lukasvadovic2 

After thinking about it some more; I think the best way to handle this is to change Sampling to Timer 2 for the Arduino AVR... I am afraid that because of the lower resolution of Timer 2, the precision can suffer affecting the hardware a lot. This is just a guess, but I don't want more problems now either;)))

gergelytakacs, 2019-05-05:
@corleone68 @lukasvadovic2 

Good job with the Sampling module, I like where this is going. 

[image]

Just a few possible things I've noted:

- The Mega is an AVR architecture too. So in `Sampling.cpp`  the second case on line `:49` is not activated, I think ,ever. (Am I wrong?:) If it sees a Mega, it will use the first case and not go to evaluate the second. So you can just change the first to `ARDUINO_AVR_UNO ` instead of `ARDUINO_ARCH_AVR`. I know that boards like the Nano and Mini are the same, however,  you cannot fit the AutomationShield device there anyways. So I don't really think we should be concerned with things like `__AVR_ATtiny24__` etc., just those boards that use the R3 pinout and can fit the boards.
- Please put the error messages that you parse in `Serial`, e.g. like 
```
    if(!setSamplingPeriod(microseconds)) 
    Serial.println("Sampling period is too long.\nMax is 4194303 microseconds."); 
```
to 
```
#if ECHO_TO_SERIAL
[...]
#endif 
```
since this is defined outside in `AutomationShield.h` , so power users can turn these on and off whether they like it. 
- No need to warn the user about Timer5, since Arduino does not really use it anywhere else.
- In your current implementation the maximal length is actually 1/16*1024*255=16320 microseconds (~15 ms), which is fine for the BoB shield. Just so you know, (I'll do this), this still needs to handle cases where this is too short of a sampling. This can be done , e.g. by setting the timer to e.g. 1000 microseconds by introducing a counter into the interrupt an count to a certain number. This would require making more cases for longer than ~15 ms sampling and also versions of the interrupt as well. As I said, if you are fine with this maximal sampling (<15ms) for BoB, you may leave this to me. My guess this is fine for BoB.
- You forgot to enable the interrupts for the Mega. You turn it off for the beginning then forget to turn it on. (It looks fine for the Uno).

gergelytakacs, 2019-05-05:
@corleone68 @lukasvadovic2 

Fixed some things, but only online - did not test it. So sorry if there are bugs, but I was trying not to introduce any.
- Fixed the `ECHO_TO_SERIAL` token. Actually, I think I was the one who forgot it:)
- Used the `ARDUINO_AVR_UNO` for the Timer configuration.
- Added the maximal possible period when using 8-bit timer in the comments.
- Added maximal error in the comments for 8-bit timer

gergelytakacs, 2019-05-06:
@corleone68 @lukasvadovic2 

Fixed basic functionality for the Uno:
- You forgot that other timer might have other pre-scalers, therefore the setting would be incorrect.
- You forgot to associate the Timer2 interrupt vector with the interrupt handler, there was still Timer1, so it was not working at all. 
- After some other changes:

@0.01 ms (absolutely not likely to be used) prescaler = 1
[image]

@0.100 ms (unlikely to be ever used) prescaler = 8
[image]

@ 0.5 ms (probably not used on the Uno) prescaler = 32
[image]

@ 1 ms, prescaler = 64
[image]

@ 2 ms, prescaler = 128
[image]

@ 4 ms, prescaler = 256
[image]

@ 15 ms, prescaler = 1024
[image]

gergelytakacs, 2019-05-06:
@corleone68 @lukasvadovic2 

For the atMEGA2560:
- There were two places saying `TCCR15B` which was not defined.
- Otherwise fine, but you can really see how using the 16 bit timer is much more precise.

Testing:
@4 ms, no prescaler
[image]

@30 ms, prescaler 8
[image]

@250 ms, prescaler 64
[image]

@1 s, prescaler 256
[image]

@2 s prescaler 1024
[image]

gergelytakacs, 2019-05-06:
@corleone68 @lukasvadovic2 

I've finished the sampling module. I've tested it for the Uno and Mega, checked periods with an oscilloscope and implemented long sampling periods over the possible resolution of the 8-bit timer. You may move to the BoBshield specific stuff and consider this done.

closed 2022-05-19 by gergelytakacs