gergelytakacs/AutomationShield #102: BoBShield: 3.3V compatibility

gergelytakacs, 2019-04-18:
Please do not forget to make the hardware compatible w/ both 5V and 3.3V logic. Should not be an issue, but if you are in doubt, ask.

gergelytakacs, 2019-04-23:
- According to a test (23.4) the Servo can take 3.3 V logic, while being powered by the 5 V rail. According to a Google search, someone tested various servos like this without any problems.
- The `AREF` should be connected to the 3.3 V rail (for the potentiometer)
- The potentiometer should be powered from the 3.3 V rail.
- The VL6180 should be powered from the 3.3 V rail, just to be sure. Does the VL6180 produce 5 V logic when powered from 5 V or does it still produce 3.3 V. THis should be even put on an oscilloscope to find out.

TiborKonkoly, 2019-04-23:
On the first picture we can see the result of the BoB shields first open loop test. The picture was taken from the Arduino IDE Serial Plotter. At this case both the sensor and the servo motor were connected to 5V. 

[image]

On the second picture we can see the result of the same test, which was executed at the same conditions, like in the previous case. The only difference is, that the sensor was connected to 3.3V. As we can see, the results are similar.

[image]

On the third picture is the result of the same test. In this case both, the sensor and the servo motor were connected to 3,3V.  If we look at it properly we can distinguish slight differences between this picture and the previous pictures. 

[image]

gergelytakacs, 2019-04-24:
Thank you for the exhaustive test and the results. See, this is something I
expect every time!:)

 To me, they seem pretty much identical and besides a phase shift, I do not
really see any difference that could be explained by the change in the
logic level voltages. I think, therefore, that both the sensor and the
motor will work fine with 3.3V and 5V logic boards.