gergelytakacs/AutomationShield #104: BoBShield: sensorRead() method

gergelytakacs, 2019-04-23:
The `sensorRead()` mehod
- Must return the position of the ball in mm
- Must correct for the shift in the zero position based on the `calibration()` method
- Must use the Adafruit VL6180X library that is located as the submodule in `src/lib`

gergelytakacs, 2019-04-23:
I have added the VL6180X library as a submodule.

gergelytakacs, 2019-04-28:
You guys said you did not have the other submodule (for the VL5x...) working, when you cloned the repo. That is because when working with submodules, you have to initialize them by calling `submodule update --init --recursive`

gergelytakacs, 2019-04-29:
[image]
- No `Serial` in the methods.
- Please comment in detail.
- How does the `calibration()` affect this function? (It doesn't). The minimal value should be calibrated, but only if the user has run the calibration sequence. So if it has been calibrated, then the minimum is that value... Instead of using "minimum" you can store "minimum calibrated" - that is then rewritten in the calibration function...
- The `status` flag is very interesting. What sort of errors does the VL6180 produce? Can this be used to e.g. interpolate values if reading fails? Please research the possible errors...
It could be (in pseudocode) - just an idea:
```
if (status == someerror)
  output the previous value... (or some other interpolation)
end
```

corleone68, 2019-04-29:
solution for cloning AutomationShield library with submodules worked, but it's without characters '<' and '>'

gergelytakacs, 2019-04-30:
@corleone68 Sure, the <> was only there to denote "insert the URL" here and that the given part is not a literal command. Git syntax doesn't use it, sorry if that was confusing.

gergelytakacs, 2019-05-05:
@corleone68 @lukasvadovic2

- Just purely to keep consistency for the API for the user, please call the "default" sensor read function `BOBClass::sensorReadMm()` instead of `BOBClass::sensorReadMm()`. All devices have this function... It is fine to leave `sensorReadPerc()` as is, for an alternative. You are also welcome to wrap `sensorReadMm()` in `sensorRead()` to achieve the same effect.
- Why is the sensor reading function returning nothing? (`void BOBClass::sensorReadMm()`). It should return a floating point number.
- You have nicely made a flag that says if the thing has been calibrated (`calibrated=1`) but then you are not really using it. :)

corleone68, 2019-05-06:
How should we call it please ? Cos those two names are the same
[image]

gergelytakacs, 2019-05-06:
Ok sorry. As it is written in the guide in the sticky note (look up), it
should be sensorRead().

All of the boards have this function name.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-05-20:
- Yet again it's an uncommented, mess, so please do comment. 
- Also where did you take "4" from the middle from? Will this be the same for all future shields?
- Why don't you use `sensorRead()` in mm in `sensorReadPerc()` anyways? You can call a method inside a mehtod... I would just ensure that my `sensorRead()` works well. 
- Also the read function does not know if the calibration routine has been launched or not... Why? That's what for the `calibrated` flag is for!

closed 2022-05-19 by gergelytakacs