gergelytakacs/AutomationShield #106: BoBShield: begin()

gergelytakacs, 2019-04-23:
The BoBShield Arduino API shall have a `begin()` method, that can possibly
- set pin modes, if necessary (probably not)
- initialize the `Servo` object and pin
- etc.

gergelytakacs, 2019-04-29:
[image]
- No Serial in the code
- Keep naming consistent, this should be `begin()`
- The self-test for the sensor should not be used here, instead you can make a testing example for the entire board, like it is done for the MagnetoShield

corleone68, 2019-04-29:
understood, thanks for the feedback
Could Servo pin and be initialized in .ino ?

gergelytakacs, 2019-04-29:
For a quick test yes, but for the final software no.

So, If you are stuck, initialize in *ino. Otherwise experiment and try to
initialize within the BoBShield API.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-05-05:
@corleone68 @lukasvadovic2

I do not think `  pinMode(POT_PIN, INPUT);			// set Potentiometer pin` is necessary. It is an analog pin, I don't think you need to set the direction for it...?!?

gergelytakacs, 2019-05-09:
- Pot input initialization is possibly not necessary.
- Choose a nicer name than `myservo`. Not `butterfly`.

raleh, 2019-05-16:
Can You @corleone68 or @lukasvadovic2 look at the description of the code on the wiki and advise me how to describe it better.

gergelytakacs, 2019-05-16:
@raleh Yes, however, due to the unfortunate events please bug @lukasvadovic2 and S. Mlady... (Also he should send me his GitHub handle so I can add him to the project).

gergelytakacs, 2019-05-20:
Is `pinMode` even needed for analog inputs?
[image]

closed 2022-05-19 by gergelytakacs