gergelytakacs/AutomationShield #107: BoBShield: Electronic and PCB Design

gergelytakacs, 2019-04-23:
- The board should be 3.3 V compatible

gergelytakacs, 2019-04-23:
[image]

gergelytakacs, 2019-04-23:
- Use the Magneto Shield (R2 / R3) for reference
- Use the "CA9" sized pot for the design, this is the smaller one we own.
- Use 3.3 V reference and logic (see another issue)

TiborKonkoly, 2019-04-23:
The schematic and PCB design of the BoB shields first version is included in the attached .zip file. 

[BoB.zip](https://github.com/gergelytakacs/AutomationShield/files/3109671/BoB.zip)

gergelytakacs, 2019-04-24:
Thank you @TiborKonkoly , will try to review your design ASAP.

TiborKonkoly, 2019-04-24:
I have made the requested changes. We can make a final check tomorrow at school, I would like to check the placement of the connectors together, just to make it sure.

So here is the scheme:
[image]

The upper part of the PCB:
[image]

The bottom part of the PCB:
[image]

And finally the .zip file:
[BoB1.zip](https://github.com/gergelytakacs/AutomationShield/files/3113811/BoB1.zip)

gergelytakacs, 2019-04-25:
Awesome, we'll check it out today. Only likely to add small changes:
- Mark the footprint of the mechanical connection on the "Top Assy" (gray)
layer, so we see where the 3D printed part goes.
- The GND of the potentiometer goes all the way to the right, to the C1.
There is a GND connector pin closer to the pot (Top left).
- The www address and the name, date on the top left corner should be
adjusted to the top. They are quite randomly thrown around...

gergelytakacs, 2019-04-25:
@TiborKonkoly Please finish and post the final schematic plan and PCB design at most by Monday morning. I would like to order the PCB by Monday evening...

TiborKonkoly, 2019-04-25:
Hello all! I have realised the requested changes and also  changed the board layout a bit, here are the results: 
[image]
[image]

- The POT was connected to the GND
- The placement of the 3D printed components is marked by the rectangles
- The text on the bottom is arranged
- I have also changed the diameter of the holes from 4mm to 3mm

Here is the included .zip file: 
[BoB_final.zip](https://github.com/gergelytakacs/AutomationShield/files/3118416/BoB_final.zip)

It should be ok, but please check it.  Have a nice weekend! :)

gergelytakacs, 2019-04-26:
Looks good to me actually. Double check everything with a clear head until Monday, then I can push it to production.

@mgulan  @erik1392  please have a look at this and check the design, or else suffer in silence if something is not to your liking;)

mgulan, 2019-04-29:
Looks allright to me too, incl. the small changes. I say go for the PCB order👍🏻

gergelytakacs, 2019-04-29:
Thanks, will do. The TugShield (Evka) will be ordered too, but that project
is in a sorry state at the moment, she did not manage timing and scheduling
well.

gergelytakacs, 2019-04-29:
[image]

[image]

gergelytakacs, 2019-04-29:
[image]

gergelytakacs, 2019-04-29:
[image]

gergelytakacs, 2019-04-29:
[image]

gergelytakacs, 2019-04-29:
Here is the "final" version that went in the production. The Wiki team will need these:
[BoBShield_R1_Final.zip](https://github.com/gergelytakacs/AutomationShield/files/3126564/BoBShield_R1_Final.zip)
[BoBShield_Production_R1.zip](https://github.com/gergelytakacs/AutomationShield/files/3126563/BoBShield_Production_R1.zip)

raleh, 2019-05-01:
These zip files should be uploaded on some servers or can remain on github, i.e. to put them to github in the PCB lajout section, as You put them here?

gergelytakacs, 2019-05-01:
All files must remain on GitHub. You have to use the AutomationShield.Wiki,
which is a separate repository from the AutomationShield code repository. I
have sent you an invitation, so you can use Git to manipulate it.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-05-01:
[image]

gergelytakacs, 2019-05-03:
@cortan @TiborKonkoly
PCB delivery is advised on next Thursday, I can bring it to school next Friday.

Make sure you follow up on the MOUSER order.

gergelytakacs, 2019-05-10:
@Cortan @TiborKonkoly 
- My wife let me know that she's got the package from the PCB manufacturer.
- @Cortan Got an advisory from our economic system that the payment is in the process. You should try to get the official order form (signed (?) by the dean etc.) and send it to Mouser ASAP.

gergelytakacs, 2019-05-10:
@corleone68 @Cortan @TiborKonkoly @raleh @lukasvadovic2 

[image]
[image]
[image]
[image]

TiborKonkoly, 2019-05-11:
They look really good. I can´t wait to test them! :)

gergelytakacs, 2019-05-12:
> They look really good. I can´t wait to test them! :)

Same here, man. :)

gergelytakacs, 2019-05-24:
@Cortan @TiborKonkoly 

The package is in the lab, I've placed everything in the top drawer next to the door. I've checked everything except the FFC connectors, since I have no idea which ones you've chosen. Everything you ordered is there, except the 6 missing servos which (if we paid for them) @Cortan should make sure we also get it. The capacitors will be a bitch to solder on, since they have no side only bottom pads... But can be done, I've tried it - but you have to use a heat gun and it's a bit tricky.

Please note, mainly @TiborKonkoly, that I would like to see not only "kind of" working and assembled devices, but neat and precise work with the soldering etc. So don't rush it, take care that the end result looks nice and presentable.

gergelytakacs, 2019-05-24:
@TiborKonkoly @Cortan 

Checked the connector it fits nicely. However, the cable is kinda "hard" I'd expect more flexibility. Might still work though. You should have used the really flat ~1 mm pitch type film ones...

Checked the ones @TiborKonkoly  ordered for the upcoming Link. The ones with the 2.54 mm pitch are of the same type - the wire is way too rigid for the Link shield. You will need to use the really flat ones with 1 / 0.5 mm pitch.

closed 2022-05-19 by gergelytakacs