gergelytakacs/AutomationShield #108: BoBShield: Generic software pointers

gergelytakacs, 2019-04-28:
- Do not print to serial by default from the library. No `Serial.print()' in these.
- Any self-test functionality should be implemented in a specific example. See e.g. [this](https://github.com/gergelytakacs/AutomationShield/tree/master/examples/MagnetoShield/Support/MagnetoShield_SelfTest).

corleone68, 2019-04-28:
Okaay thanks for the info

gergelytakacs, 2019-04-29:
Also checked the pinned Issue for software hints...

You should comment any generic idea on software development for the AS
project there.

G

gergelytakacs, 2019-04-29:
[image]

gergelytakacs, 2019-04-29:
What is this? 

Is this necessary?

[image]

gergelytakacs, 2019-04-29:
[image]
This is fine, but
- Potentiometer not potenciometer
- Please comment line-by-line.

corleone68, 2019-04-29:
First one is not used and needs to be removed. Initial idea was to create a function for BaudRate (found in libraries tutorials), but it not necessary. 

Second issue - debug() is used in FloatShield to check if sensor was initialized.

Third issue - yeaah typo :D

gergelytakacs, 2019-04-29:
You can and should use a sort of debug functionality. Create an example
that provides a self-test function for the system. Look at the
MagnetoShield how this can be done.

g.

Best regards / S pozdravom / Udvozlettel;

Gergely Takacs, PhD.
Associate Professor
___________________________________________________
   web:       gergelytakacs.com
   e-mail: gergelytakacs@gergelytakacs.com
   mobile: +421(0)907 165 720

<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
I'm using Inbox When Ready
<https://inboxwhenready.org/?utm_campaign=signature&utm_medium=email&utm_source=signature>
to protect my focus.

gergelytakacs, 2019-05-20:
Please comment your code!
[image]
????
[image]

gergelytakacs, 2019-05-20:
Hello there;

I've added CI for automated build checks. Your branch status is:
[image]
Every time you commit the entire system is automatically re-built. You can click on the link to see how this looks like.

gergelytakacs, 2019-05-30:
According to @TiborKonkoly the temporary (from the mechanical standpoint) prototype based on the PCB is working. Please @lukasvadovic2 update the code ASAP and push to your branch, so I can review it...

closed 2022-05-19 by gergelytakacs