gergelytakacs/AutomationShield #111: BoBShield: Wiki

gergelytakacs, 2019-05-01:
Please use this for Wiki related communication.

raleh, 2019-05-02:
I created a wiki page for BOBShield here is the link https://github.com/gergelytakacs/AutomationShield/wiki/BOBShield. Look and give an opinion, what else should be added or edited. I wrote two sections on the wiki page. As for the SW team, I asked them to send me a code, but they replied that they would gladly send, but that their code still does not work. When they send me the code, then I will edit the Arduino Library section. Also Marek did not give me any of his documentation, so I can not continue any further.

gergelytakacs, 2019-05-08:
Never really had any sorts of issues with this. When I clone it through GitHub desktop, I can work with it simply through the graphics interface. I cloned by clicking on the web Wiki, then I can  Push/Commit/Pull etc through the graphical software. I have never used the command line to work with the Wiki, so I don't know how that behaves. I cloned it locally several times without issues by this method.

Mine looks like this:
[image]

[image]
Thus my address is also `~.wiki.git`. I don't see an issue with that. You have to realize that the `Wiki` is most probably a repository on its own and behaves as any other repository.

I tried to pull from the command line, it works just fine:
[image]

gergelytakacs, 2019-05-08:
Also, why do you live in the future?:) 
[image]

raleh, 2019-05-10:
I had success in uploading new images and rar and zip files, so links were changed and they are downloadable. Now the problem is the table, I worked it through the Markdown Tables generator
[image] 
[image]
and then I copied it into a git. But in the git will not appear as a table but as plain text. As You can see on https://github.com/gergelytakacs/AutomationShield/wiki/BOBShield. It's already going on my nerves, because the table will not appear as it should. I will now continue to add and improve descriptions in sections.

mgulan, 2019-05-10:
Did you try to look at a wiki template of another AS shields?

raleh, 2019-05-11:
Yes, I looked. They are same as my table, I don't know what the problem is.

gergelytakacs, 2019-05-12:
> Yes, I looked. They are same as my table, I don't know what the problem is.

Have you looked if your text has any special characters that MarkDown may interpret differently? You can also try to send the table to me as - say - excel and I'll try to see what's wrong.

gergelytakacs, 2019-05-12:
@raleh @mgulan 

You need a free line between text and table. It's fixed now. You are welcome. :)

gergelytakacs, 2019-05-12:
@raleh 

Please make this "denser" and nicer, see HeatShield how:
[image]

gergelytakacs, 2019-05-12:
@raleh Please use less empty space with the renders. If the renders have a big enough resolution (they do, I've checked), you don't need new ones, just crop it out a bit.

[image]

gergelytakacs, 2019-05-12:
@raleh What causes the blue lines on the top and left of the PCB? If you can, please remove these.

[image]

gergelytakacs, 2019-05-12:
@raleh 

You may already write more on the methods used in the API! We already know what they *should* do - please consult with the software team.

[image]

gergelytakacs, 2019-05-12:
@raleh 

Please use clickable links for the hardware. 

[image]

raleh, 2019-05-16:
Thank You for correcting the table, I did changes You told me to change. I'm going to write to the software team, let them send me improved code description.

gergelytakacs, 2019-05-16:
Welcome. We agreed today on how to proceed, so please do that.

closed 2021-12-22 by gergelytakacs