gergelytakacs/AutomationShield #119: TugShield:begin()

EvaVargova, 2019-05-07:
void TugShieldClass::begin(){  
      pinMode(TUG_YPIN, INPUT);  		// Flexi pin
      pinMode(TUG_UPIN, OUTPUT); 		// Servo pin
	  servo.attach(TUG_UPIN);			// Set pin for servo
	  servo.write(0);					// Set zero position 
Mala by som tu dať aj Serial.begin(9600) ?

gergelytakacs, 2019-05-07:
> @gergelytakacs <https://github.com/gergelytakacs>.

- English only, please.
- No, do not use Serial inside the API. Initialization is up to the user in
the application. - Please read issue #42 for some generic information.

gergelytakacs, 2019-05-14:
`pinMode()` is used to set the direction of digital pins. You do not have digital pins (in the classical sense) in your shield. As far as I know, you don't need to set the direction for analog pins (Flexi pin, potentiometer pin), while I think the Servo library takes care of these things. So I'd remove them, but test evertything to be sure...

[image]

gergelytakacs, 2019-05-22:
For now finish the thesis, but after that:
- The comment says that you set the position to zero (of course that you choose freely), however, the define token `SERVO_MAX` says "max" that could be confusing later. So I suggest you change the name of this token to something more sensible. Like `SERVO_ZERO` or something that makes sense when you read the code.
- Align the code lines nicer so it's more readable.
- English comments only.

closed 2021-12-22 by gergelytakacs