gergelytakacs/AutomationShield #120: TugShield: Thesis

gergelytakacs, 2019-05-07:
Thesis outline:

1. Úvod (1-2 s.)
2. Motivácia (5 s.)
3. Tvorba hardwaru (10-15 s.)
 - Komponenty
 - Vývoj predbežného protoypu
 - Vývoj plošného spoja
 - Finálny prototyp
4. Tvorba aplikačného rozhrania 
 - Knižnica AutomationShield (súčasný popis)
 - Vývoj API pre TugShield (progress, základné princípy, problém so Sampling etc., samotná trieda TugShield)
 - Metódy v TugShield (pod-podkapitoly pre každú metódu
5. Príklady 

6. Záver (1 s.)

gergelytakacs, 2019-05-10:
@EvaVargova (A bit unrelated to the issue but) ...the PCB have arrived.
 
[image]
[image]
[image]
[image]
[image]

EvaVargova, 2019-05-10:
:D I am very happy....Thanks!

gergelytakacs, 2019-05-11:
> :D I am very happy....Thanks!
I do hope the MOUSER parts will be here soon as well. Liaison w/ @Cortan about this, fingers crossed.

gergelytakacs, 2019-05-29:
Thesis has been submitted.

closed 2019-05-29 by gergelytakacs