gergelytakacs/AutomationShield #122: TugShield: actuatorWrite()

gergelytakacs, 2019-05-08:
The TugShield `actuatorWrite()` will be a `void` function that takes a floating point number of the angle and then sets the servo to that certain angle. Please note, that the angle is always relative. It depends on what you choose and how you assemble the device. So you must really think about how to assemble it at the beginning, the angles way only make sense (across devices) then.

gergelytakacs, 2019-05-08:
Note that this is related to #105

gergelytakacs, 2019-05-14:
- Please change the name of this variable to something more sensible, e.g. `servo_angle`.
- I'm not sure what is the reason to have this variable as a public one, as opposed to simply let it live inside the method. It is after all not really needed anywhere else, or is it?
[image]

EvaVargova, 2019-05-16:
[image]
[image]
I think Tugshield.h for some reason cant include Servo.h.
I have tried already this: 
[image]
and this:
[image]
but still don't work.

gergelytakacs, 2019-05-16:
You have not initialized the object `servo`. You have the class `Servo` but no object `servo`. In the beginning, you shall initialize the object by `Servo servo;`. Also please note that you are mixing the reference to the object (`servo`, upper code portion) with the class (`Servo`, lower code protion) that may create some trouble.

gergelytakacs, 2019-05-22:
In your header, there are some issues with this, namely:
[image]
- You don't need a variable in the prototype function, just the input argument type
- The type should be `float` and not `int`. Of course this affects the rest of your code (possibly)

closed 2022-05-19 by gergelytakacs