gergelytakacs/AutomationShield #123: TugShield: sensorRead();

gergelytakacs, 2019-05-08:
The TugShield `sensorRead()` method will take no arguments and will return the percentual deformation of the beam as a floating point number. Please use the `OptoShield` class as an inspiration.

First the method shall check if the user has run the `calibration()` method or not (see #121). This is based on a flag. If not you will set the minimal and maximal values to some sort of defaults, which you can determine based on experimentation. The analog reading can be then mapped (these are floats, so use `mapFloat()` from `AutomationShield` to give the percents.

gergelytakacs, 2019-05-14:
Same issue as in #122. I do not think you need private variables in the class, as you do not share these *across* the methods. The only exception is `iscalibrated` (or whatever you call it), because that one you set in the calibration method, but use it in the sensor method.

[image]

EvaVargova, 2019-05-15:
[image]
I have a problem with this method....IDE show me this error:
[image]
I really don't know where is the problem. Method should return float _sensorValue.

gergelytakacs, 2019-05-15:
@EvaVargova

Allright. What does the `.h` say? Because it seems to me that your method prototype is incorrect.

Now I have looked at your `.h` and it prototypes `void sensorRead()` so that's why the compiler things these are actually two different methods, since have two different output types.

gergelytakacs, 2019-05-22:
- Seems ok by visual inspection.
- Horizontal align the code lines nicer so it's more readable.
- English comments only.

closed 2022-05-19 by gergelytakacs