gergelytakacs/AutomationShield #124: TugShield: Identification example

gergelytakacs, 2019-05-08:
This example will be based on the HeatShield identification example. You may copy the example as is, and do the minimal necessary changes; namely change sampling to much shorter (e.g. 0.1 s...) and change the input trajectory.

The goal of the example is to produce an open-loop test for system identification purposes. That means, that the servo will be set to a range of step changes in angle, then the output (calibrated) will be read. The input and output are passed to the serial port.

It is up to you what input trajectory you choose. Mind you that the trajectory in HeatShield is percents of power, you should change that angles you choose.

gergelytakacs, 2019-05-14:
You now have a sampling of 50 ms which sounds about right. Look at `Sampling.period()` where you multiply by 100, this should be 1000 (milliseconds to microseconds) and then at the beginning you should have 50 instead of 500 (ms).
[image]

gergelytakacs, 2019-05-14:
The variable `T` holds how long an experimental "section" (saved in `U[ ]`) should last. This should not be too long or too short, and you have to determine by experiment. So how did you get 1350?:) Also the comment is wrong for your case.
[image]

gergelytakacs, 2019-05-14:
- Please provide an actual experiment so I can see how it looks like.
- If you (and I) are satisfied with the experiment, you need to log and save the results to Matlab. (and display it from there in your thesis).

gergelytakacs, 2019-05-22:
I see no change, improvement or results currently (22.5.2019, 8AM) :-1:  :confused:

EvaVargova, 2019-05-22:
I have finished this example. 
[image]
It looks fine to me.

gergelytakacs, 2019-05-23:
Hey there, good job. Agreed.

This can go into the thesis, I think it looks quite fine. Please repeat the experiment and export the results to a MATLAB plot so it looks nicer in the thesis.

closed 2022-05-19 by gergelytakacs