gergelytakacs/AutomationShield #125: BoBShield: Identification example

gergelytakacs, 2019-05-08:
This example will be based on the HeatShield identification example. You may copy the example as is, and do the minimal necessary changes; namely change sampling to much shorter (e.g. approximately 0.1-0.01 s...) and change the input trajectory.

The goal of the example is to produce an open-loop test for system identification purposes. That means, that the servo will be set to a range of step changes in angle, then the output  ranging (calibrated) will be read. The input and output are passed to the serial port.

It is up to you what input trajectory you choose. Mind you that the trajectory in HeatShield is percents of power, you should change that angles you choose. Nobody said that angles cannot be positive and negative, this is up to you and how you choose them to be interpreted (and how the hardware is assembled).

gergelytakacs, 2019-05-20:
Please do a little housekeeping and decide which version do you want to keep
[image]

gergelytakacs, 2019-05-20:
(I can't even decide which ones to test etc.) You pushed a mess into the repo!

[image]

lukasvadovic2, 2019-06-06:
So I attach two graphs, first one is with the ball all the times stucking for a while at the end near a sensor (the flat part),
Second one is better but the ball was not going all the way it can..
[image]
[image]
also the code:
`#include <BOBShield.h>                // Include the library
#include <SamplingServo.h>            // Include sampling

unsigned long Ts = 14;            // Sampling in milliseconds
unsigned long k = 0;                // Sample index
bool enable=false;                  // Flag for sampling 

float y = 0.0;            // Output
float u = 0.0;            // Input (open-loop)
float U[]={-30.0,-29.0,-28.0,-27.0,-26.0,-25.0,-24.0,-23.0,-22.0,-21.0,-20.0,-19.0,-18.0,-17.0,-16.0,-15.0,-14.0,-13.0,-12.0,-11.0,-10.0,-9.0,-8.0,-7.0,-6.0,-5.0,-4.0,-3.0,-2.0,-1.0,0.0,1.0,2.0,3.0,4.0,5.0,6.0,7.0,8.0,9.0,10.0,11.0,12.0,13.0,14.0,15.0,16.0,17.0,18.0,19.0,20.0,21.0,22.0,23.0,24.0,25.0,26.0,27.0,28.0,29.0,30.0};  // Input trajectory
int T = 1;              // Section length (appr. '/.+2 s)
int i = 0;              // Section counter
int a =0;               //flag for BoB tube shifting right / left

void setup() {
 Serial.begin(115200);                // Initialize serial
 BOBShield.begin();                   // Define hardware pins
 BOBShield.initialize();              // Check if Adafruit VL6180 sensor is available

   // Initialize sampling function
  Sampling.period(Ts *1000);         // Sampling init.
  Sampling.interrupt(stepEnable);     // Interrupt fcn.
}

// Main loop launches a single step at each enable time
void loop() {
  if (enable) {                      // If ISR enables
    step();                          // Algorithm step
    enable=false;                    // Then disable
  }  
}

void stepEnable(){                   // ISR 
  enable=true;                       // Change flag
}

// A signle algoritm step
void step(){ 
if (a==0){                           // if tube is shifting right
if (k % (T*i) == 0){        
  u = U[i];                          // Set reference by modulo math operation
  i++;
}
k++;                                 //increment K index
}
if (a==1){                           // if tube is shifting left
if (k % (T*i) == 0){        
  u = U[i];                          // Set reference by modulo math operation
  i--;
}
k--;                                 //decrement K index
}                  
y = BOBShield.sensorRead();           // Read sensor 
BOBShield.actuatorWrite(u);           // Actuate
   
Serial.print(y);                     // Print output  
Serial.print(", ");
Serial.println(u);                   // Print input

if(i==60){                           // if tube was shifting right, switch direction of shifting to the left
  a=1;
}
if(i==0){                            // if tube was shifting left, switch direction of shifting to the right
  a=0;
}
}`