gergelytakacs/AutomationShield #126: TugShield: PID control example

gergelytakacs, 2019-05-08:
This will be an example that realizes PID control. You can use the HeatShield PID control example as a basis for your example. You must edit and think about
- The object name `TugShield` instead of `HeatShield` and of course the correct header
- Must set  a shorter sampling, probably on the order of 0.1 s
- Must set different trajectory according to your needs. The trajcetory (desired, setpoint) is in percents, your output should ideally reach these after a time.
- Must set a different length of an experimental section (T, in samples)
- Of course, and the hardest part is, to set a proper PID controller tuning.

gergelytakacs, 2019-05-14:
Why would you use different sampling from the identification case? Please use the same sampling period (e.g. you can change the ID to 100)... 
[image]

gergelytakacs, 2019-05-14:
Same as in sampling. This should be set by experiment to look nice. Also the comment is wrong.
[image]

gergelytakacs, 2019-05-14:
- I can't really judge whether it works if I don't see any results.
- Please show a Serial Plotter output so I can see that it works.
- If you (and I) are satisfied with this, then you have to log the data in Matlab, as you need to show the results in a Matlab plot anyways. (in your thesis)

gergelytakacs, 2019-05-22:
I see no change, improvement or results currently (22.5.2019, 8AM) -1

EvaVargova, 2019-05-22:
Here in the output graph. It doesn't work at all.
[image]
I think problem should be in two things: 
1. Problem with variable int - float for servo
2. Outputs from flexi are percentage values, but values written to servo aren't.

gergelytakacs, 2019-05-23:
I agree with the possible issues with the types, but first you must be sure that your tuning is right after blaming it to somehting else.

I believe the green line is your input to your motor. As you can see it aggressively and violently switches to extremes, which is not what you what. So I think, that your PID tuning is definitely incorrect. The PID controller reacts to the small changes in your input (kind of noise) and blows it out of proportion. You need to experiment (a lot) with the tuning.

closed 2022-05-19 by gergelytakacs