gergelytakacs/AutomationShield #128: BoBShield: Self-test example

gergelytakacs, 2019-05-08:
This example performs a sort of self-test and diagnostics of the hardware device. It can also help to check the functionality of the assembled hardware. What it actually does depends on the possibilities of the hardware. You may base this one on the self-test example found amongst the MagnetoShield examples.

For example:
- Turn the servo to one end and see what the sensor reading gives. Determine some sort of range that can be considered "normal"
- Do the same to the other end.
- Check if the sensor has some self-diagnostics function.
- etc.

gergelytakacs, 2019-05-12:
@lukasvadovic2 @corleone68 

Please push to your branch regularly, so I can check your progress. :) Thanks.

gergelytakacs, 2019-05-22:
Also, what's this?
[image]

gergelytakacs, 2019-05-22:
Mainly @TiborKonkoly  and @lukasvadovic2  at the time.

lukasvadovic2, 2019-06-06:
[image]
Self test looks good, tolerance of meassured distance however needs to be adjusted as soon as the real prototype became to exist :D because what I am using right now ... well lets just say that the ball is stucked all the times..

`#include <BOBShield.h>                // Includes all the neccesary libraries nad hardware components
#include <SamplingServo.h>            // Include special Sampling running on timmer 2

  void wait() {
  delay(5000);
}

Adafruit_VL6180X vl = Adafruit_VL6180X();
void setup() {

 Serial.begin(115200);               // Initialize serial
 BOBShield.begin();
  // wait for serial port to open on native usb devices
  while (!Serial) {
    delay(1);
  }
  // finds out if sensor is active
  Serial.println("Adafruit VL6180x test!");
  if (! vl.begin()) {
    Serial.println("Failed to find sensor");
    while (1);
  }
  Serial.println("Sensor found!");

  
  Serial.print("Turn servo to -30 degrees!  ");
  Serial.print("\t");                // print a tab character
  Serial.print("\t");                // print a tab character
  Serial.print("\t");                // print a tab character
  BOBShield.actuatorWrite(-30);         // Tilt beam to the minimum so the ball falls towards the sensor
  wait();                // Wait for things to settle
  int y = vl.readRange();
  if (y >= 80 && y <= 100) {
    Serial.println(" Ok.");
  }
  else {
    Serial.println(" Fail.");
  }

  Serial.print("Turn servo to +30 degrees!  ");
  Serial.print("\t");                // print a tab character
  Serial.print("\t");                // print a tab character
  Serial.print("\t");                // print a tab character
  BOBShield.actuatorWrite(30);         // Tilt beam to the minimum so the ball falls towards the sensor
  wait();                // Wait for things to settle
  int a = vl.readRange();
  if (a >= -5 && a <= 10) {
    Serial.println(" Ok.");
  }
  else {
    Serial.println(" Fail.");
  }`

closed 2022-05-19 by gergelytakacs