gergelytakacs/AutomationShield #129: TugShield: Self-test example

gergelytakacs, 2019-05-08:
This example performs a sort of self-test and diagnostics of the hardware device. It can also help to check the functionality of the assembled hardware. What it actually does depends on the possibilities of the hardware. You may base this one on the self-test example found amongst the MagnetoShield examples.

For example:
- Turn the servo to one end and see what the sensor reading gives. For this you must use the absolute values, not percents Determine some sort of range that can be considered "normal"
- Do the same to the other end.
- etc.

gergelytakacs, 2019-05-12:
@EvaVargova 

Please push to your branch regularly, so I can check your progress. :) Thanks.

gergelytakacs, 2019-05-14:
Do not forget about the self-test example.

gergelytakacs, 2019-05-22:
I see no change, improvement or results currently (22.5.2019, 8AM) :-1:

EvaVargova, 2019-05-22:
I have finished Self-test example. 
[image]
Maybe the range of "OK" is too wide, but in case it is narrower sometimes print "FAIL".

gergelytakacs, 2019-05-23:
Ok, looks about right. But why are you mixing languages?:)

closed 2022-05-19 by gergelytakacs