gergelytakacs/AutomationShield #132: LinkShield: Hardware Design

gergelytakacs, 2019-05-11:
This issue is devoted to the generic hardware design issues of the upcoming LinkShield device. Post hardware design related issues and ideas here, unless the problem is big enough to deserve its own  issue.

gergelytakacs, 2019-05-12:
Got these from eBay. They are 4 pin and look absolutely fine for hand soldering. Combine these with the strips I got earlier, and that's a pretty cool flexible lead to the end of the accelerometer. (Don't know if it analog or digital yet, but pretty sure you have to make your own breadboard - if possible).

[image]

gergelytakacs, 2019-10-31:
*Motor selection*
- Try 120 degrees, with non-default signal width (according to datasheet).
- Try the analog output. What is the maximal amplitude of the singal?
- Take a full swing of rotation with the beam, try to see if the analog feedback detects any sorts of vibrations.

*How many g-s*
- First tests with 30 deg idicate 4 g.
- Take a full swing (120 deg) to see.
- The standard SG90 TowerPro servo had power issues. Try if a capacitor and diode solve it.

*Mechanical design*
- Turn the link to the correct position.

*Accelerometer selection*
- Which one can be hand-soldered.
- Candidates for accelerometers
- use soldering kit - learn to solder

* Cables *
- Study flat ribbon cables and connectors

gergelytakacs, 2019-11-14:
- The feedback servo signal has voltage peaks on it, but it does not provide information about the vibrations. Conclusion: the feedback is not needed.
- Make screenshots with tests using a 90-120 degree sudden change of the beam, then test all the different accelerometers. Scale Y axis to "g"s or SI units. Use analog and digital too...
- Maximal acceleration is about 4 g.
- Analog seems to be better because of the limited resolution of digital.
- Try to determine which chips are hand solderable. Those with connections on the bottom we cannot do by hand. Try to look in forums if people mention hand soldering these.
- The FFC cables with 0.5 mm pitch seem to be the best. The connectors are hand-solderable.

gergelytakacs, 2019-11-21:
Notes

- Motor current is seen on accelerometer.
- Other than that analoge looks good.
- Range is up to 3 G
- Why is the DC response of the accelerometer not 0?
- Digital looks acceptable, a bit of quantization is visible.
- 4 G range is recommended.
- All cheap accelerometers suitable for our purposes have contacts from the bottom.
- If the voltage peaks cannot be removed, all we have is the digital accelerometer.
- Cables: 0.5 mm pitch, 15 cm length.

Todo 21.11.2019
- [ ] Consult the 3D design w/ Erik
- [ ] Try to stabilize the supply of the analog accelerometer to see if it filters out peaks. 
- [ ] If you've picked an accelerometer, who sells it?
- [ ] Try to read up on the voltage spikes on the accelerometer signal.
- [ ] Do some *more* tutorials in DipTrace.
- [ ] Determine the placing of the servo. Don't forget about Due, Zero. Use a CAD redner from grabcad (or elsewhere) and play around with the placement.
- [ ] Give me a list of accelerometer candidates. (It wouldn't be bad if we had the same on a breakout.)
- [ ] Go to TME.sk and pick a suitable 0.5 mm pitch, 15 cm length and suitable connectors too.
- [ ] Look for a same sized servo (micro?) and see if there are any "faster". Faster is better.
- [ ] Start to write a "Motivation / Lit. review chapter of your thesis.

martinvrican, 2019-11-21:
Accelerometer sensors used for testing: 
digital - ADXL345
analog - ADXL335

Results of tests with ADXL345 - 180, 60, 10 degrees
[image]
[image]
[image]

Results of tests with ADXL335 - 180, 60, 10 degrees

[image]
[image]
[image]

gergelytakacs, 2019-11-21:
@martinvrican Thank you, excellent.

Can you please do a repeat measurement with the digital sensor
- with no weights on the beam and
- with weights that you think would be "normal" and appropriate to use?

martinvrican, 2019-11-21:
TME.sk - pitch
https://www.tme.eu/sk/details/pca-6b-04-hu-3/konektory-ffc-fpc-raster-0-5mm/adam-tech/

https://www.tme.eu/sk/details/sfv4r-3stbe1hlf_c/konektory-ffc-fpc-raster-0-5mm/amphenol/sfv4r-3stbe1hlf/

Connector/ cable - I couldn't find good ribbon cables on TME.sk

gergelytakacs, 2019-11-21:
@martinvrican 
Allright, got it. Thanks.

gergelytakacs, 2019-11-28:
Consultation / todo 28.11.2019

- Peaks stayed unfortunately, that is why the decision is to go with digital.
- Will use the ADXL345.
- If we want to avoid outside supply, there are no faster servos. Faster servos usually use 6 V or hihger supply. We go SG92 or the SG90? Metal or plastic screw? 

- [ ] Decide on the servo.
- [ ] Determine the placing of the servo. Don't forget about Due, Zero. Use a CAD redner from grabcad (or elsewhere) and play around with the placement.
- [ ] Do some more tutorials in DipTrace.
- [ ] Consult the 3D design w/ Erik. Print your first candidate / idea by next week.
- [ ] Start to write a "Motivation / Lit. review chapter of your thesis.
- [ ] Draw the base PCB in an -almost- finished form.
- [ ] Look into the ways to program Arduino libraries. Consult the existing hardware, e.g. OptoShield, HeatShield, etc. You can also consult with FloatShield and MagnetoShield (quite complicated). Begin to write LinkShield API for Arduino.

gergelytakacs, 2019-11-28:
ADXL345 breakout tutorial: https://learn.adafruit.com/adxl345-digital-accelerometer/downloads

[image]

[image]

martinvrican, 2019-11-28:
- simplified ADXL345 breakout and connection
[image]
[image]
https://www.the-diy-life.com/how-to-connect-an-adxl345-3-axis-accelerometer-to-an-arduino/

gergelytakacs, 2019-11-28:
@martinvrican Sure! These seem to be not 5V tolerant that is the only thing. However, mind you that judging by the photos may not be enough, as these breakouts might have parts on their other side.  In any case, our guess was correct - there are 2 pullups and a capacitor seen on these. You still need to figure out if you can leave the CS/SDO pins on floating or it is required to connect them to ground. The Adafruit one looked like it need to be grounded to set the address. Please have a look at the datasheet.

gergelytakacs, 2019-11-28:
Oh okay. From the connections it seems that CS is on 3.3V, SDO is grounded. That is fine, you can do that on your own board, so it will only have a cap and the acceleometer on it. In any case, double check these connections in the datasheet so we don't make mistakes.

gergelytakacs, 2019-11-28:
Note that the connectors from TME are at my home, will bring them in Monday. Stop by, and you can decide which one to use and which one to design on these boards.

gergelytakacs, 2019-11-29:
@martinvrican New connectors arrived from TME, you can stop by monday:

[image]
[image]

gergelytakacs, 2019-12-02:
@erik1392 @martinvrican 

More connectors to choose from:

[image]

You can find it in your "drawer" in the lab, with your name on it.

martinvrican, 2019-12-03:
https://componentsearchengine.com/ga/search.html?searchString=52745-0497

martinvrican, 2019-12-03:
Servo
https://www.tme.eu/sk/details/kp-mb0090/servomotory/k-power/mb0090/

Connector
https://sk.farnell.com/molex/52745-0497/connector-ffc-fpc-4pos-1row-0/dp/2679573?st=52745-0497

gergelytakacs, 2019-12-05:
[image]

gergelytakacs, 2019-12-09:
@martinvrican @erik1392 The component list is as follows
100 nF 0805: https://www.tme.eu/sk/details/c0805c104m5rac/kondenzatory-mlcc-smd-0805/kemet/c0805c104m5ractu/10 uF, 1206, tantalum: https://www.tme.eu/en/details/t491a106m016at/smd-tantalum-capacitors/kemet/ Unfortunately I have not found 0805 at TME, so this is a bit larger.Servo: https://www.tme.eu/sk/details/kp-mb0090/servomotory/k-power/mb0090/22 uF, 0805, tantalum (instead of your too small 0.1 uF) for the Servo: https://www.tme.eu/en/details/tajr226m004rnj/smd-tantalum-capacitors/avx/ This is larger capacitance than your original intentions, I think it won't hurt at all.10 kOhm resistor 0805 https://www.tme.eu/en/details/smd0805-10k/0805-smd-resistors/royal-ohm/0805s8j0103t5e/  
Please re-design your boards accordingly, however, it seems to me that all the parts are compatible in their sizing to your original intentions, so no actual changes may be required.

@martinvrican  please upload your design files to the AutomationShield server so we can review and work on them. Please also put a screenshot of the current state here;) THank you:)

gergelytakacs, 2019-12-09:
I'm re-posting from @martinvrican 's mail:
[image]

Please @erik1392 have your say.

erik1392, 2019-12-10:
looks OK for me. Just one thing... It may be useful to extend soldering pads further outwards in the accelerometer footprint for ease of hand soldering.

gergelytakacs, 2019-12-10:
Good idea!

gergelytakacs, 2019-12-10:
[image] @erik1392 @martinvrican

gergelytakacs, 2019-12-10:
@martinvrican @erik1392 - This is the breakout:
[image]

gergelytakacs, 2019-12-10:
[image]

gergelytakacs, 2019-12-10:
@erik1392 @martinvrican 

[image]

gergelytakacs, 2019-12-10:
@martinvrican @erik1392 

[image]
[image]

Currently in audit.

[image]

gergelytakacs, 2019-12-10:
@erik1392 @martinvrican 

In production:
[image]

gergelytakacs, 2019-12-11:
Update: The components have arrived, I can bring them to the lab tomorrow.

gergelytakacs, 2019-12-11:
@erik1392 @martinvrican 

Santa has came early. All will be in your drawer in the lab by tomorrow. Note that some components are shared between Magneto and Link, you might need to check that too. (I don't want to split the contents, because it would be hard to identify some components without the original bag.)

[image]

gergelytakacs, 2019-12-18:
@martinvrican @erik1392 
Ho-ho-ho, Santa was here again. (I'll have it in the lab tomorrow.)

[image]
[image]
[image]

gergelytakacs, 2019-12-20:
@martinvrican @erik1392 

Ho, ho, ho - Santa has visited us in D.S. today!:)

[image]
[image]
[image]
[image]

gergelytakacs, 2020-01-02:
@martinvrican @erik1392 

Oh, santa was late with these ones: ADXL345 breakouts for "cannibalizing" for our own ones:
[image]
[image]

martinvrican, 2020-01-09:
It's cute, but will it work?
[image]

martinvrican, 2020-01-09:
Yes, it works!!!
[image]

gergelytakacs, 2020-01-10:
@martinvrican Yeah!!!!! Awesome!!!! You are the man!!!!!:)))) Nice job!

closed 2021-12-22 by gergelytakacs