gergelytakacs/AutomationShield #160: FloatShield: "Linear flow" insert

gergelytakacs, 2019-06-18:
I have left one device in the lab which has this "insert" to try to laminarize airflow. You can also ask Erik to print you more.

The base question is: Is this insert needed? Is it good and / or beneficial for the device? Essentially you need to find out whether it should be used at all.

PeterChmurciak, 2019-06-19:
Currently the ball sort of dances around the tube so it is not ideal. The insert might help a bit so I am going to try it. 

Also bigger ball might be the solution - the inner diameter of the tube is approximately 34mm and diameter of the ball is approximately 30mm, so I will try to find a slightly bigger one to see if it helps.

gergelytakacs, 2019-06-20:
Allright. As far as I remember the original results were not too great either. I was hoping the original authors posted some sort of measurements or control results online, but they haven't. So based on my memory, it wasn't too great.

I'd surely try to improve the behavior of the ball mechanically, at least to see what these things do and if they are even worth considering.  (Larger ball, different weight for the ball, "linearizer" etc.)

However, it is also worth trying to re-tune the controller. Have you tried that?

gergelytakacs, 2019-06-20:
You should post some screenshots of the serial plotter so I can see how unstable it is... Try to make a profile of setpoint changes, and see how well it is followed in closed-loop... Post some results online. But don't worry their results were not too pretty either, that's why you are working on it - to improve it a bit:)

gergelytakacs, 2019-06-25:
@PeterChmurciak You should try to pull toghether a PID controlled example without the insert and tune the controller as nice as you can. Then do the same with the insert. The only question is

**Is the insert needed? Does it make things better?**

PeterChmurciak, 2019-06-25:
@gergelytakacs I have yet to test the benefits of the insert, however sometimes it seems to make things worse - specifically, when the ball is still on the ground, the insert sometimes makes it harder for the ball to lift off the ground, see:

[image]
(sorry for possible neck fracture)

On the video I set the power output of the fan to max, and the ball does not seem to care for a while. This never happens without the insert. This behaviour is not consistent - it does not always happen and requires further testing. I agree with your idea to test benefits of insert, I am currently trying to make PID work to be able to do that, see https://github.com/gergelytakacs/AutomationShield/issues/161#issuecomment-505570938.

gergelytakacs, 2019-06-26:
Okay. So let's say the insert at the moment is inconclusive at best, or based on your analysis it may be even detrimental to control. It is interesting how it needs time to get flying. That is weird. What do you think the physical explanation could be? I have no clue.

One idea, not completely unrelated to the insert is, that nobody said you cannot have a "startup" procedure. By that I mean that before PID takes over (or anything else) you can do some sort of logical/heuristic algorithm to start the ball flying. Like, for example, push the ball to 100% and start the PID after then.... Or something like that.

My neck is fine. :)

gergelytakacs, 2019-07-17:
This has been solved: @PeterChmurciak has determined that the insert does more harm than good. It will not be considered from now on.

closed 2019-07-17 by gergelytakacs