gergelytakacs/AutomationShield #161: FloatShield: The Status Quo

gergelytakacs, 2019-06-18:
@PeterChmurciak After you have successfully tested the device, please report the current state of things here. I'm interested how things work (or don't) so we can begin throwing some ideas on how to begin.

PeterChmurciak, 2019-06-19:
@gergelytakacs Today I tested the examples provided for FloatShield, unfortunately neither of them seemed to work for me, both got somehow stuck on I2C laser sensor initialisation. To find out more about the issue, I have tested the parts of FloatShield separately - potentiometer and fan seemed to work just fine, the  problem was and currently is I2C laser sensor - I am unable to get past initialisation phase. I have tried to run examples provided by the Adafruit library for this sensor, but without any luck. I have also tried the I2C Scanner to see if the I2C device is recognizable, but this scanner also got stuck. When I unmounted the FloatShield from my Arduino Uno, the I2C Scanner worked correctly and did not find any devices. I am starting to think that the FloatShield that I took home with me might have a faulty connection to its I2C sensor. I will continue to ask google about solution to this problem and if I am not able to find one, I will come to school to test other FloatShield device to find out the true nature of the issue.

gergelytakacs, 2019-06-20:
@PeterChmurciak Allright, then there is some issue with the sensor. This could be either 
a) a faulty connection, so check cables, or
b) a faulty sensor unit. 
In either way the easiest idea would be to check another unit. You are welcome to stop by and take one from the lab. I'm not going to be at school tomorrow and Monday, however, you should find someone around with access to the lab.

The sensor is an Adafruit breakout board (clone) so that either works or not. The only issue could be physically with the cables.

However, based on your other comment, you got it working. So what was the issue afterall?

PeterChmurciak, 2019-06-20:
@gergelytakacs The other comment was misleading, sorry about that. It did not work, at least not the laser sensor. In the other comment I described the behaviour of the ball based purely on just manual control of the fan with potentometer. I did not manage to make the I2C sensor work, so I will come tommorow to school to try another one.

PeterChmurciak, 2019-06-21:
@gergelytakacs Today at school I have tried other devices, and they worked as they should. It seems I was just really unlucky - to pick the single faulty unit from the bunch of working ones. I have labeled the faulty FloatShield unit and left it in your office, and took two working units with me - one with the insert and one without it, so I can compare them. I will now continue with the testing and inform you about my progress.

gergelytakacs, 2019-06-23:
Awesome, thank you for the update! That's just bad luck, you really managed to pick the single broken unit from the stack and it seems our guys did not test them properly!:)

As soon as you get going, we can and should meet in a more regular basis. For now, just keep mapping the "status quo" !:)

gergelytakacs, 2019-06-25:
The rewritten library looks good. Let's test it in an indetification and closed loop (PID) example to see where we stand now.

PeterChmurciak, 2019-06-25:
@gergelytakacs  I am currently testing PID regulator example, the Serial Plotter output looks like this:

Blue colour - r (reference);
Red colour - y (ball position);
Green colour - u (PID output);

[image]

As you can see, it looks like tennis match between upper and lower part of the tube. By trying out the "sweet spot" idea  based on the ball behaviour mentioned in https://github.com/gergelytakacs/AutomationShield/issues/167#issuecomment-505529872,  I have managed to make it look much better - by basically remapping the output of PID to "sweet spot" range 40.0-45.0. With usage of this remapping it looks like this:

[image]

[image]

[image]

This looks almost good, but surely not perfect, the problem might be for example slow start from bottom position - the ball rising speed is much lower at 45.0 than at 100.0. I am not sure whether this is the right way to go. What do you think ?

gergelytakacs, 2019-06-26:
Okay. So I'm guessing the tennis match is the PID tuning as is currently (done by the guys before), and you are right that is quite useless and unstable. One of the reasons may be too aggressive tuning. This is what you've heard about (possibly) on TARII: the system is getting unstable. But before that you have transition, such oscillatory behavior. Look at how the input gets saturated... A good tuning should not go to the extremes with the input (e.g. fan power).

The second chart (and the rest) look really reasonable. However, there should be no "remapping" in the sense you are writing. You simply need to change the tuning!:) Now if remapping helps, then that means that a different PID tuning helps as well. Just think about what the "P" does! That's actually linear remapping:) So, I suggest you re-tune the PID completely. Without a model pretty much all you can do is play around with the constants until it works, or more diplomatically "tune by trial and error". 

Your last comment is correct and any behavior that is "odd" (e.g. different from the working point) is basically what we call nonlinearity!:)

TLDR; Retune the controller manually until you get the best possible results.

PeterChmurciak, 2019-06-26:
@gergelytakacs I am currently playing the guessing game with the PID example. So far I have been able to achieve something like this:

Blue colour - r (reference);
Red colour - y (ball position);
Green colour - u (PID output);

[image]

[image]

Visually, the device with ball looks pretty nice - it is able to hold the ball almost at the exact position, however it needs more time to stabilise - it is not able to adapt to rapid changes quickly enough. Also it has a pretty slow start from ground position. You were right about not needing the remapping part. I shall toy around with the constants a little more to see if it can be improved.

gergelytakacs, 2019-06-27:
[image]

Noice, indeed. This looks quite good to me. Don't forget that it is not entirely random guesswork for PID, and that is one reason why it is the beloved darling of industry. You should possibly read up on the intuitive meaning of P/I/D and its effect on control to have a better basis of understanding. I e.g. like the table included (of all the things) in the (Wikipedia article)[https://en.wikipedia.org/wiki/PID_controller]. See the table at the middle. A better tuning (than random) can be actually learned by experience.

Some other (semi-) random notes:
- Don't push the reference all the way to 100%. Logically, that's pretty easy to achieve. I wouldn't go there with `r`. (See second screenshot).
- After you are pretty satisfied (or sick) with your manual tuning, you can also do exact comparisons of quality other than visual. The easiest would be to create identical experiments (by an identical course of `r`), log the data to MATLAB and and do this
[image]
or this
[image]
where of course instead of an integral you have a sum and by the end you get a nice single scalar number telling you how good you performed. See more e.g. (here)[https://slideplayer.com/slide/5266544/]. This is not a must or something I expect, just to give you a tip how you can compare results more objectively. But don't underestimate "visual" comparison either.

PeterChmurciak, 2019-06-27:
@gergelytakacs I have been trying different sampling periods as you suggested in https://github.com/gergelytakacs/AutomationShield/pull/165#discussion_r298025831, and I have found quite big problem - the basic function provided by Adafruit for getting measurement from laser sensor is really robust/slow and as it is, does not allow use of faster sampling periods. The fastest sampling period is... 41 miliseconds - just to get data from laser sensor without two step functions overlapping. That means the results I got with using sampling period 30 miliseconds were flawed. These results I posted from PID https://github.com/gergelytakacs/AutomationShield/issues/161#issuecomment-506060632 used sampling 50 miliseconds so they are accurate. I am at loss now about what to do, because as it is, we are limited by this sensors library. What do you thing would be the best course of action now ?

gergelytakacs, 2019-06-27:
@PeterChmurciak

Oh well, that sucks. I would now fork my efforts to two directions.

1.  First of all, go on testing the system with 50 ms. Sounds a bit slow to me, but it might just work. In any case, tuning PID's and data gathering at this speed should be still very valuable. The worst that may happen is you re-do some things with a faster sampling later, but by that time you should have enough experience to do things promptly. 50 millisecond is:

2. Second of all, I would look into speeding up the readings. Try to find a completely different, alternate library, if such exists and try to implement it. Look at the documentation of the device (the datasheet of the sensor itself) and find out what is its limitation when it comes to speed. Dig up the internet. Possibly look for #DEFINE switches that affect speed. To test this, you don't even need the FloatShield, just do a minimal example with sampling + the sensor.

gergelytakacs, 2019-07-17:
The current status has been mapped:
- Arduino IDE API reworked
- Found a HW bug with the stickers on the fan
- Very decent PID results obtained

closed 2019-07-17 by gergelytakacs