gergelytakacs/AutomationShield #166: FloatShield: First principles model

gergelytakacs, 2019-06-25:
Although a first principles model of the FloatShield may be a bit tricky because of turbulent flow etc. we should try to look if this is possible. Please use this issue (also pictures, scans, links) to gather any info on the modelling of system dynamics.

gergelytakacs, 2019-06-25:
Have not read it, but it's a beginning: http://rabi.phys.virginia.edu/1050/2014/Slides/ball_sports-air.NQ.4.0.pdf

gergelytakacs, 2019-06-25:
Maybe the Drag equation can help?

https://en.wikipedia.org/wiki/Drag_(physics)

gergelytakacs, 2019-06-26:
@mgulan Requesting your ideas when you'll be back from SG.:)

gergelytakacs, 2019-07-08:
@PeterChmurciak Any progress here? @mgulan Care to join in?:)

PeterChmurciak, 2019-07-09:
@gergelytakacs Not much. I have looked at the drag equation (from wikipedia):

[image]

And I got stuck in figuring out how to get the flow velocity. Firstly I started to search for the fan used in the FloatShield. According to FloatShield wiki it is `Sunon PMD1204PQB1` and here is its [datasheet](https://www.gme.sk/data/attachments/dsh.625-170.1.pdf). The fan specifications that could lead us to speed of the air are (from datasheet):

[image]

Assuming the `Rated Speed` is speed of the fan while being at the maximum power output (PWM=255), we could  linearily approximate  the RPM to PWM dependence When PWM=0 then RPM=0 and when PWM=127 -> RPM~7500. From the RPM and fan blades dimensions, we could probably get the air velocity. This sounds somewhat logical, a bit simplified and it might or might not be the correct approach.

What confused me were the other two parameters. They are commonly used as fan specifications - `Air Delivery` and `Static Pressure` with their dependency chart (from datasheet):

[image]

Where as I understand, 
* Static Pressure - is the resistance of some object (ball) to the air flow - pressure with what ist the air flow able to "push" the object 
* CFM is air flow - how big a volume of air can the fan move per minute

After this I got confused what parameters to take into account in the model, and stopped looking into this issue for a while.

gergelytakacs, 2019-07-10:
@PeterChmurciak 

Okay, this may not be all that easy. So first let's look at what you wrote and how to create our custom first principles model (if it can be done at all).

So first, it is important to state what we can measure and control. We know the ball position relative to the top (or ground, doesn't matter) and the PWM signal to the fan. Now this can be related I guess to power/flow/pressure etc. fairly easily but possibly not accurately. Let's say we have "power" in and ball position out, between that we have a "black box" of dynamics.

One starting point is to do a force equilibrium, this is true for a falling object
[image]

Now from this I started to deduct something which might be completely wrong, please @PeterChmurciak and @mgulan try to see if my thinking is okay:
[image]
[image]
Try to check if my logic holds up to this point.

If it is okay to this point, it would be enough to relate airspeed to fan power:
- We can naively say that the relationship is linear(ish), so va=alpha*p where alpha is some unknown constant. 
- We can use tables and whatnot to try to relate the two, like you started to. 
- We could even try to measure it to see how it behaves then fit a curve over it. I have pressure gauges around, we might try to do an experiment like that. 
- All of the above:)

I any case first I would think if the other base assumptions are correct (on the scanned paper).

Given these, your thinking is good with the velocity, but not complete. As I see this data is given for "maximum power" at 12 V. Sure, we can interpolate linearly from this and figuring out how air speed relates to these factors at maximum speed. However, I suggest you look into this static presure / CFM thing in more detail, because it can be helpful actually. Or think about how the fan power can logically affect airspeed in the tube... I too have to think about and read about it a bit...

Now, a completely alternate route to get a model might be to see if someone has already done it:) Your source instead of YouTube videos and other similar things might be scientific articles. I am not sure if I told you this, but I will assume I have not. Our university pays big $$$ to access some of the leading databases like https://ieeexplore.ieee.org and https://www.sciencedirect.com/. Now I do not suggest to read everything related to airflow etc. but try to hunt down a summary of the model somewehere. Mind you that these databases are paid, but you have access to them through any univesity IP address. To get that from home, please activate your VPN account, which you have as a student. Here is a seriously outdated guide: https://www.stuba.sk/sk/pracoviska/centrum-vypoctovej-techniky/cinnosti-a-sluzby/vzdialeny-vpn-pristup-do-siete-stu.html?page_id=3750 but it should help.

PeterChmurciak, 2019-07-11:
@gergelytakacs Today I have tried the VPN account thing. No you did not tell me about it before and I had no idea. So I looked into it, namely the two sites that you mentioned. I found [one article](https://www.sciencedirect.com/science/article/pii/S2405896318303781) that might be useful, where author mentions model, and other works that have studied this problem. 

I will surely look into those other works, but for now the model mentioned in this article:

[image]

It looks kind of similiar to the one you proposed, but uses speeds instead of drag force. Possibly they are the same thing only in different form.

 I think your logic makes sense and seems correct. I am now thinking from where to get the air speed, I thought about calculating it from RPMs and fan blades dimensions - each rotation of the fan, the height of air equal to height of the blades gets pushed out, so it travels one height of the fan blades per rotation, this somehow made sense to me but I was not able to find anything supporting this idea. 

The other way the air speed could be calcualted is from CFM, as that is volume per minute, if we divide it by cross-sectional area of the tube  we should get speed out of it. However then we would need to know the CFM and power relationship. Datasheet provides only one graph for only 12V... so I am not sure about which way to go.

gergelytakacs, 2019-07-12:
This access to the world databases is actually an important advantage, and you can find lots of information otherwise not accessible on the open Internet. Anyways, the article you've found is a good one. To learn using sources such as this, you shall find the bibliographical references related to the model and have a look at those as well. These may have new pieces of information and so on and so on.

Now, it seems that I was quite close to the solution but forgot about the fact that the ball moves as well and that the lift force must use the difference of the airspeed and the ball's speed. Now, if you have a look at Jernigan et al. (2009) you see that the only difference is that they have this speed *difference*:
[image]

Jernigan et al. made further assumptions about the airspeed using a similar logic to what I wrote previously. They said that the airspeed must be a function of the voltage to the fan and the height of the ball. They measured some profiles and then used this in their model. 

As of the model you cited above, it is actually the same one as the one in Jernigan et al. with a little bit of manipulation of variables. They introduced the steady-state velocity (But at what power and hight? This is nonlinear!) and using that simplified the equation. So essentially based on Chacon et al. (2017) it reads like this:
[image]

I am not sure if this is any better than the previous representation though, also you have a new "constant" v_eq, which is not a constant at all, especially if you take a look at Jernigan et al.'s measurements. However, this may be a quick way to test a nonlinear model in System Identification Toolbox, see the last line for an approximation.

Now the remaining question is, shall we use this model (why not) we need to know the airspeed. So you can focus on this question. Figuring it out from the datasheet is quite a tricky endeavor, because everything is given for maximal speed! We can try to measure it... (But how?)

gergelytakacs, 2019-07-12:
Continuing a bit of reading Escano et al. (2006) includes another model that relates the airspeed and the velocity of the ball explicitly in a much more sophisticated way:
[image]

If you search for the pressure force, you can find McDonald (1994) that seems to be a great read, but I just looked at the pressure part. It actually derives the frequency of oscillations around the equilibrium point! I suggest you give it a good study.

References:
[McDonald1994.pdf](https://github.com/gergelytakacs/AutomationShield/files/3385651/McDonald1994.pdf)
[Uploading Escano2006.pdf…]()

gergelytakacs, 2019-08-06:
@erik1392

AirSpeed modeling review and TL;DR:

- [Jernigan2009] uses a combination of a 2D lookup table and a transfer function to model input (voltage) and airspeed. They have measured it and use tables, I would like to avoid look-up tables if possible, but it is a nice exercise. They have their model in Simulink only.
- [Chacon2017] uses essentially the same model, does some kung-fu and introduces the equilibrium speed (or a new constant) into the equation. Still needs velocity though. 
- [Chacon2018] linearizes the system from above, getting a first order TF+Integrator. Input voltage and wind-speed are taken as first order! So finally they get second order + integrator. Could not gather I/O data in open-loop, used a P regulator in closed loop to gather data. Best fit by ARX: na=nb=4, nk=3. They got 85%.
- [Escano2006] Uses a nicer model including pressure force, in addition to drag. That is all beautiful, but then they use linear models (black-box, 4th order cont.) taken at various inputs.
- [Cholodowiz2016] References Jernigan2009 and Escano2006, in addition Timmermans. After a lot of discussion on the Bernoulli equation (haha), they end up using a 1st order discrete TF with no zeros, and get a 74% match. (I guess to modeling data near working point.)
- [Saenz2017] Uses Jernigan2009 model. Notes that the drag coefficient depends on the Reynolds number, that depends on relative airspeed btw. object and airspeed. Notes that for small velocities this can be taken constant. Resorts to a similar representation to Chacon2018. Ends up linearizing the model, derives how. Could be an interesting first approximation from data.
- [Ovalle2019] Begins with DC motor dynamics, we however use a PWM signal and cannot really measure motor current etc. So such a detailed first principles analysis of the motor is not feasible. Ends up at a linear TF, of course. Assumes similar logic to Jernigan2009.

gergelytakacs, 2019-08-20:
@PeterChmurciak @mgulan @erik1392 

Bump on this one, please contribute!:)

gergelytakacs, 2019-08-26:
@PeterChmurciak @mgulan @erik1392

Updated the TL;DR on modeling in [here](https://github.com/gergelytakacs/AutomationShield/issues/166#issuecomment-518780005)

mgulan, 2019-08-26:
@gergelytakacs are the above papers in DB?

gergelytakacs, 2019-08-26:
@mgulan Yes, they are in DB, Reading subfolder.

gergelytakacs, 2019-08-27:
Okay so let's review the [Chacon2017] model once again, because that second order TF with an integrator might actually work (for the linearized case)
[image]
[image]

gergelytakacs, 2019-08-27:
For the linear SS it is
[image]

closed 2022-05-19 by gergelytakacs