gergelytakacs/AutomationShield #167: FloatShield: System identification

gergelytakacs, 2019-06-25:
This issue is related to creating a system identification examples, its results and possibly using the results to identify parameters in a first principles model.

PeterChmurciak, 2019-06-25:
@gergelytakacs I am currently trying to create identification examples - for example the one that changes input values multiple times and plots response, howerver I am not sure what values to use, because of the behaviour of this system. For example look at these plots from serial monitor (sampling period used was 30 miliseconds):

[image]

[image]

[image]

The behaviour that can be somewhat seen from these plots is that when the power output of the fan is from 0.0% up to 38.0% (percentual value of pwm signal 0-255) the ball does not lift up at all, around 40.0% up to 45.0% the ball sort of levitates, and anything higher makes the ball go up the tube all the way, with the only difference being the rising speed. I am now thinking about this "sweet spot" of levitation, that this range of mild descent of the ball around 40% and mild rising of the ball around 46% could be used for more precise control of ball position - for example the PID could work within this range. What do you think about the testing values used for this identification mehod ? Should I try some different method ?

PeterChmurciak, 2019-06-25:
For clarification, the red line is current ball position 0.0 (floor of tube) -100.0 (upper wall of tube), and blue line is current power level of fan 0.0 (pwm 0) - 100.0 (pwm 255).

gergelytakacs, 2019-06-26:
Yeah, these results are absolutely abysmal and it is most probably not your fault. :) Nice work though.

- First of all for testing it is fine to use percentage points for ball position (in the interest of scaling) but for it to be useful in system identification, the results should be in millimeters.
- If we even dream of using a linear (linearized) model, the output (position) cannot really saturate. This means, any result, where the ball hits the upper limit is not really useful. Mind you that we can just "cut" those parts out of the data, however, it is still better to try to achieve a data collection where the ball does not go all the way up. 
- Now I think out of the three results the bottom one is most useful, yet, by judging purely by eye, the input-output does not correlate very nicely. The things we learn from this may be that a.) there is not much "richness" you can inject in the levels injected into the actuator, this testing must remain around the sweet spot of ~45% power and then you introduce small changes. So my suggestion is you try to levitate the ball **without** control and then try to inject small changes into that signal to see what happens.
- However, this might be not possible at all. These systems are called open-loop unstable. Just think of magnetic levitation: you probably cannot stabilize it in open loop. That is fine and that is not your fault, but in that case the identification must be done under control, but then again that will cause a lot more trouble later.

[image]

TLDR; My suggestion is not to give up open loop ID at this point, but use a relatively small bandwidth of changes in the input. What you can do is to generate those changes more frequently (not every 100 th sample) and you can even choose a nominal power level that seems to stabilize the ball pretty well, but then try to inject a random (yes!) noise to the input. So the noise would be u = u_{nominal level}  + w where w is the maximum noise you can inject without losing stability.

gergelytakacs, 2019-06-26:
@mgulan Requesting your ideas when you'll be back from SG.:)

PeterChmurciak, 2019-06-27:
@gergelytakacs I am currently trying to get response of this system to step input, and this is what I got:

Blue colour - y (ball altitude in milimetres);
Red colour - u (fan power percentage - input to the system);

[image]

And more detailed view of the first minute:
[image]

Note that the ball altitude is in milimetres, and reaches at most around 65 milimetres - which would be around 20% of the tube height - the ball position is not saturated. However it is pretty slow process. I got this data by sampling every 50 miliseconds for 10000 samples, to be sure that it somewhat stabilises. The input needed was around 42% of fan power, higher inputs would send the ball flying to the top. I am now wondering if this kind of data can be really used, when it can be achieved only under specific conditions - like specific input. What do you think ?

gergelytakacs, 2019-06-28:
@PeterChmurciak 

Sure you can create an example with a step test, however, mind you that it has more of an educational value than usefulness for real system identification. What you can do from a step test is to calculate the gain and time constant(s) of a low order linear system. For a nonlinear system (such as this one, I suspect) this is of limited use. 

- The time constant: That the ball behaves differently when taking off from 0 mm than from changing its position around the working point, say 60 mm-ish. The charts would suggest that the time constant is 30-ish seconds (is that correct at all?) which is certainly not true. 
- The second thing is the gain of the system, which is the ratio of the input for a certain output. This would suggest to me that the gain K is around 1.2-ish. Here comes the catch: that should apply for every input and every input, hence, linear. But, according to your experience, it doesn't.
- The third thing is the "order" of the system. The chart suggests its of low order. This may be true(ish:), but it's definitely nonlinear. Order means the largest derivation in the model of the dynamic behavior.

TLDR; May actually hold an educational value to create a step test to explain the above things to someone, but it is not too useful for real identification thing.

PeterChmurciak, 2019-06-28:
@gergelytakacs Ah, I was afraid that it will be of no use, probably just wanted to try if it is possible to get stabilised step response without output saturation. Yes those numbers are correct, with this level of power input, it takes around 30 seconds to achieve stabilised position. When using more power, this phase is of course much quicker, but the ball position will certainly saturate and that is what I wanted to avoid. As you say the gain is surely not 1.2 for every input, it certainly does not behave linearily. This gain and this time constant would then be correct only for this specific power input of 42% and therefore are not really "constant". We can agree on the nonlinearity of this system. What kind of identification example would then help us with identification ? Step response is not good. Should I revisit Open loop using small changes as you mentioned in https://github.com/gergelytakacs/AutomationShield/issues/167#issuecomment-505742453 ?

gergelytakacs, 2019-06-28:
> This gain and this time constant would then be correct only for this specific power input of 42% and therefore are not really "constant".
Correct. However, this is a trick they often use with nonlinear systems. Do such identification with all sorts of inputs and create a set of models for different setpoints. Sometimes its called "gain scheduling" and other fancy names. But the base idea is you approximate a nonlinear system by a bunch of linear ones at different working conditions. (Regardless of what type of model do you end up using).

>What kind of identification example would then help us with identification ? 
I think you mean the test signal, not the model. If so, then yes its #167. Try to do open-loop for now around this nominal working point. Make small changes (also called perturbations). These changes can be even random noise injected. The idea in short is: Use the largest perturbation that a.) does not make the system unstable and b.) does not push the system into saturation (completely up or down). This would be a relatively good measurement to get. You can even get more of these at different levels, as e.g. the MATLAB System Identification Toolbox allows to combine several experiments in one batch.
-

gergelytakacs, 2019-08-06:
Attempting system identification in Open-loop by a PRBS signal. I think this will work:

[image]

gergelytakacs, 2019-08-20:
@PeterChmurciak / @erik1392 / @mgulan 

I'm sharing here the small progress I made before the whole "theses" writing issue came up, with the hope that it is useful. The script (and the connected functions) `matlab/FloatShield_PRBS.m` generates a PRBS sequence and saves it to a header `U.h`. The example `FloatShield_Identification.ino` attempts to have a go at open-loop identification. This reads in the PRBS sequence and changes upwards and downwards (bias) from a center of the input. Feel free to experiment and get the best out of the signal. I got something like this: [resultID.zip](https://github.com/gergelytakacs/AutomationShield/files/3520525/resultID.zip), given the input

[image]
the output
[image]

Now if this sequence is useful or not, I don't know. I'd attempt to do a black-bock identification with  System Identification Toolbox first to have an idea about the way the data/system behaves. See if something even converges at all to something. Anything around 70%+ is not bad, I think. Then I'd try to go back to first principles models in #166 that are based on literature and see if I can fit anything to a sensible to these. I'd start with linearized models, then try nonlinear. I'd review them over and over and try to figure a way to represent the air velocity which we don't have, since #184 was not too successful so far (got some forces, not speeds).

As I said before, this might be a though cookie though.

PeterChmurciak, 2019-08-20:
@gergelytakacs I have started looking into it. The sysid toolbox provides two types of non-linear models, `Nonlinear ARX` and `Hammerstein-Wiener`, we might possibly want to use one of them. 

As I currently see it, the most important part of identification is the kind of data you feed to the toolbox, as it tries to create model based on it. I have tried using data that you have acquired, but the best fit I could get was 20% with `Hammerstein-Wiener`. The input might be possibly too wild/jumpy, but I have not tried using only parts of that, so that could get better fit. What portions did you mean ? Divide it into parts where its mostly rising and mostly falling ?

I have tried it with different data (that included saturated output) but could not get better fit. I have yet to understand all the settings in the toolbox and models. 

The heatshield blackbox identification data, based on which was its model created were pretty good, as I was able to create nonlinear model with 90% fit using default settings. So I think I should focus on acquiring the kind of data the toolbox is able to work with.

gergelytakacs, 2019-08-26:
Hello there @PeterChmurciak ! Sorry for the slow response, as I said - a bit busy, but I got some breathing space for today.

The H-W models are used in [Shaefer2019] (see Dropbox for paper) and they actually provide a pretty good manual to reproduce results. However, they argue - and I think they are right - that a PRBS signal is not good for nonlinear systems, since it only handles binary input states. You will need a richer signal, they recommend something called "amplitude modulated" PRBS (APRBS). So one thing is that you could try to reproduce their methhod, it's actually pretty reasonable and cool. So this is important. I would not use PRBS for nonlinear models, only linear ones.

> The input might be possibly too wild/jumpy, 

The "jumpiness" of the input can be set. Look at the MATLAB script `FloatShield_PRBS.m` for that. The amplitudes are settable from within the Arduino application, but the bandwidth (speed) not. The script will generate a header file that you can include. I have been not experimenting with changing the speed of the signal, so don't know if it'd help. Sure worth trying.

> What portions did you mean ? Divide it into parts where its mostly rising and mostly falling ?

I mean that the larger span of the position you cover with the data, the more of the nonlinearity is obvious. So by portion, I mean a span of the amplitude within the tube. So, say, you pick a protion of the signal that stays within e.g. 40 mm of the middle, and does not venture too far from that. The identification for a "lienar" (linearized) model will be easier in that portion. Of course, it also means that we did not cover the entire amplitude. But that's kind of the point, let's start with something half-usable.

>  I think I should focus on acquiring the kind of data the toolbox is able to work with.

That too. But also focus on reading the existing papers on the modelling / literature and learning about identification. I'd also try to use the same (structure) models they claimed to use to see where you get with this. As I said, this identification story will be less straightforward than programming the API:)

Let me know where you got with this. Also try to bug @mgulan  and @erik1392 for ideas;)

gergelytakacs, 2019-08-26:
@PeterChmurciak 

Assuming the 2nd order model with integrator from [Chacon2017] with a 2nd order noise model, I can get about a 50% match, assuming one has a limited range of outputs.
[image]

gergelytakacs, 2019-08-26:
A fourth order black-box (fully parametrized) model also converges to ~ 40% match. Not bad, Imho:

[image]
[image]

This would suggest that a linearized SS model is also possible, e.g. based on [Castillo2017]

gergelytakacs, 2019-08-26:
@PeterChmurciak 

Here is a working process model identification. It gets unstable very-very fast. The difference between a useless model (diverging, unstable) and this one is adding a sample or two into the dataset or changing the intial guess by 5%... So it's also about luck if you can get it working:

[image]

Don't forget, that even if you pick a fairly small "linear region" of data and even if you assume a linear model - the identification problem mathematically is nonlinear. So it can diverge, etc.

Have a try with it: [ID.ZIP](https://github.com/gergelytakacs/AutomationShield/files/3540977/ID.ZIP) If you don't change anything, you should get the plot above, with 75% match. Of course this covers that region of the system, it is invalid elsewhere... The beauty of nonlinearity. If you'd use this model elsewhere, it would not be very useful.

gergelytakacs, 2019-08-27:
@PeterChmurciak @mgulan @erik1392 

So a grey-box ID of a linear TF is done, and it's actually not bad. Its committed to the repo as `FloatShield_Identification_Linear_TF.m`. 

[image]

This is done, but there are still a bunch of possibilities:
- Linearized state-space model
- A better representation of fan dynamics
- Natively nonlinear identification
- etc.

gergelytakacs, 2019-08-27:
Linear state-space:
[image]
see `FloatShield_Identification_Linear_SS.m`.

PeterChmurciak, 2019-08-27:
@gergelytakacs Sorry for my slow response... Those are some nice fit percentages. I see that you spend your "breathing space" very productively. Good job sensei! 

I forgot about those articles... will revisit them soon for more information on non-linear identification. Also will look closer into the ID data you provided.

The thing that my mind can not yet properly process, is that model found using these small time portions would somehow help us with the whole thing, they seem to be too different...

gergelytakacs, 2019-08-28:
@PeterChmurciak 

> Sorry for my slow response... 

No worries.

> Good job sensei!

Arigato gozaimasta!:)

>  The thing that my mind can not yet properly process, is that model found using these small time portions would somehow help us with the whole thing, they seem to be too different...

You are right, the models from those small portions will not help identify the entire model. The key here is that we chose the small time portions because they did not diverge way too much from a certain height. If the output stays near a certain linearization point (e.g. height) then we can create a model for that area. Say, the model is valid for the height of 100+-20 millimeters. For a linear system, this model would be valid for the entire dynamics, but not here - it is more nonlinear than that. Also imagine a curved function, you can cover that with small segment straight lines and approximate it. This is what we do now, but now it's time to move to natively nonlinear models. So in TL;DR:
1. These are linear model so it's expected they only cover a certain setpoint (certain area of dynamics).
2. Real machines often work near pre-set setpoints so linear models can still be useful for nonlinear plants.
3. Getting to know the linear models can help in identifying the nonlinear dynamics later. It's sort of a practice.

PeterChmurciak, 2019-08-29:
@gergelytakacs When reading a bit of Schaefer2019, I have started working on the APRBS signal generation (bf0a6b5), as they used it there, and it might be the right way to go for non-linear model acquisition.

I have then modified Identification example (3ada9a7) for it to be able use both APRBS and PRBS signals based on what the user wants. 

Additionaly, the example now uses PID control at the start, to stabilise the ball at selected altitude. The power used at stabilised position is then saved and further used as a "middle" around which using PRBS/APRBS small changes (+/- span) to power are made.
So now main inputs to the example is height where the ball should stabilise, and the span - how big changes will be made. Thanks to this, we should be able to choose around what height the ball will move.

PeterChmurciak, 2019-08-31:
@gergelytakacs The APRBS signal they mention in Schaefer2019 has additional parameter `k` - number of levels the signal can have. I am not sure if that is important. Our signal can have as many levels as there are PRBS jumps - each jump in APRBS has different value, within the range of [minu,maxu].

I have managed to get some possibly usable data: [IdentificationData 31.8.2019.zip](https://github.com/gergelytakacs/AutomationShield/files/3562515/IdentificationData.31.8.2019.zip)

[image]

All were acquired at the lower portion of the tube. When trying to move around upper half of the tube, saturation was inevitable. It is possible that our tube is too short - for example we can not see how high the ball would go at 70%/80%/90% power, because it is limited by the "wall". Maybe if the upper wall was higher, we could dare to use more wild inputs without fear of immediate saturation, and therefore cover wider portions of the tube. On the other hand if we changed the lenghth of the tube, we would possibly change the bahaviour of the system. Also the ToF sensor might have problems with bigger distances... Not sure... What do you think about the idea of using longer tube for identification purposes ?

When trying to use the data in sysid toolbox I did not manage any reasonably good fit. It is possible that APRBS with different parameters (maybe different band) would be better suited for higher portions. I shall try that.

gergelytakacs, 2019-09-03:
@PeterChmurciak 

In reaction to [this one](https://github.com/gergelytakacs/AutomationShield/issues/167#issuecomment-526329014): All right, looks good to me. Checked the files visually, the concept is also quite good!

>Additionaly, the example now uses PID control at the start, to stabilise the ball at selected altitude.

But PID is off the rest of the (recorded) flight path, I guess?

> The APRBS signal they mention in Schaefer2019 has additional parameter k - number of levels the signal can have. I am not sure if that is important.

I do not think that is relevant.

> All were acquired at the lower portion of the tube. When trying to move around upper half of the tube, saturation was inevitable. It is possible that our tube is too short 

The system is (borderline) unstable. It is inevitable to hit the upper or lower part of the tube eventually. I think the question here is to be able to get the maximum span out of the identification procedure without using a feedback control method. If the nonlinear model fits the lower portion, the model should also take care of the upper portion. The key question here is to get **data that is good enough for identification** and not perfect data.

> What do you think about the idea of using longer tube for identification purposes ?

I do not think that is absolutely necessary. It even may change the dynamics of the system. I'd say, let's try to play around with the data and nonlinear identification as-is and keep this as a possible option if all else fails.

> When trying to use the data in sysid toolbox I did not manage any reasonably good fit.

Have you tried the grey-box model according to `HeatShield_Identification_Greybox.m`? If we get a decent linear fit, this should mean that we get a decent nonlinear fit as well. So, I think it is possible.

PeterChmurciak, 2019-09-08:
@gergelytakacs 

>But PID is off the rest of the (recorded) flight path, I guess?

Yes, the PID stabilisation part is not recorded. It happens during setup. Do you think it should be ?

I have started doing MATLAB tutorials for system identification toolbox as you proposed, to get better understanding of its functions, for now I am at linear models. The tutorials provide some nice training data, it is hard to not get good model fits with them...

mgulan, 2019-09-09:
@PeterChmurciak What was the exact ball size that you used to gather the APRBS iddata set btw?

mgulan, 2019-09-10:
It was a struggle, but I managed to identify a nonlinear greybox model (see below) with a ~79% match with the live data from system. I think it might be further improved as I still don't know the exact mass and exposed area of the ball (@PeterChmurciak pls) that was used; more reliable initial guesses of fan parameters would be useful too.
Based on an extensive literature review I converged to the system model proposed recently by Chaos et al which is most reliable. This shall be used for the paper and for the linear(ized) model.
[image]

gergelytakacs, 2019-09-10:
@PeterChmurciak in response to [this](https://github.com/gergelytakacs/AutomationShield/issues/167#issuecomment-529253504)
 
> Yes, the PID stabilisation part is not recorded. It happens during setup. Do you think it should be ?

No, we don't need that. I was just making sure I understood what you meant.

> The tutorials provide some nice training data, it is hard to not get good model fits with them...

Yeah, well that sort of data is a bit neater than ours!:)

gergelytakacs, 2019-09-10:
@mgulan @PeterChmurciak in reaction to [this](https://github.com/gergelytakacs/AutomationShield/issues/167#issuecomment-529769033)

> It was a struggle, but I managed to identify a nonlinear greybox model (see below) with a ~79% match with the live data from system. 

Fantastic! Excellent news!

> I think it might be further improved as I still don't know the exact mass and exposed area of the ball (@PeterChmurciak pls) that was used; more reliable initial guesses of fan parameters would be useful too.

Assuming the cork ball is used (@PeterChmurciak pls. confirm), the diameter is 30 mm (measured it just now) and the mass is 3.84 g (measured it just now.

>Based on an extensive literature review I converged to the system model proposed recently by Chaos et al which is most reliable. This shall be used for the paper and for the linear(ized) model.

Can you please try the linear(ized) model according to this then? They examples are in `AutomationShield/matlab/examples/FloatShield/` as "Linear_SS" and "Linear_TF", see if the model is even better than what you get with the existing identificaiton...

gergelytakacs, 2019-09-10:
PS.: @mgulan 

Instead of overwriting `AutomationShield/matlab/examples/FloatShield/ ` as "Linear_SS" and "Linear_TF" with the Chaos et al. model, you could put there a switch, so user can choose which way to go... Or even use both and compare them. I'm just saying that since there are already usable results it would be a shame to throw them out completely.

mgulan, 2019-09-25:
Hey there, forgot to post the results last week. Something like this
[image]
yet I was not satisfied and think the code (`FloatShield_Ident_Greybox.m`) needs to be revised, e.g. unit of output data, PWM style input (not present in the actual file so that one can see the same result as above).
I also put the basic stuff that I assumed into the FS paper TeX file (Identification part) for inspection.