gergelytakacs/AutomationShield #168: FloatShield: Sensor (VL53L0X) sampling speed issue

gergelytakacs, 2019-06-27:
According to #161  there is an issue with the sampling speed of the VL53L0X TOF sensor. It should be thoroughly investigated if this can be mended somehow, or if not, why. So, here are some questions to answer:
- Is it a hardware issue? What does the datasheet say about sampling speed?
- Is it a software issue? Can settings be changed in the API? Are there alternate libraries?
- Has anyone experienced or investigated this problem?
- Is it a computational power issue? Would more powerful hardware achieve faster sampling?

gergelytakacs, 2019-06-27:
The datasheet is here: [vl53l0x.pdf](https://github.com/gergelytakacs/AutomationShield/files/3336211/vl53l0x.pdf)

gergelytakacs, 2019-06-27:
Now this should get you started @PeterChmurciak : 
[image]

PeterChmurciak, 2019-06-27:
@gergelytakacs With a different library (specifically https://github.com/pololu/vl53l0x-arduino), I was able to reduce it to the time of 20 miliseconds, this seems to be the limit. Therefore we can lower it, by using different library, as you suggested. Can you elaborate on your suspicion that there is not much to gain ?

gergelytakacs, 2019-06-28:
@PeterChmurciak 

The "not much to gain" note was just me thinking aloud that reducing it to half (not like an order of magnitude lower to 1 ms) may not be worth the trouble at all. But if a simple library swap works already, then by all means!:) 

So now I suggest you try to test if the Polulu library is sufficient. If yes, we can change the library alltogether. That is actually a bit more complicated than basic GitHub activity, so let me know if that's the route you want to take. I think it is a good one.

Outside libraries are placed in `/src/lib/` so maybe change the current Float API to the Polulu one, and assume it is in the previous path. I will link the Polulu library there as soon as you tell me I should. (It is not a simple copy, it is something called a Git Submodule).

PeterChmurciak, 2019-06-28:
@gergelytakacs I am trying out the library and it seems sufficient. The single function that will probably be used from the library is that for distance in milimetres acquisition, and that one seems to be working fine - even with 20 ms sampling. I think that I want to take this route and use Pololu library instead of Adafruit.

gergelytakacs, 2019-06-28:
@gergelytakacs 

Sure thing! I'll integrate it when I get back on the 8th. Remind me please.

PeterChmurciak, 2019-06-29:
@gergelytakacs I have looked into Github Submodules and concluded that while it may be easy to add a submodule into repository, according to many people it may cause future problems if you dont know what you are doing. I shall wait for your return with swapping the library, to avoid breaking anything. 
Enjoy your vacation :slightly_smiling_face:

gergelytakacs, 2019-07-03:
@PeterChmurciak I think if you've got to this point in reading, you know what I know about submodules :-)

I've added the Polulu one, it's in `AutomationShield\src\lib\vl53l0x-arduino\'. Let me know if it's okay (at least as a submodule) and I change it if not. I guess I can remove the Adafruit one?

PeterChmurciak, 2019-07-03:
@gergelytakacs I have encountered some trouble while trying to make this new submodule work. Specifically in file `AutomationShield\src\lib\vl53l0x-arduino\VL53L0X.cpp` there is a line at the beginning:
> #include <VL53L0X.h>

It gives me fatal error:
>D:\NewDokumenty\Arduino\libraries\AutomationShield\src\lib\vl53l0x-arduino\VL53L0X.cpp:6:21: fatal error: VL53L0X.h: No such file or directory
compilation terminated.

when trying to compile examples. The library is included in FloatShield like this:
> #if __has_include("lib/vl53l0x-arduino/VL53L0X.h")        
#include "lib/vl53l0x-arduino/VL53L0X.h"                   
#endif

However when that line at the beginning of the `AutomationShield\src\lib\vl53l0x-arduino\VL53L0X.cpp` file is changed to:
> #include "VL53L0X.h"

There are no problems and everyting seems to work as it should.

My question then is, can the <...> be changed to "..." inside the submodule ? I tried to change it and look at desktop github to see if it will register the change, however it did not look that simple.

gergelytakacs, 2019-07-04:
Oh fun!:) You've found a bit of bug in Polulu's library:) 

If the Polulu library is installed through Arduino IDE, it works fine, but as you've found it, it doesn't when you use it as a submodule in our library. The difference between `<>` and the quote mark is that while the former is searched in the install directory, the second in the compilation directory.

- Now, what you can do, is you **Fork** their library (make your version), correct this mistake then start a pull request (at their repo) explaining why you made the change.
- If they approve, then case solved, we use their version.
- If not, we use your fork!:)

PeterChmurciak, 2019-07-04:
@gergelytakacs As you suggested I have forked their library and made my [corrected version](https://github.com/PeterChmurciak/vl53l0x-arduino). I have also started a pull request at their repository, explaining my situation. For some weird reason `continuous-integration/travis-ci/pr` check has failed when checking my [pull request](https://github.com/pololu/vl53l0x-arduino/pull/36#issue-294654294).

gergelytakacs, 2019-07-05:
@PeterChmurciak - Okay, excellent. Let's wait out if they pull or not. If not, we'll just use your fork!:)

Our Travis fails or their travis?

gergelytakacs, 2019-07-05:
@PeterChmurciak Oh, now I see. If you inspect their logs, you can see that the fail has nothing to do with your change or the library, there is an issue w/ their config.

Click on "View Log" to see the details...

gergelytakacs, 2019-07-08:
@PeterChmurciak I've checked the pull requests for the Polulu library, however, there are 3-4 outstending PR's at the moment. The oldest one is from 2017, they seem not to be caring much about the library anymore.

So what I've done is I removed all the unneeded sumbodules from our repo (including the Adafruit one for this device) and added a submodule from your fork! Let me know if you run into any problems. 

TLDR; You have full control over Polulu version of the library as a submodule now.

PeterChmurciak, 2019-07-09:
@gergelytakacs The files in [/lib](https://github.com/gergelytakacs/AutomationShield/tree/FloatShield/src/lib) look like they are not from my fork, but from original library repository - they are not modified despite my fork being their [source](https://github.com/gergelytakacs/AutomationShield/commit/3fd270fcbabaa869973ce661cfe0af2fee1696b0#commitcomment-34236448). I am not sure how to update them to match those in their source. Please have a look at it, it might have a simple solution that I am just not aware of.

gergelytakacs, 2019-07-10:
@PeterChmurciak This has been solved, thanks to you!:)

gergelytakacs, 2019-07-17:
This issue has been solved:
- The ToF sensor has so called time budgets, it seems, that the shortest possible sampling speed is around 20 ms anyways.
- The Polulu library allows to achieve this easily, so the AS library adapted this.

closed 2019-07-17 by gergelytakacs