gergelytakacs/AutomationShield #170: FloatShield: Possible hardware changes

PeterChmurciak, 2019-06-29:
Purpose of this issue is to discuss hardware problems of FloatShield and possible changes or enhancements. The hardware should try to adhere to AutomationShield guidelines https://github.com/gergelytakacs/AutomationShield/issues/19.

PeterChmurciak, 2019-06-29:
One of @gergelytakacs issues with the current design is that it's not 3.3V board compatible, as stated in https://github.com/gergelytakacs/AutomationShield/issues/169#issue-461985184.

PeterChmurciak, 2019-06-29:
One of my issues is the ball/tube size. The crucial attributes of the ball are its weight and its diameter with regard to the tube diameter. Therefore ideal ball would be made of some light material - cork, cotton, polystyrene or it would be hollow plastic ball - like the ones used in table tennis. And at the same time, it would be of only slightly smaller diameter than the inner tube diameter - that there would be almost no room for horizontal movements of the ball. These balls are made in specific normalised sizes, most often 30 and 40 sometimes 35 milimetres. Currently used balls are 30mm cork ones. Therefore ideal tube inner diameter should be 31 or 36 or 41 based on the ball used, in our case 31/32mm. Unfortunately, the current tube has inner diameter of almost 35mm, "almost" being very important word here - recently I got my hands on cotton balls of 35mm diameter because of my conviction that they would fit, well... they didnt. I also got another plastic ball used for table football which had special size of 34 milimetres, and that one was a perfect fit. The problem then was that it was too heavy for the fan. What I want to say, is that there was not enough thought put into choice of tube/ball size combination, and it is surprisingly harder to find balls of specific diameter than it is to find tube of specific diameter - the tube should have been chosen based on available balls. How much big of an improvement would this change of tube be, I am not sure, I am just pointing it out, that it could improve the ball behaviour. However that would be quite an intrusive change because the tube is the biggest part of the FloatShield body and all the 3D parts were created based on its size.

gergelytakacs, 2019-07-03:
It is not a problem to change the tube if you wish to do that. I'd suggest you try to get your hands on alternative tubes (e-bay, any other source) - I can refund the price. Re-printing the top/bottom is also a possibility, you just have to contact Erik M. to print out the pieces.

Thinking about it according to the above comments, I recommend the following course of action:

- I'd consider using the FloatShield (let's call it R1) in its current state and get the most out of the given hardware. The reason for this is, that we have 10 pcs. of it and we should get it to a state that is usable (useful) for the students in the future without re-doing it a lot. Small and not intrusive changes are fine (different bal, "linearizer" etc.) , but focus on software at the moment.
- For hardware changes, In th meantime, I'd completely redesign the thing, let's call this FloatShield R2. This will be "your" bettered version, and we do  You may even want to think about making it considerably smaller if necessary.  Nobody said it has to be this big, e.g. So if you wish to get your hands dirty, we should start a parallel effort to get a second generation FloatShield designed. 

This would be ideal, as HW design takes a lot of trial and error, lots of waiting on orders and while that is done you can work with the current one. (Later there can be switches in the SW to adjust for the versions).

gergelytakacs, 2019-07-08:
@PeterChmurciak Shall I look for balls too? (That sounds funny when I type it like that!:)

gergelytakacs, 2019-07-08:
@PeterChmurciak [This was](https://www.ebay.com/itm/34mm-Weight-Lockable-Stainless-Steel-Penis-Stretcher-Metal-Ball-Enhancer-Care-US-/112651789516) the first result of my search.

gergelytakacs, 2019-07-08:
Jesus, I'm going to get ads for cock rings now for the next month.

PeterChmurciak, 2019-07-09:
@gergelytakacs You can try looking, I have looked mostly on the Slovak/Czech stores. Ball with diameter 33 or 34mm would be ideal, however it also needs to be made of some light material. The internet is surely full of balls, unfortunately most of them are the kind you would not like to get your hands on.

gergelytakacs, 2019-07-17:
Moving some info from #44 that will be closed:

- sensor and vent connectors too close together (possible solution - smaller vent connector size fits, for future PCBs might be redesigned as well)
- short-circuited resistor (due to contacts touching usb port, has to be moved on the PCB in future design, for now, duct tape will do)
- Fan connector too close.
- 3.3 V compatibility
- Description of the sensor is missing (silk?)
- Smaller potentiometer (#19) a bit farther away from the fan for easier accesbility.

gergelytakacs, 2019-08-20:
@PeterChmurciak 

This is more of a long term question: after the identification is done, (see other threads) you / we have to decide how and where to continue. For this we may have to get together to have a short chat. You'll have to see whether you are interested getting your hands dirty by a bit of hardware design (e.g. to create a new revision of Float 2), or just to do a basic upgrade with 3.3 V compatibility, or just to focus on software altogether. etc.

gergelytakacs, 2019-08-26:
@PeterChmurciak 

Any luck with learning DIP Trace?

PeterChmurciak, 2019-08-27:
@gergelytakacs Somewhat yes. For now I have made following changes to the PCB layout:
* To make it 3.3V compatible I have changed the voltage source of the `Potentiometer` and the `VL53L0X` sensor from 5V to 3.3V.
* To solve the short-circuit I have repositioned the resistors (R2, R3).

Original layout:
[image]

New layout:
[image]

If I understand it correctly, we also want to use a smaller potentiometer, specifically `CA9MV-10K` as mentioned in [hardware guidelines](https://github.com/gergelytakacs/AutomationShield/issues/19#issuecomment-381040064), is that right ? Do we have those ? That could save up some space on the board and allow for more repositioning of resistors (possibly the R2 and R3 resistors might be too close right now).

Few questions based on https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-512206822, shall we also use smaller vent connector size fits ? What is the Fan too close to ? And how did you mean sensor description ? Currently the sensor connection is labeled as `J1` should it be any different ?

gergelytakacs, 2019-08-28:
@PeterChmurciak 

Great, I see you are getting started!;) I'll try to answer your notes as efficiently as possible:

- Yes, please use the smaller potentiometer, CA9MV-10K. I'll attach a design that uses it, so you can just copy and reuse. We have them physically available in the lab too.
[Magneto.zip](https://github.com/gergelytakacs/AutomationShield/files/3549221/Magneto.zip)
- Now that the pot is powered by 3.3 V, is compatible with CMOS level boards, but not with TTL level ones (AVR). Solution: you have to connect the 3.3 V source to the AREF pin - then in software tell the AVR boards that the analog reference is external. So basically the software will have to have various parts for board versions (e.g. `IFDEF BOARDRELEASE = R2` ).
- The MOSFET should be marked with something else than "C", it is not a capacitor. E.g. you can use "Q" that's usual for transistors and similar components.
- "Description of the sensor is missing (silk?)" - I have absolutely no idea what I wanted back then, lol:) Just ignore it. Maybe I wanted to put there `y(k)` near the sensor connector, as it was in Opto... Bu since neither the actuator nor the sensor is on the board itself, it does not make sense to label it like that. So, don't worry about this.
- "Sensor and vent connectors" [...] I don't remember the exact reasons why I had a problem with that, maybe it was just too close together for my taste. Inspect that area and decide if you see an issue or not. Maybe just move them a bit further away (in the y-direction)... Again, don't worry about 
that too much.
- Don't forget that the schematic and PCB views should remain connected and you can synchronize them. I hope you are not working only in the PCB view to make changes. (So that we will have the scheme as well).
- Perform any other change you see fit.

PeterChmurciak, 2019-08-28:
@gergelytakacs Specific parts in the PCB schemes are taken from `AutomationShield.eli` Diptrace library. Is there a way for me to get access to that library ? I could not find it anywhere in the files.

PeterChmurciak, 2019-08-28:
@gergelytakacs I believe that Uno schematic block, that has identical layout as the real board itself should be used in this case, like this one:

[image]

With GND and A4,A5,SDA,SCL ports somehow differentiated.

Reason for this is that currently, there is only one GND port, while in reality there are three (on UNO) GND ports. In different situations we might want to use different ports, like in https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-525464330 - the `C1` MOSFET is connected to the second GND port - because it is closer. Same thing with `J1` that uses SDA/SCL ports instead of A4/A5, because the path is more straightforward. Currently in the scheme is used block:

[image]
 
When the layout as in https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-525464330 is used, and the layout is synchronised with the scheme, the paths to the second GND and to SDA/SCL ports are removed, and want to forcibly use the default GND and A4/A5:

[image]

This is caused by the other ports not being defined (not connected with blue line) in block-pattern connection, because there is currently nothing to connect them to:

[image]

It could be fixed by creating Uno block that has identical pins as the hardware, so each virtual pin (in scheme) could be connected to its real counterpart (in the board pattern). This solution, while making the scheme block slightly bigger would allow more flexibility in the routing department, as the pins that are closer or in more favorable position to the component could be used, and then synchronising the layout with scheme or using the autorouting tool should work as intended.

I will probably need the `AutomationShield.eli` library as there is proper board pattern and components that I currently have only copies of. I only hope that it still exists. I can then implement the new Uno block into it. What do you think about that ?

PeterChmurciak, 2019-08-30:
@gergelytakacs 

Original scheme:

[image]

New scheme:

[image]

I am not sure about their correctness but these two schemes are equivalent. The new one also uses new arduino block, new-smaller potentiometer and connection 3.3V-AREF. There are also lesser amount of "cables" shown on the new scheme. What do you think about it ? Should I change it somehow (for example show more cables) ?

PeterChmurciak, 2019-08-30:
@gergelytakacs I have saved the blocks that were used in files that you and the FloatShield wiki provided me, and modified them to suit my needs.

New layout (top):

[image]

New layout (bottom):

[image]

This includes new potentiometer, bigger gaps between connectors and 3.3V-AREF connection. (On the back mask - FloatShield v1.1 R1 09/19)

 I would like to ask you about this connection (blue line on the second picture), whether it can be realised like this - wider trace (from 3.3V) that goes to potentiometer 100% pin, and out of there thinner trace goes to AREF pin. Is that correct way of doing it ? 

Please review both scheme and layout and tell me if you would like to see something done differently.

gergelytakacs, 2019-09-03:
Hello there @PeterChmurciak 

I will try to be as efficient as possible with my reply:
- Regarding [this comment](https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-525758385): I have shared my Dropbox containing the entire AutomationShield with you. There is a "common" parts library for the AutomationShield project. The original idea was, that each new part we create will be saved in a custom library for future use.

> {...] With GND and A4,A5,SDA,SCL ports somehow differentiated.

You are correct. The original UNO part was fine, but now it grew over us and we need GND's and SDA/SCL pins differentiated. We ran into that problem before, we chose to leave the original (simple) UNO part in the schematic view, then changed routing manually in the board view.

> Same thing with J1 that uses SDA/SCL ports instead of A4/A5, because the path is more straightforward. 

That is not the only and most important reason. The SDA/SCL ports are compatible across different boards (e.g. Mega, Zero etc.) but the Uno is the only one that has SDA/SCL on the A4/A5. So I'd rather remove the SDA/SCL notes from A4/A5, because that would make our designs Uno specific.

> It could be fixed by creating Uno block that has identical pins as the hardware

I think that would be the best course of action. And I would say you should call it "Uno R3 Pinout" so as to not to refer the board itself but the pinout. And use only the default SDA/SCL connectors not A4/A5. After reading the rest of your comments, I think you already did change it around. In that case it would be good if you'd save it in the common library for future designs.

>New scheme:

- It looks good to me, but I'd prefer the more cables version and no net ports. I tell you why. For working with the scheme the less cables and net ports are better. But we will have to show your new design eventually (e.g. your thesis, possibly further academic paper) and for that I think the cabled version is preferrable.
- The other think I'd ask is you change the schematic symbol of the MOSFET because it's really disproportionately huge compared to the rest of the components. This is just an aesthetics issue.
- Why are the D7, D8 connections on the sensor jumper active? Are they used in any way? I don't actually think so! Can you please investigate this?

> I would like to ask you about this connection (blue line on the second picture), whether it can be realised like this - wider trace (from 3.3V) that goes to potentiometer 100% pin, and out of there thinner trace goes to AREF pin. Is that correct way of doing it ?

The wider traces are used at places, where there are more current to be expected. That is not the case with the potentiometer, so you can change it back to regular trace width. You should inspect ground and voltage traces to and from the fan/MOSFET for trace width, since these carrry a bit more juice.

>  (On the back mask - FloatShield v1.1 R1 09/19)

Please call it FloatShield R2. You will need to alter code between versions, and it will be easier to refer to it as "no. 2" when using `#ifdef`-like precompiler definitions:)

PeterChmurciak, 2019-09-04:
@gergelytakacs 

>Why are the D7, D8 connections on the sensor jumper active? Are they used in any way? I don't actually think so! Can you please investigate this?

This is interesting question, I did not think much about them, I just assumed that they should be there because they were there before. I have looked into it.

Our sensor looks like this:

[image]

* `GPIO1` on sensor is connected to `D8` on board
* `XSHUT` on sensor  is connected to `D7` on board

As I understand it (based on [this](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/0e/0a/96/1b/82/19/4f/c2/DM00280486/files/DM00280486.pdf/jcr:content/translations/en.DM00280486.pdf)), they are required when there are more VL53L0X sensors being used. This [site](http://osoyoo.com/2019/04/21/arduino-lesson-vl53l0x-time-of-flight-distance-sensor/) provides short description about their additional usage. 

In our case, there is only one sensor of this kind being used, and that will probably not change. Therefore usage of those two pins is obsolete. Actually, despite the pathways created on the `R1 v1.0` board, there are only 4 cables being used to connect the sensor with the jumper `J1` - the `D7` and `D8` pathways are not used.

To make sure I have tested the sensor using cables directly connected to arduino to confirm that only the 4 first pins are required - `VCC`, `GND`, `SCL`, `SDA`. That is indeed the case.

You were right, the `D7` and `D8` connections are not used, and in our case will not be used. I therefore believe that we can safely remove those pathways from the scheme&layout. What do you think?

gergelytakacs, 2019-09-05:
@PeterChmurciak 

Please remove these connections, they won't be used or necessary.

PeterChmurciak, 2019-09-08:
@gergelytakacs I guess that matter concerning 3D printing should be discussed with @erik1392 ? Do you know about a good way of contacting him ? I have tried contacting him using university email but it is possible that he does not use it as often during summer as during school year.

I would like to change the 3D sensor holder at the top of the tube, as it can move around pretty easily  (at least on my unit), and I dont see a duct tape as long term solution. Does Erik (or do you) possibly have the CAD model of current sensor holder ? It might need just minor tweaks.

erik1392, 2019-09-09:
@PeterChmurciak sorry I did not respond to your email on Friday, I will print it today. 3D model is downloadable form wiki [here](https://github.com/gergelytakacs/AutomationShield/files/1942312/assembly.zip), there is the whole assembly in .step and also some .iam (inventor assembly) which is probably useless without separate files for all the parts. The .step file should work just fine.

gergelytakacs, 2019-10-23:
If a fan with a hall sensor is used
- What is the amplitude of the signal of the fan, and how to connect electrically to the Arduino?
- What TC will be used on an Uno to count? Timer2? Which pins can be used as counters with Timer2?
- Will be a different (possibly 3 pin connector)
- The lower sensor connector could be 4 pin!
- The MOSFET schematic sign is extra ugly. Please correct.

Will need a components list to buy stuff!!!!

PeterChmurciak, 2019-10-23:
@erik1392 could you please provide me with the model name of the new fan you propose to use in R2 ?

PeterChmurciak, 2019-10-23:
@gergelytakacs Timer0 is reserved for Uno functions, [Sampling](https://github.com/gergelytakacs/AutomationShield/blob/master/src/Sampling.h) library on Uno uses Timer1 so we are left with Timer2. The Timer2 is also responsible for controlling PWM output signals generated by pins 3 (which we use for PWM) and 11. I am not entirely sure whether that could be a problem.

gergelytakacs, 2019-10-24:
@PeterChmurciak I'm pretty sure that is an issue though. So we are stuck with Timer 2 in this case that also uses PWM. 

So TL;DR: @erik1392 It is a great idea to have a fan with RPM feedback but which counter would we use (on the Uno) to count the signal? Or would this be reserved to other boards than the Uno?

We need to decide if the Fan+RPM is actually doable or not...

PeterChmurciak, 2019-10-24:
The Timers responsible for PWM pins based on [this article](https://www.robotshop.com/community/forum/t/arduino-101-timers-and-interrupts/13072) are:
Pins 5 and 6: controlled by timer0
Pins 9 and 10: controlled by timer1
Pins 11 and 3: controlled by timer2

So when we use one of those PWM pins, the timer can no longer be used for something else. If we used Timer1 for sampling, Timer2 for counting RPM, we could then use pin 5 or 6 for PWM fan control. Just using PWM created by Timer0 should not compromise any of the Timer0 delay()/millis()/etc. functions. What do you think, is that correct ?

PeterChmurciak, 2019-10-24:
Cited from [this article](https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM):
>The Arduino uses Timer 0 internally for the millis() and delay() functions, so be warned that changing the frequency of this timer will cause those functions to be erroneous. Using the PWM outputs is safe if you don't change the frequency, though.

gergelytakacs, 2019-10-24:
@PeterChmurciak Nice find, and you are actually right. I've completely forgot about this. They've combined the timing functions with PWM capability! So basically theoretically Timer 0 should be capable of producing both PWM and timing for delay() etc. In this case any sort of PWM should be handled by either pins 5 or 6 (on any future shield).

However mind you that 5 and 6 uses a faster PWM frequency because of this (980 Hz vs 490 Hz). It might change absolutely nothing, or it may make a difference. Can't say right now. Can you test this? E.g. to patch the current FloatShield to work from pin 5 (or 6)? I think using a piece of cable should be fine. Just to be sure.

If you can confirm that nothing changes, then we can use Timer 2 for inputs from the fan. Then the next step would be to find out how the fan signal works at all.

PeterChmurciak, 2019-10-24:
@gergelytakacs I have tested at what power level the ball starts to lift from the ground - when using PWM pin 3 the ball starts to lift around 46% of power and when using PWM pin 5 the ball starts to lift around 52% of power - using the PWM from pin 5 moves up the required power level of the fan to lift the ball off the ground by approximately 5%.

To make sure I tried the PID example:

[image]

Results are very similliar, in this case slightly better with PWM from pin 5 but that might be a fluke.
When taking closer look at the power levels:

[image]

Indeed there is approximately 5% power level difference to keep the ball at stable altitude. Not sure how big of a problem this is. What do you think ?

gergelytakacs, 2019-10-25:
@PeterChmurciak The results look good, there is no need to worry about anything you wrote. I think we shall safely move the PWM output to  no. 5/6 pins. 

You may make this change right in the scheme / PCB for R2.

Now please have a look at the way the fan gives the feedback signal. Ask Erik for the type, you can try to contact him through Facebook if he is unresponsive.

gergelytakacs, 2019-10-25:
@PeterChmurciak PS: I think he said it's completely the same + the signal, so you may be able to find it on your own.

erik1392, 2019-10-25:
@PeterChmurciak @gergelytakacs   Sorry for the late response. This is the [fan](https://www.tme.eu/sk/details/pmd1204pbb1af/ventilatory-dc-12v/sunon/pmd1204pbb1-a-2-f-gn/) it is exactly the same one but with one extra wire. I also found some basic [example](http://www.theorycircuit.com/reading-dc-fan-rpm-arduino/), how to read the RPM but the software side won't be sufficient enough for us. The good news is that the hall sensor circuit inside the fan is just connecting the pin to the GND through some transistor so it can work with any voltage.

erik1392, 2019-10-25:
I have some 3pin fans at home so I can verify ti next week.

PeterChmurciak, 2019-10-25:
@gergelytakacs First part of [this article](https://showmeyourcode.org/how-to-measure-the-rpm-of-the-3-pin-pc-fan-with-arduino/) combined with following graph taken from [fan specification](https://www.tme.eu/Document/23780aed3853b140e993af4ec53d537c/PMD1204PBB1AF.pdf):

[image]

Leads me to belive that this is how the feedback signal looks, where each "step" represents one quarter of a rotation. However I was not able to find out the signal amplitude (on the top of this picture there is +12.0V but I am sceptical about that). In every example of this kind that I have found, the signal cable was connected directly to interrupt pin, using resistor only as a pull up (like in example provided by @erik1392). To make sure we will probably need to test it.

PeterChmurciak, 2019-10-25:
@gergelytakacs MOSFET schematic old vs new:

[image]

What do you think ?

gergelytakacs, 2019-10-25:
@erik1392 So you put any voltage onto the signal pin and it is connected to ground from time to time? That is actually smart. So please do verify it in experiment if you have such a fan available.

@PeterChmurciak Testing it will be the easiest way. If I understand Erik correctly you are supplying a signal to this pin and measuring it's state. The fan circuitry pulls it to gnd.

PS: Yes, the mosfet looks good!;)

erik1392, 2019-10-29:
@gergelytakacs @PeterChmurciak I have som interesting foundings. It would be very difficult to use a fan with a hall sensor and control the fan with PWM because the power for fan and sensor is cut off and it is impossible to read rpm correctly,
[image]
 I tried even connecting the cap in parallel to the fan.
[image]
(PWM at 80%)
 It helped a bit but it drastically changed PWM vs. fan characteristic ( increased the effective value of current) which is not neceserly the bad thing but at low PWM values, ti still made quite an ugly signal at the hall sensor output. 
[image]
(PWM at 15%)
So I propose that we use a fan with 4 wire connections, where the fan is connected to power +12V, GND, hall sensor output and PWM input and the RPM controller is inside the fan. We also won´t need the Mosfet on the shield. 

I found this [one](https://www.tme.eu/sk/details/pf40281b1-s99/ventilatory-dc-12v/sunon/pf40281b1-000u-s99/), the same size except it is 4mm toller and should be a bit more powerful in terms of airflow and pressure but it should draw less current.

I also tested this type of fan and rpm measurement works correctly. 
[image]
(hall sensor output with 4wire fan)

[image]
(rpm and PWM sent to the fan)

gergelytakacs, 2019-10-30:
@erik1392 @PeterChmurciak 

Erik, thank you very much for your extensive research into the fan. If I understand it correctly, the power to the hall circuit is cut off with each "0" of the PWM signal? So that means, that it should be controlled by an analog signal?

To me the 4 wire solution sounds okay. I am not bothered by the lack of MOSFET actually. We should try it! Peter - can you maybe order one of these fans for testing? I will order anything more expensive you need, but the thing is it is way too complicated do the paperwork for orders (not worth it if they are small).

erik1392, 2019-10-30:
@gergelytakacs yes because we are interrupting connection to GND with the MOSFET, it may work if we use the high side MOSFET but I haven't tested it. It looks like a PC motherboard uses some kind of linear voltage regulators to control fan rpm. 

@PeterChmurciak just thing I noticed, I would advise connecting all GND pins on the shield together and run the GND trace for Fan connector from that two pins at Power header. Right now the fan GND is connected only to GND pin at digital I/O header and I am not sure how much current can traces connecting that pin withstand.

gergelytakacs, 2019-10-30:
@erik1392 Please try to test it with a high-side MOSFET too, just out of curiosity.

@PeterChmurciak I agree w/ Erik on connecting the ground connection. Also, please try to order a single fan for testing, if you can!;)

PeterChmurciak, 2019-10-30:
@gergelytakacs Alright, I have ordered the [fan](https://www.tme.eu/sk/details/pf40281b1-s99/ventilatory-dc-12v/sunon/pf40281b1-000u-s99/) that @erik1392 has proposed to use. It will be delivered probably during next week so when it does, I will try it out.

@erik1392 interesting idea with the GND pins, I shall do that.

PeterChmurciak, 2019-10-31:
@gergelytakacs The latest scheme:

[image]

And corresponding layout:

[image]

With `J1` being the I2C sensor jumper and `J2` being the 4 wire fan jumper. Last pin of the fan header will be connected to the pin responsible for counting RPMs. I am currently not sure which pin can be used for that with Timer2.

erik1392, 2019-11-02:
@PeterChmurciak nice work but the fan header has wrongpin order. Pin 3 is hall sensor and 4 is PWM input.

gergelytakacs, 2019-11-05:
@PeterChmurciak 

Haven't you selected a fan that has both PWM input and signal output? Because in that case you will need 4 wires into the connector. So I agree with @erik1392 that something seems to be amiss. Also, has the fan arrived and have you tested it?

PeterChmurciak, 2019-11-05:
@erik1392 @gergelytakacs Indeed I have selected the 4 wire fan. I did not know the order of the pins at the time of creating the layout so I just randomly connected the 3rd pin to PWM and left the 4th pin unconnected. Other reason of not connecting the 4th pin was that I was/and still am not sure what pin should it be connected to - using which pin will be the counting realised. Can you please advise me with that ?

The fan has arrived yesterday but I did not test it yet. Will do that tomorrow.

PeterChmurciak, 2019-11-07:
@gergelytakacs I have began testing of the new fan. It has 4 wires (from left to right) black, blue, yellow, red. Their function is represented by their colour:
>  BLACK - GND
>  BLUE - PWM signal
>  YELLOW - signal from Hall sensor
>  RED - VCC (+12V)

The cables are thinner than the ones on the previous fan and are separate - do not stick together, so connecting them into the board in the same order as they come out of the fan is not necessarry. I guess I will still move the GND connection into the last pin of the `J2` header to keep their order.

The new fan is more powerful but also lot louder than the previous one (at the 200 PWM sounds like a jet taking off), thankfully our working levels when balancing the ball are not that high. It also has a faster reaction time for PWM change - because of that the current PID example does not work with it - the constants for R2 will have to be different.

The Hall sensor (YELLOW) wire works as we expected - the INPUT pin the wire is connected to on the board needs to be pulled up using external or arduino-internal pull-up resistor (to 3.3V/5V), and through the YELLOW wire the INPUT pin state is being pulled down to zero. What do you think should we use external resistor or built-in resistor for this ?

Currently working on getting the RPM.

PeterChmurciak, 2019-11-10:
@gergelytakacs I figured out that a good way of getting RPM might be counting the pulses in a set period of time - for example in one second. At max speed of the fan there are 547 pulses in one second - that would represent 16410 RPM what looks like a valid value based on [fan specification](https://www.tme.eu/Document/04ecbbafd5c3a23dfb7428f2eeb8581e/PF40281B1-S99-DTE.pdf) stated speed 17600 RPM +/- 10%. 

I am now thinking about how wide the period of time should be - how often should the RPM get updated ? For example half a second works as well and quarter might also work, but is that necessary ? At the slowest speed of the fan there are around 100 pulses in one second - that means 1 pulse in 100 milisecond while we need at least 2 pulses (representing 1 rotation) - so the fastest update of RPM could be every ~200 miliseconds using this method of counting. 

The problem could be solved the other way around - to measure the time between pulses - how much time does it take for the fan to make 1 rotation (or N rotations). I can try to make it work this way if we need faster updates. What do you think ?

gergelytakacs, 2019-11-11:
@PeterChmurciak @erik1392 This is in response to your last three posts in this thread:

> I was/and still am not sure what pin should it be connected to - using which pin will be the counting realised.

Should be the pins left for the only active timer-counter, that is Timer2 - so it should be 11 or 3. You may pick, e.g. Pin 3. I am unsure how the location of timer/counter pins is preserved in other architectures though. I don't think you/we should fixate way too much on this RPM feedback signal though.

> The new fan is more powerful but also lot louder than the previous one 

Not necessarily worried about this personally.

>  because of that the current PID example does not work with it

That is expected, as you have a slightly different system model. You can handle this in code by different tuning switched by the `SHIELDRELEASE` token in the preprocessor.

>  What do you think should we use external resistor or built-in resistor for this ?
 
I would suggest external. This, at least, makes the board a bit more populated:)))

> I figured out that a good way of getting RPM might be counting the pulses in a set period of time - for example in one second.
 
I am not really sure what you mean by this. I'll take a guess: you read in the pulses into a counter, then poll that number at pre-set time periods, re-set it afterwards and go at it again? Because if yes, this sounds quite reasonable.

> At max speed of the fan there are 547 pulses in one second 

You forgot to tell: have you figured out a proper way counting?:) Or is this just an initial test, by e.g. polling a pin?

> I am now thinking about how wide the period of time should be - how often should the RPM get updated ?

Well, as we are varying the input power each sample, I guess the most ideal case would be to update it along with the sampling... This actually has the same logic as input/outputs etc. in the control loop.

>  so the fastest update of RPM could be every ~200 miliseconds using this method of counting.

I'd say while for measuring fan speed for most applications that is fine, for ours is not. So a fan speed update of ~200 milliseconds would only mean it can be used for auxiliary (utility) tasks and self-calibration, but the dynamic charts or any live control would be pretty useless. I'm not saying in this case we should not use it, I'm just saying it has limited usefulness.

> The problem could be solved the other way around - to measure the time between pulses - how much time does it take for the fan to make 1 rotation (or N rotations). 

If that can be done easily with a MCU timer (I think it definitely can) then I'd say that is a lot more reasonable way to do it. Look up how you can measure the width (duration) of a pulse, and you have your solution.

@erik1392 care to join in?

PeterChmurciak, 2019-11-11:
@gergelytakacs 

> I would suggest external. (resistor)

According to fan specification, value of the resistor should meet the condition `Ra >= V+ / If(max)`. Where `If(max)` is equal to 5 mA (milliamperes), and `V+` based on the interpretation of the [specification](https://www.tme.eu/Document/04ecbbafd5c3a23dfb7428f2eeb8581e/PF40281B1-S99-DTE.pdf) (V+ is used for different things there), can be either supply voltage 12V or working voltage 3.3/5V. The working voltage makes more sense, as this resistor is meant to pull it down, therefore the resistor should have >= 660 ohms for our 3.3V compatible board.

> I am not really sure what you mean by this. I'll take a guess: you read in the pulses into a counter, then poll that number at pre-set time periods, re-set it afterwards and go at it again? Because if yes, this sounds quite reasonable.

Currently, I use Timer2 as a source of 1second long interrupt intervals, meanwhile I use external interrupt pin to count the amount of pulses. Each second from Timer2 I calculate the RPM from the amount of pulses and reset the pulse counter. This I believe is proper way of counting. 

>If that can be done easily with a MCU timer (I think it definitely can) then I'd say that is a lot more reasonable way to do it. Look up how you can measure the width (duration) of a pulse, and you have your solution.

To get the RPM, we need the number of pulses and the time it took. At the very least, we need the duration (width) of single pulse. Beginning and end of the pulse can be catched nicely with interrupt, but what about the time ? I do not think using for example millis() is a good way, therefore I am using timer for timing right now. Thinking now how that can be combined to get the time of the pulse.

Now that I look at it closely...
>At the slowest speed of the fan there are around 100 pulses in one second - that means 1 pulse in 100 milisecond while we need at least 2 pulses (representing 1 rotation) - so the fastest update of RPM could be every ~200 miliseconds using this method of counting.

This is not true...  I have interpreted the decimal point incorrectly... 100 pulses in one second means 1 pulse in 10 milliseconds while we need at least 2 pulses (representing 1 rotation) - so the fastest update of RPM could be every ~20 miliseconds using this method of counting. 

Anyway better way might be to get the width of the pulses. I will try to do that.

PeterChmurciak, 2019-11-11:
@gergelytakacs Current solution: I am using Timer2 to create something like "TenthOfMillis()" function that increments every tenth of millisecond. This resolution is suitable for our problem as one impulse at maximum power lasts around 0.8 milliseconds - 8 "tenthsofmilliseconds". I then use this time to get time between two RISING signals, using external interrupt pin - to get time of half of a rotation and then count RPM from this (time between any CHANGE could be measured to get the time of quarter of a rotation also, bit I think half is more than enough). (Also the counter of the tenthofmillis is resetted periodically so there is no slow-down caused by calculating with big numbers or problem of overflow)

This method works very well as it is, only thing that I am unsure of is if everything will work as it should when put together.

gergelytakacs, 2019-11-12:
@PeterChmurciak @erik1392 @mgulan 

I was part of a PhD defense, where the author developed a remote-experiment framework. His test case actually was similar to our FloatShield. There are some notes regarding his work that may be pertinent to our own efforts:

- Would it be reasonable to place the sensor directly above the fan (to the bottom) the same way it is shown in this work? See, p. 39, Fig. 12 b.
- The author said, that there is a library for reading signals from fans, it is maybe worth to look at this... This is actually the `FanSpeed` library, see. p. 53. @PeterChmurciak might take a look at this.
- He mentions that VL53L1X in comparison with VL53L0X has some advanced functions. One is, that you may specify and change the field of view (FOV) so as to minimize interference from the tube. Would it be worth exploring this possibility and changing the sensor to the newer version?
- He has a literature review in Sect. 4.1, which includes some papers that we might have missed with the IFAC paper. It might be interesting to ask him, what sort of fan did he pick.

[dizertacka_Rabek_fin (2).pdf](https://github.com/gergelytakacs/AutomationShield/files/3836615/dizertacka_Rabek_fin.2.pdf)

gergelytakacs, 2019-11-12:
@PeterChmurciak @erik1392 

>[...] therefore the resistor should have >= 660 ohms for our 3.3V compatible board.

Sounds reasonable, but don't forget to test it. Also, don't forget there is only a 620 or 680 Ohm resistor. I don't think it matter though. However, if this is truly a pull-up resistor, its value seems to be extremely low - so we talk about a "strong" resistor. I think you can safely assume that 1k should be fine to use. After reviewing your table, yes V+ seems to be whatever you want in the range of 3-12, and based on the logic I'd guess 1 k is fine, but maybe even 5 k - 10 k are good (it says larger than or equal:)

> Currently, I use Timer2 as a source of 1second long interrupt intervals, meanwhile I use external interrupt pin to count the amount of pulses. Each second from Timer2 I calculate the RPM from the amount of pulses and reset the pulse counter. This I believe is proper way of counting.

Sounds good but a bit of a slow update time. Nevertheless, still useful. The other way would be to measure the duration of a pulse. You have a counter which is stopped, re-set each time the signal goes to the low state. I think this should be pretty reasonably done with a counter. 

> To get the RPM, we need the number of pulses and the time it took. At the very least, we need the duration (width) of single pulse. Beginning and end of the pulse can be catched nicely with interrupt, but what about the time ? 

As said above. Try to measure the duration of a single pulse or two pulses. How about you use a timer that increments as any other timer. But you start/stop it based on an external hardware interrupt! Say an interrupt detects a rising edge, your timer goes 1,2,3 ... i ... N-1 N but then a falling edge (or better another rising edge) is detected and you write the actual value `i` to a variable, re-set the timer to zero. Something like this.  

>  I do not think using for example millis() is a good way,
Well, it might not be, dunno yet.

>  so the fastest update of RPM could be every ~20 miliseconds using this method of counting.

That does not sound that bad as originally. But in that case, can't you just increment a counter in an interrupt each time a rising edge is detected and compute the RPM at each sample? (Since you are sampling at 20 ms anyways)?

> Anyway better way might be to get the width of the pulses.

I think it definitely deserves a try.

>  Current solution: I am using Timer2 to create something like "TenthOfMillis()" function that increments every tenth of millisecond. This resolution is suitable for our problem as one impulse at maximum power lasts around 0.8 milliseconds - 8 "tenthsofmilliseconds". 

That sounds awfully low resolution. So you like have 8 levels of RPM information? Does not really sound to be enough...

> This method works very well as it is, only thing that I am unsure of is if everything will work as it should when put together.

I don't think its appropriate. You are limiting the timing resolution to 3 bits!:) Why would you use your timer at each 10th millisecond if you can configure that timer practically any way you want?

Also, check out this one: https://github.com/anru/fanspeed

PeterChmurciak, 2019-11-12:
@gergelytakacs 
>Would it be reasonable to place the sensor directly above the fan (to the bottom) the same way it is shown in this work? See, p. 39, Fig. 12 b.

I do not think that this change would make any reasonable difference. Additionaly the sensor would need to be fixed there in a position as to not obstruct the air flow, while also being somehow protected from the falling ball in case of sudden fan turn-off as I believe the sensor would not like hits like that.

>The author said, that there is a library for reading signals from fans, it is maybe worth to look at this... This is actually the FanSpeed library, see. p. 53. @PeterChmurciak might take a look at this.

Will look at it.

>He mentions that VL53L1X in comparison with VL53L0X has some advanced functions. One is, that you may specify and change the field of view (FOV) so as to minimize interference from the tube. Would it be worth exploring this possibility and changing the sensor to the newer version?

Cited from [pololu forum](https://forum.pololu.com/t/vl53l0xv2-vs-vl53l0x-vs-vl53l1x/15072):
>In general, increased range is the main difference between the L1X and L0X, although there some other differences like the new ability to select the ROI (region of interest) on the L1X...

The main difference is increased range - we do not need that, but added function is ability to select region of interest (or FOV) which might potentionally interrest us in terms of reducing interference as you said. The questions are though, how big the actual interferrence is and what does it cause ? Is the reading more noisy because of it ? Is this interferrence a relevant issue ? I can get one and try it out if you believe it would make a noticeable change, although I am bit skeptical about it.

PeterChmurciak, 2019-11-12:
@gergelytakacs 
> Sounds reasonable, but don't forget to test it. Also, don't forget there is only a 620 or 680 Ohm resistor. I don't think it matter though. However, if this is truly a pull-up resistor, its value [...]

I currently have been testing it with arduino internal resistor which based on [this page](https://hackingmajenkoblog.wordpress.com/2016/08/12/measuring-arduino-internal-pull-up-resistors/) has value around 35kOhm (or in range 20kOhm - 50kOhm). With this it works as it should, therefore I agree on using 10kOhm resistor for this. 

>That sounds awfully low resolution. So you like have 8 levels of RPM information? Does not really sound to be enough...

>I don't think its appropriate. You are limiting the timing resolution to 3 bits!:) Why would you use your timer at each 10th millisecond if you can configure that timer practically any way you want?

I am not sure if we understood each other correctly. The pulse width varies with fan power - at 10% power there are ~230 tenthsofmilliseconds in one pulse (one pulse lasts ~23 milliseconds), at 50% power there are ~18 tenthsofmilliseconds in one pulse (one pulse lasts ~1.8 milliseconds), and at 100% power there are ~9 tenthsofmilliseconds in one pulse (one pulse lasts ~0.9 miliseconds).

I actually measure time for half of rotation - time between one rising part of signal and next rising part of signal - basically period of the signal. This time is ~double of those I have previously mentioned - so at 100% power I get value of ~18 tenthsofmiliseconds, and I compute maximal RPM out of that. 

If you think that this is not accurate enough, I can go lower - to hundrethofmillisecond - where at 10%  power I would get ~2300 and at 100% power I would get ~180, or to microseconds where at 10% power I would get ~23000 and at 100% power I would get ~1800. Which resolution would you then prefer ?

>Also, check out this one: https://github.com/anru/fanspeed and maybe ask Mr. Rabek from the PhD work above what library he used. (Might actually not be a lot better than your concept, just checking).

I have tried the library. Based on what I have seen, the library depends on polling of the state of digital pin the hall signal goes to somehow measure its width - it does not utilize interrupts to do that. (If there were no interrupts used, we could use arduino function [pulseIn()](https://www.arduino.cc/reference/en/language/functions/advanced-io/pulsein/) that uses similliar principle to get pulse duration) With any value of delay itroduced into the sketch, the function returns invalid values. I believe this library can not be effectively used in our case, as there is a lot of other stuff going on besides monitoring RPM. I am not sure how  Mr. Rabek used it and whether this is really the library he used, as on page 59 of his work he uses `fan.getSpeed();` function while the [FanSpeed.h](https://github.com/anru/fanspeed/blob/master/FanSpeed.h) library does not contain such function.

PeterChmurciak, 2019-11-13:
@gergelytakacs I have asked @erik about the purpose of protection diode in our circuit, as I am not that knowledgeable in the workings of electrical components, and we came to a conclusion that the diode might not be needed with the new fan - as a fan that has inbuilt hall sensor almost surely has inbuilt protection diode. We might keep it there to be sure, but it most probably does not need to be there. What do you think ?

PeterChmurciak, 2019-11-13:
@gergelytakacs It seems that @erik1392 and also Mr. Rabek were right about linear RPM-Power characteristic of the fan. I have got:

[image]

With 25 millisecond sampling period and 25 seconds at each power level from 5% up to 100% incremented by 5%. It looks to be linear, at least from 5% to 90% (the line was fitted based on this region), at levels higher than 90% it slowly (nonlinearly) gets closer to max RPM value. Equation that characterizes this line is `RPM = 172 * Power(%)`.

gergelytakacs, 2019-11-14:
@erik1392 @PeterChmurciak 

Certain things from your [comment](https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-553073875) have been resolved and discussed yesterday. Your way of reading in RPM given the constraint that we need to use Timer2 are in general correct and I have no issue with that. My idea of using capture mode is not feasible, since this is only available on TImer1.

Regarding the issue of removing or leaving the [diode](https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-553415281) I leave this decision up to you guys, either is fine. On one side if the circuit of the fan already contains one, then this is not needed, thus making the setup even simpler (and cheaper). On the other side, at least there will be someting on the PCB:))) But really, up to you.

Nice job with the [static RPM characteristics](https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-553653762). We did learn something. Such a chart will be welcome in your thesis, but 
- I'd prefer if you'd have PWM levels (0-255) on the x-axis, since having percents introduces numerical errors. 
- Make a test that does the test automatically (e.g. raises PWM and dumps results to Serial) so that you have a better resolution, e.g. each number 0,1,2,3 ... 255. 
- The fitted line can be then used to create a PWM vs. RPM function (actually RPM -> PWM) and I'd also constrain the function to exclude this upper saturated part.

Overall, nice job.

PeterChmurciak, 2019-11-14:
@gergelytakacs 
RPM - PWM dependency:

[image]

Data were acquired with 25 millisecond sampling period and 25 seconds (1000 samples) at each PWM level from 7 (minimal PWM for the fan to start rotating) up to 255 incremented by 1. It also looks to be linear, at least from 7 to 230 PWM (the line was fitted based on this region), at levels higher than 235 it slowly (nonlinearly) gets closer to max RPM value. Equation that characterizes this line is `RPM = 68 * PWM`.

PeterChmurciak, 2019-11-14:
@gergelytakacs I have used breadboards and cables to connect Arduino with R1 shield and new fan in a way to simulate R2. I have used external pull up resistor and did not use the diode. At the first glance everything seems to work as it should, only PID constants will have to be modified. We should now decide if we want to add something more to the R2 shield (some shiny LEDs or some spectacular thermometer) and if not, I think we can regard this as a finalised R2. 
One note that the measurement from post above will have to be repeated when everything is put together, as this measurement was made with fan positioned horizontally lying on the table. With fan connected onto the shield and positioned vertically, at highest power level I was getting slightly higher RPM - specifically around 17300 (at vertical position I got around 16200). I think this is caused mainly by the fan orientation.

PeterChmurciak, 2019-11-14:
@gergelytakacs @erik1392 Latest scheme and layout:

[image]

[image]

[image]

Note1: The 3.3V connections could be technically solved with single line but @erik1392 advised me that it is better practice to bring supply voltage to each component separately, this is how I interpreted it. 

Note2: I have moved the potentiometer to the very edge of board for better access. The position was inspired by [BoBShield](https://github.com/gergelytakacs/AutomationShield/wiki/BoBShield#pcb-layout) (with ground connection touching the USB port).

Note3: Version is `FloatShield v2.0` and `R2, 11/19`, not sure if the the `RX` and `vX.X` numbering works like this.

[23 more replies]

closed 2020-02-24 by PeterChmurciak