gergelytakacs/AutomationShield #183: FloatShield: Simulink API and Examples

gergelytakacs, 2019-07-17:
The FloatShield will have an API for Simulink. Please look for the OptoShield and HeatShield Simulink API for details of how this can be done. Some starting notes are:

- The hardest part will be to implement the VL51L0X library for the sensor.
- Look into the OptoShield on how to do a calibration.
- The first example should be the PID one. Again, look for Opto/Heat for inspiration. Aim to replicate the results of the Arduino IDE PID example, e.g. with the same sampling, tuning and references. If it doesn't work for some reason, no worries - but it should.

PeterChmurciak, 2019-07-18:
@gergelytakacs I have started with implementation of the sensor. It is troublesome as expected, but it seems doable. 

I encountered this error when trying to look at the other APIs:

[image]

It can not be opened in older versions of MATLAB so I am currently installing newest version - 2019a to possibly solve this problem. Hopefully the newer versions are capable of opening older ones.

gergelytakacs, 2019-07-19:
@PeterChmurciak

Yes, this is a well know "feature" of Simulink. Models can be only opened in the same or newer versions, the are unfortunately not backwards compatible. It's normal....

PeterChmurciak, 2019-07-22:
@gergelytakacs I have managed to sucessfully port the VL51L0X distance sensor library into Simulink. 

With that I was able to create the Simulink API for FloatShield, and add it into AutomationShield model library -  b2eaa3b. Please review the API when you can. The blocks do not have mask pictures yet, but I will try to find some suitable ones. 

I have a few questions.

I have used MATLAB version R2019a to save the files, is it acceptable ? Or should I rather save it in some older version compatible format ?

I am currently working on the PID Simulink example, and the functioning of Simulink confuses me - specifically the concept of time in Simulink. In Arduino IDE and in MATLAB the time was somewhat real-time, and we got it using timers/interrupts and tic-toc functions. 
In Simulink, multiple blocks can have different sampling periods and additionaly you can set "main" sampling period using `Fixed-step size:` parameter in solver settings. After setting all of this, the Simulink simulation runs at its own pace, using its own sampling period. I am currently very confused about how does the Simulink sampling work and will look for some logical explanation on the internet. If you could help me understand it a little bit more, I would be very grateful.

PeterChmurciak, 2019-07-23:
@gergelytakacs I seem to be having some memory issues:

[image]

And

[image]

The warning about source sample time is caused by 6a33885 change, and it is somewhat intentional.

What concerns me are the "not enough memory" warnings - the program runs despite them but when they show during compilation, the data are not being displayed on the Simulink sinks like `Display` or `Scope`. They can be sent to workspace using `SimulinkToWorkspace` block and then manipulated, but I am not sure about their correctness.

For example I have tried to run the Simulink PID example using sampling 20 miliseconds (it was done before 6a33885 so I was forced to use 20ms) and I have set the simulation time to 600 seconds. Each 60 seconds the reference changed to next level - same test as in MATLAB and Arduino IDE. The whole thing was supposed to take 10 minutes, but for some reason, It took the Simulink 1 second to advance 0.1 second in simulation, so the whole thing took 6000 seconds - 1hour and 40 minutes. The first image in this post is from compilation of this simulation - there was some problem with memory. The result of the simulation - sent to workspace and plotted:

[image]

The PID worked pretty good - the ball seemed to follow the reference trajectory pretty nicely. What is weird about it is the inconsistent noise in readings at different reference levels. Possibly it can have something to do with the memory warning ? Not sure.

The duration of this experiment further confused me about workings of Simulink "time". There were 30000 samples in the results - at 20ms during 600 seconds, this is a valid number, however it took 6000 seconds of real time. Is this normal in Simulink ?

PeterChmurciak, 2019-07-27:
@gergelytakacs I have tried sampling 25 miliseconds and this was the result:

[image]

Again 600 seconds simulation stop time, with 25 miliseconds, got 24000 samples - good, but it took much longer in real-rime than 10 minutes.

In this result the noise is even worse than before for some reason, especially above value of 50%. 

I have tried using just `Sensor Read Distance` block with scope block to see if there is someting visible on there - used 25 milisecond sampling and manually changed the position of the ball:

[image]

But there was no significant noise at any level.

To make sure I have moved the ball to around 70% position and let it there:
 
[image]

No significant noise at 70% position - there seems to be nothing wrong with the sensor readings this time.

I am currently not sure what could cause it. When doing these two tests, I have used only simple blocks so there was no "Not enough memory" warning (when there is this warning, the results can not be displayed with scope/display blocks) - It might have something to do with it, or the issue might be somewhere in the `FloatShield` simulink block. Further testing will follow.

PeterChmurciak, 2019-07-27:
@gergelytakacs This is result with 50 millisecond sampling:

[image]

And this with 100 millisecond sampling:

[image]

I was trying if different sampling periods would change anything, but besides faster simulation real-time duration, they did not change much.

I have looked at the ball behaviour during the simulation, and based on that, the results might be actually accurate - for some reason, the simulink has trouble keeping the ball in a stabilised position at higher portions of the tube - the ball tends to visibly go up and down. 

This might be possibly because stabilisation at higher portions of the tube requires faster reactions, that simulink can not provide. Arguably, the best results were acquired while using 20 millisecond sampling, and with increasing value of the sampling period, the results got worse. 

I thought that 20 milliseconds is the sampling speed limit, because of the laser sensor restrictions, but it seems that 20 milliseconds in simulink are not 20 milliseconds of real time, so I will try to go lower to see what happens. The simulation will surely take even longer than before...

PeterChmurciak, 2019-07-28:
@gergelytakacs This is result with 15 millisecond sampling:

[image]

And this with 5 millisecond sampling:

[image]

These tests took unholy amount of time to complete...

If I had to draw some conclusion from these tests, the best results were acquired while using minimal hardware-possible sampling of 20 milliseconds. When trying to use faster or using slower sampling, the quality of ball-position control would deteriorate, and there would be much more vertical oscillation (or noise ?). The memory warning possibly might not have any effect on the results, just on the ability to display them using scope/display blocks.

In terms of quality of the results, the best sampling to use would then be the 20 millisecond one. (I will probably repeat the experiment once more with this sampling to make sure it was not just coincidence).
However, as I have mentioned, the test with 20 millisecond sampling and 600 second simulation stop time (10 reference levels with 60 seconds to stabilise - to be consistent with previous PID examples) takes about 1 hour and 40 minutes to complete - that seems a bit too much to me. Is that acceptable ? I would like to hear your opinion on this issue.

gergelytakacs, 2019-07-29:
@PeterChmurciak I'm trying to answer your questions in order from previous posts.

> I have used MATLAB version R2019a to save the files, is it acceptable ? Or should I rather save it in some older version compatible format ?

Well, this is a problem that is new to the AutomationShield at this point and we will have to decide together. I suggest the following two possible scenarios, either is fine: Save it in 2019a, but then (maybe) re-save the other two device API's as well, so that there is no discrepancy between Simulink versin compatibility, or b.) save it in 2018b (or whatever the others were saved in) and that way at least there is some backwards compatibility. In this case the entire library will have to be re-done at some point in the future for Simulink, if we are forced for some reason (new features, incompatibility etc.)

> the functioning of Simulink confuses me - specifically the concept of time in Simulink. [...]

What Simulink does, is that it takes your block scheme (proprietary "programming language") and transcribes it into C code. You can actually check out and inspect this ugly and inefficient machine code in the temporary files. Now, the way Simulink does time is exactly the same as you've done and learned in Arduino IDE. It uses timers and interrupts, and almost the same logic as Arduino IDE! :) So you don't have to worry about timing in most cases. The settings are a bit confusing for one very good reason: Simulink allows continuous math as well, and for that it needs solvers! We don't have to worry about such things at this point, so the main setting should be `Discrete` and `Fixed Step` (if I remember the keywords correctly). The sampling you enter here will be used as a sample size, as you are used to it in Arduino API. 

What might have confused you are the sampling times in the blocks. Now this of course has a meaning for some more advanced stuff, but in a nutshell, it just lets you to slow down certain parts of your simulation. Imagine coding in C a part where you say that you update the sensor only every 10th sample. It would be actually pretty easy to do (but it would complicate the "math" behid discrete control a lot.) In older versions, at least, it lets you to enter a.) only multiples of the "main" sampling time, and b.) only slower than the "main" sampling time. What I suggest is that you use the same everywhere or inherited (-1). So don't worry too much about this, Simulink does sampling pretty well, and being able to use different sampling in different blocks is a feature:)

gergelytakacs, 2019-07-29:
> I seem to be having some memory issues.

I actually expected this to jump at us at some point, As I've explained above, Simulink translates the blocks (your programming "language") into C by machine and it is not very efficient. What you do by hand takes a lot less memory and it is most of the time more efficient. This PID example then (and I'm referring to the version you've encountered the memory issues) is just too large to fit on the memory of the Uno. What you can do is "make it simpler", which is not really an option for this already quite simple example or just declare that the API is not compatible with the Uno and use a Mega at least. (I'll lend you one). This is acceptable for me, as we are not here to solve the limitations of MATLAB/Simulink... Now the fact that it still runs is interesting, and you always need to differentiate between compiler (even if it's Simulink) warnings and errors. If it is a warning, you, should be fine, **but** (and this brings us to the next problem:

>data are not being displayed on the Simulink sinks 

**but** to solve the memory issues above, Simulink chooses to turn off certain functionality, and one of the most memory and time intensive ones is actually the "External Mode" - e.g. sinks and switches with your computer. I've seen this before. Fun fact: older versions of Simulink did not let you run in External Mode on an Uno at all because of this, only Mega! :-) Again, if all else works well (and haven't read the rest of the thread yet) this is not a critical problem for us, just say that the Uno is too simple for External Mode (but otherwise works) and use a Mega.

> the whole thing was supposed to take 10 minutes, but for some reason, It took the Simulink 1 second to advance 0.1 second in simulation, so the whole thing took 6000 seconds - 1hour and 40 minutes. [...] The duration of this experiment further confused me about workings of Simulink "time". There were 30000 samples in the results - at 20ms during 600 seconds, this is a valid number, however it took 6000 seconds of real time. Is this normal in Simulink ?

Now that is worrying though and is not normal. If the closed-loop control works, that's possibly a small wonder. To explain this you have to understand what happens when you violate your real-time constraints on hardware: Simulink (on your computer) does not know about this at all. Actually this can be explained like this: because the scheme is too complex for the memory, Simulink throws out the "External Mode" to communicate with the host (your PC). So, if the sample time is violated, your PC will not know about this. Instead, the target (Arduino) just proceeds with the simulation, albeit, with a reduced speed.

What you can do about it is check in settings to enable overrun detection, e.g. to light up the LED onboard:
[image]
if the LED is lit up, you are violating real-time constraints! Now the fact that it takes 10 x more time means that you can only achieve ~200 ms :( The trouble may be the sensor and its driver, since ~20 ms is doable in Simulink even in External Mode. So you have two choices: speed up your example by finding the bottleneck and use 20 ms, or just use a (much) longer sampling and re-tune PID. Of course the Mega will be just as slow as the Uno, since they are clocked at the same speed. 

TLDR; 
- Use a Mega to test the Simulink API (I will lend you one) and just claim that Uno does not allow for External mode use because of memory restrictions.
- Either speed up your example by finding the bottleneck (sensor for sure) but this may be not feasible or reduce sampling speed and re-tune PID. Either is fine, we are talking here hardware / Simulink limitations not our limitations as programmers. This case there will be a Simulink API, there will be a PID example, it will be just much slower sampled...

gergelytakacs, 2019-07-29:
@PeterChmurciak  This is in response to [this comment](https://github.com/gergelytakacs/AutomationShield/issues/183#issuecomment-515686425):

> But there was no significant noise at any level. 

Leave the noise alone for a while, you have more important issues - the sampling. The noise (or nonlinearity) may seem to behave entirely differently as your one basic thing (sampling) is not entirely okay. See comment above.

> This might be possibly because stabilisation at higher portions of the tube requires faster reactions,

Could be. If you don't find the bottleneck, you may never achieve that speed. So just use a different sampling, create a different reference trajectory (not necessarily an hour long) and acknowledge the problem.

gergelytakacs, 2019-07-29:
@PeterChmurciak This is in response to the [next comment](https://github.com/gergelytakacs/AutomationShield/issues/183#issuecomment-515796878):

>I was trying if different sampling periods would change anything, but besides faster simulation real-time duration, they did not change much.

Yes, because neither of those sampling times were actually achieved. It maxes out at some level, could be aorund 200-250 ms, and that's it. So no matter what sampling time you choose below that limit, it will always be the same. The rest of your discussion relates to this fact, so I'm not going to discuss it.

gergelytakacs, 2019-07-29:
@PeterChmurciak This is a TL;DR for all the things before. Now 

**TL;DR**
- Use a Mega to test the Simulink API (I will lend you one, if you need) and just claim that Uno does not allow for External mode use because of memory restrictions. This will solve your memory issues. However this may allow the External Mode (unlike previously) which may reduce your sampling even further...
- Your example cannot keep up with real-time sampling, it maxes out at much slower speeds. To see this confirmed, turn on "Overrun detection" and send it to a digital pin (e.g. D13, the LED). As long as that is active, your sampling is too slow for real-time and the example is not correct.
- Attempt, to find the bottleneck. It could be that the automatic translation to C from the block scheme is just crap and you can do nothing about it. On the other hand, there may be something you did that slows down the system compared to the other API's.  My guess the bottleneck is in the sensor or I2C communication for sure. Unfortunately removing the bottleneck (slow part) , may be not feasible at all. If it is by any chance, problem solved, if not, then...
- Reduce sampling speed, create another reference run (that doesn't take an hour:) and, if needed re-tune PID. Either is fine, we are talking here hardware / Simulink limitations not our limitations as programmers. This case there will be a Simulink API, there will be a PID example, it will be just much slower sampled...

PeterChmurciak, 2019-07-29:
@gergelytakacs I find this very unfortunate, but it would be weird if there were no problems with it. I have tried the overrun detection feature as you proposed, and unfortunately you are right... Your guess was spot on, as the minimal sampling period without overrun is 200 ms. 
I will try to find the culprit of this slowdown and if possible, resolve it.

gergelytakacs, 2019-07-29:
@PeterChmurciak  Thank you for the confirmation, sorry to be a bearer of bad news!:)

PeterChmurciak, 2019-07-29:
@gergelytakacs For the record, this is result using 200 ms sampling period:

[image]

The result is not spectacular in any way, but at least the test took 10 minutes of real time as it should - the problem was with the sampling speed.

I will do my best to allow the simulation to use faster sampling and if that will not be possible, I will retune PID constants to better suit this sampling period.

gergelytakacs, 2019-07-29:
Well, this is what I expected. I would at this point 

- See if faster sampling is possible by finding the bottleneck and if not
- Set 200 ms sampling, change the reference and re-tune the PID 

As this won't be the same as the MATLAB or Arduino API's you can use a different reference! You already know that eventually the controller will settle, even if it takes a lot of time (your previous experiment). So don't obsess over trying to do the same run as in Arduino, just assume this is a new case and make it pretty:)

PeterChmurciak, 2019-07-29:
@gergelytakacs I have managed to resolve the sampling issue. The bottleneck was sensor library implementation to simulink. The problem was... Trivial... 

When setting the sensor to fast-sampling mode I have to set its sampling limit to minimum - to 20 milliseconds. The function to do that takes input in microseconds - so it would be 20000. I have somehow put additional 0 there, so the sensor was forcibly limited to use 200000 - 200 millisecond sampling.

I have set it to 20 milliseconds (39f9159) - now the fastest sampling simulink can use without overrun is 20 ms.

This is result of PID example with 25ms sampling period for 600 seconds (without overrun) - same reference as in MATLAB and Arduino API:

[image]

The simulation took 10 minutes as it should this time.

The overshoot at the begining is probably caused by a lack of lift-off routine, and there is still some oscillation at higher portions of the tube.

What do you think ?

gergelytakacs, 2019-07-29:
@PeterChmurciak 

Awesome job and kudos on finding and pinpointing the bug! :-) I'm happy that the issue got resolved and the results look very pleasing to me actually! I'm happy w/ how it turned out so no re-tuning or anything like that is necessary.

Now, about the liftoff routine. Why not try to implement it? Would it be hard to do? The Simulink models are actually pretty flexible about these things and you could insert a "before run" block into the collection. Let's say this block runs before the actual simulation and lifts the thing up like you do in Arduino IDE. Just and idea though, this is already quite cool/fun.

gergelytakacs, 2019-07-29:
@PeterChmurciak 

I had a look at the Simulink API and the blocks themselves. You have done an excellent job, thank you. I possibly have some minor comments on naming / duplicates and placement. See the other comments in the commit... Now, let's try to make a short-term plan to keep up the work:

- Go get a beer, you deserve it for finding the sampling bug:)
- Attempt to make a block that implements a lift-off routine, like you did in the Arduino IDE. Not overly important, but since your results are excellent, why not.
- If there is a liftoff routine, perform an experiment identical to the MATLAB/Arduino API ones, save the results for later use in the academic paper presenting the results.

After this, the fundamentals are done. The API works well. However,  a more serious phase follows that will need more control engineering. But first, let's try to come back to the question of modelling and acquiring a good test signal (discussed elsewhere).

gergelytakacs, 2019-07-30:
- Created graphics for existing FloatShield blocks and included them in the scheme
- Renamed FloatShield.slx to Float.slx for consistency, corrected it all in AutomationShield.slx
- Created a script to list model versions (modelManagement.m). This may later include automatic re-saving the blocks in other versions.
- Re-saved models in 2019a for consistency
- Changed photo in FloatShield block collection (featured in AutomationShield.slx)

PeterChmurciak, 2019-07-30:
@gergelytakacs Thank you, the blocks will look better with their own characteristic pictures. 

Interesting and practical idea with the modelManagement script. 

Was there not a problem when renaming the FloatShield to Float ? I remember having a problem when trying to do that, as the "float" filename was somehow reserved by MATLAB.

gergelytakacs, 2019-07-31:
@PeterChmurciak There was no problem, but I did not compile or run the blocks!:) So please check, and if there is a conflict with the MATLAB keyword (which, I think you are right about), then just rename all to `*Library` as you've suggested in the other comment.

gergelytakacs, 2019-07-31:
@PeterChmurciak 

To present the results (in an interesting way), we'd need not only a data file to plot from the Simulink results, but also a scope screenshot from a live run. Now, to do this you'll need to swap the Uno to Mega because of the memory issues, but otherwise it shouldn't be too much of a trouble.  Have a look at how the Scope is configured in the HeatShield example: it is divided into two parts (top and bottom). The top has reference and output, while bottom has input. Also, (for visual reference) can you try to make an experiment that sets height in mm directly, instead of percentages?

PeterChmurciak, 2019-08-03:
@gergelytakacs I have tried if the model name causes any issues but it does not. The warning that caused me concerns was:

[image]

And it was caused by the name "float":

[image]

To test effect of using capital letter:

[image]

The matlab knows a file named "float" and is therefore concerned about the name Float/float used somewere else. 

However the files are differentiated by using first capital letter, so they should not be interchangeable.
As the usage of name "Float" causes no problems and no warnings or errors, I will keep it as it is.

PeterChmurciak, 2019-08-03:
@gergelytakacs When working on moving the files based on your feedback, I have encountered a warning after compilation: 

[image]

So I changed my mind, as there is actually a warning about that issue, and renamed the `Float`, `Heat` and `Opto` to `FloatLibrary`, `HeatLibrary` and `OptoLibrary`, to solve this shadowed files warning.

gergelytakacs, 2019-08-05:
@PeterChmurciak 
Sure! I had no preference whether it should be `Float` or `FloatLibrary` - so this is completely acceptable. Also I did not think of the naming conflict!:)

gergelytakacs, 2019-08-05:
@PeterChmurciak 

Short-term goals when you have the time and the will:)

- If you could re-tune for direct position feedback in [mm] for Simulink that would been swell, so it does not look too boring in the paper. You can create a separate example for this. It should not be too big of a hassle, I guess. Say you scale the reference to [mm] by the factor k (e.g. 0.5), then you can attempt to change just "P" by the same factor, so you get "k.P" and I think it should work. The only meaning of this would be to create a visually more interesting plot.
- Screenshot of the "Scope" with PID control. 4:3-ish sides, not too big, leave in the window controls.  Use two separate inputs, top will be reference and output, bottom will be input.
Goes in the paper.  e.g. 
[image]
- Arrange the PID example Simulink scheme nicely for the blocks, so it's compact, etc. Export the scheme into a vector format. This can be a bit of a hassle, but printing to a PDF is a possibility. As far as I remember, Simulink had trouble with it, and it saved the scheme in a PDF but as a vector. To force it to pint in vector, you have to use MATLAB command line - I Googled it as well. e.g.
[image]
- After this is done, we have to go back to modelling, that is a priority. Gathering input output data should be the next job.

PeterChmurciak, 2019-08-06:
@gergelytakacs I have tried to upgrade the `Sensor Read` block, but ran into some limitations 0ac49bf.

I have added PID example 9788359 that can be run on Arduino Uno. The results look like:
(Could not pinpoint the cause of "noise" in upper positions) 

[image]

The cropped version gets saved into .mat file and plotted on the screen: 

[image]

Variables with both uncropped and cropped versions stay in the workspace and can be inspected if needed.

I will try to modify if to use millimetres instead of percentage.

Tomorrow (wednesday) I will come to school to borrow Arduino Mega, as I dont have one, and can not use Simulink Sinks because of Aruino UNOs memory size.

I am not quite sure about the `Simulink Coder` as I have not seen that name until now. The responses in [post1](https://www.mathworks.com/matlabcentral/answers/347856-upload-code-to-arduino-from-simulink-coder) and [post2](https://www.mathworks.com/matlabcentral/answers/306163-generate-c-code-from-simulink-and-place-it-into-arduino) make it seem that running a model on Arduino does not use Simulink Coder. 

On the other hand, in the [video](https://youtu.be/eVXQBUHMYCU?t=27) from MATLAB there pops up a `Code Generation Report` window, that indirectly signifies a usage of the coder, and that window does indeed pop up at the begining of my every simulation. Based on that I would say that the coder is needed.

gergelytakacs, 2019-08-12:
@PeterChmurciak  To respond to your latest post in this thread.

> I have tried to upgrade the Sensor Read block, but ran into some limitations 0ac49bf.
Yes, sure. This is a known limitation. Your solution is excellent, I have no issues with how you've done it.

> I have added PID example 9788359 that can be run on Arduino Uno.

Excellent, I have no issues with the model. Looks good and I like the Reference block, I was not aware of its existence.

> I will try to modify if to use millimetres instead of percentage. 

Yes please.  Leave the Uno example alone, you can do it for the Mega now. Just copy the example and change it for Mega. We'll only need screenshots of the model (vector) and the scope of the results for the paper.

> The results look like: (Could not pinpoint the cause of "noise" in upper positions)

I think the results are fine. I am not too worried about the "noise". It can be that their PID block has a slightly different implementation? Could be, but I'm not disturbed by this at this point so just leave it be. Could you maybe post a `mat` file with these results? I don't want to include that in the paper, but I'd like to save it - just in case it is needed in the future.

> I am not quite sure about the Simulink Coder as I have not seen that name until now. The responses in post1 and post2 make it seem that running a model on Arduino does not use Simulink Coder. [...]

I am not entirely sure about this either. This was just out of curiosity to know, don't sweat it. Based on the [description](https://www.mathworks.com/products/simulink-coder.html) it seems like that is the module that allows to translate the block scheme to C/C++ and it is handled as an extra product, not a part of the garden variety Simulink. They say it's required [here](https://www.mathworks.com/matlabcentral/answers/262462-how-to-install-simulink-coder), and they mention it [here](https://www.mathworks.com/hardware-support/arduino-simulink.html) on the product support page but don't list it as a requirement.  I've also ran into an error message when I tried to compile on a machine without Simulink Coder. So I'm leaning into "yes, it's required" but still not completely sure.

PeterChmurciak, 2019-08-12:
@gergelytakacs Actually the `Reference` block does not exist, or at least I could not find any that would suit this cause. 

It was custom-made for the PID example using `Matlab Function` block - that is another wonderful tool to incorporate matlab code, specifically matlab functions, into Simulink. Whole `FloatShield` block including calibration could be made using this single block (I have tried that hoping that it would reduce memory needed and possibly allow Scope usage on Uno), but when testing it It caused later errors in PID example concerning algebraic loops, that could not be solved (actually only solution that I have found was to use `Memory` block that created single step delay in feedback loop, but I did not want to add any delay).

With some modifications the `Reference` block could be used as default reference block in our library if you wanted to, I just was not sure where we would place it.

I have finalised the Uno PID example and modified some blocks in library 2aefb32. The results that I got from this PID example (including .mat): [PID_Uno 12.8.2019.zip](https://github.com/gergelytakacs/AutomationShield/files/3494441/PID_Uno.12.8.2019.zip)

The `FloatShield` block can now output ball position(altitude) either in % or mm, so the Mega example will simply use this second (mm) option, with modified Reference Vector and PID constants. I have not tried to use the Mega board yet, so I hope that these will be the only changes to the example needed. 

I will start working on that.

gergelytakacs, 2019-08-13:
> Actually the Reference block does not exist [...]

Oh, I was silly and assumed it is a built-in function, but now I opened the mask and I see it's custom.

>  but when testing it It caused later errors in PID example concerning algebraic loops, that could not be solved (actually only solution that I have found was to use Memory block that created single step delay in feedback loop, but I did not want to add any delay).

Memory / Delay in this case should be the same. You may actually insert a [z^-1] block, which from your studies in TAR II should be clear that, yet again is the same. But this is no concern for worry. Actually I found that sometimes Simulink can be weird about this and assume that you want to send the input as soon as it is computed - so like mid-sample. (Played a lot with this a while ago). So to prevent that you actually needed the [z^-1] block (delay/ memory) and then the input was sent at the right moment. But this is not really well documented and you shouldn't worry about it at this point. 

> With some modifications the Reference block could be used as default reference block in our library if you wanted to, I just was not sure where we would place it.

I think that is a great idea. If you could make it more universal, you should include it in the library. I'd just create a "Common" or something library module and collect blocks not specific to any given hardware there.

> The results that I got from this PID example (including .mat): PID_Uno 12.8.2019.zip

Thank you, I just archived it.

> The FloatShield block can now output ball position(altitude) either in % or mm, 

Seen it, looking very good!

>  I have not tried to use the Mega board yet, so I hope that these will be the only changes to the example needed. I will start working on that.

Sounds like a plan. So just to repeat, the goal now is to get a nice screenshot from a "Live" test w/ units in mm (PNG, scope controls shown, not too big so they are in good proportion) and a vector oputput (PDF/EPS) of the model. I killed time with something else today/yesterday but the remaining factor at this phase is the modelling.

PeterChmurciak, 2019-08-13:
@gergelytakacs I wanted to show you what I talked about in my last post. 

See 77fb0fe, in this example, the `FloatShield` block under its mask consist of single block - `Matlab Function` block. This model also uses `Memory` block to get rid of wild errors caused by usage of persistent variables in the matlab function in algebraic loop.

By using this modified `FloatShield` block, it is possible to see results directly on Scope, while also saving them in .mat file and plotting them on figure, even on Arduino Uno. The results look like this:

[image]

[image]

More Scope screenshots and .mat file with results (if needed): [Alternative_SimulinkPID 13.8.2019.zip](https://github.com/gergelytakacs/AutomationShield/files/3497594/Alternative_SimulinkPID.13.8.2019.zip)

Surprisingly the `FloatShield block` change or addition of `Memory` block improved the results - specifically the "noise".

Usage of this modified block seems to bring some benefits - like improved results and also not needing to create separate PID examples for different boards, as it can run on Arduino Uno. 

This brings the question if creating whole FloatShield block using single matlab function is acceptable by your standards. What do you think about it ?

gergelytakacs, 2019-08-15:
@PeterChmurciak 

As I said in my comment related to your commit, I find this very interesting. If you verify that real-time sampling is not affected (it shouldn't though) I have no problem with a FS block that uses the MATLAB function instead of the discrete blocks. 
**TL;DR**: If it works well, you may swap the original FS block to the new alternative using the MATLAB Function and have just a single PID example (and not separate). It seems to be a better solution, if real-time sampling is not affected.

Can you do another test / screenshot where you fit a bit more to the X (Time axis)? Also please change "Ball position" (e.g. output) to be the top chart) and the other one the bottom. Also change "Regulator output" to "Input" or even rather "Fan power". 
[image]

Note: Our guys lecturing TAR I / TAR II tend to be inconsistent in naming "inputs" and "outputs" and create a mess in the heads because of this fundamental mistake. I tell you how it works: the signal naming is related to the plant, not to the controller. That means, any sensor reading is an output (from the plant) and any actuator effect is an input (to the plant). Any other naming scheme is wrong.

PeterChmurciak, 2019-08-15:
@gergelytakacs To answer your questions from https://github.com/gergelytakacs/AutomationShield/commit/77fb0fe94e9936a87c2153191b70197a86682aea#commitcomment-34699817 :

* I have found no drawbacks regarding sampling speed. There is no overrun detected even if using 20ms sampling - so this should not be an issue.
* That is a good point, if hipster, then hipster all the way! I will implement PWM block directly.
* Only thing that is directly comparable that I know of is the report after compilation. So I compared that:

When using "classic" `FloatShield` block we get:

[image]

And when using "new" MATLAB Function created  `FloatShield` block we get:

[image]

Conclusion - When using "new" block, Program uses up slightly more memory (+~2%) and Data take up less memory(-~6%).

If we compare whole PID examples:

When using "classic" `FloatShield` block we get:

[image]

And when using "new" MATLAB Function created  `FloatShield` block we get:

[image]

Conclusion - When using "new" block, Program uses up approximately same amount of memory (+~1%) and Data take up less memory(-~9%). (Note that when using "new" block PID example there is additional `Memory` block for it to work)

It seems that even when using a single block, during compilation simulink manages to take up more than half of the Arduino Unos memory. 

If memory is an issue, there seems to be a real benefit in using `MATLAB Function` block for buidling other blocks, as that difference in the PID examples - 66.2% vs 75.2% (Data) means in reality working Scope block vs not working scope block.

Only downside I see to usage of this block is the disappointment when curiously looking under the mask of the block, only to find a single MATLAB block there. (Surprisingly MATLAB Function is better at being Simulink block, than Simulink block is)

PeterChmurciak, 2019-08-15:
@gergelytakacs I have tried it with the simplest block of FloatShield library: `Actuator Write`

Using normal Simulink Blocks:

[image]

Using `Matlab Function` block instead of normal blocks:

[image]

While in this case the difference is not that big, it seems that `MATLAB Function` block is the most memory-efficient way to build more complex simulink blocks (And therefore best way for External mode simulations). 

I therefore propose the idea of recreating all FloatShield library blocks while using  `MATLAB Function` block in them whenever possible. What do you think about that ?

PeterChmurciak, 2019-08-15:
@gergelytakacs My bad, simplest block is `Reference Read`

Using normal Simulink Block:

[image]

Using `Matlab Function` block instead of normal block:

[image]

So it depends on the case. The Radio options that are selected on the mask need to be taken into account. If it uses 2 and more blocks and can be replaced with `Matlab Function` block then it will probably be more efficient and should be replaced. If it is not more memory-efficient, it will stay as it is. This `Reference Read` block would then stay as it is, as there is only 1 Gain block used.

gergelytakacs, 2019-08-16:
Thank you for the comprehensive comparison. Looking at the memory footprint of the compilation was a good idea and provided the needed insight into the problem. To me, it seems that there is only a minimal difference between memory footprint - however, this remains in this case essential as it means the difference between the availability of external mode of the Uno. I don't think the differences are not that dramatic... With higher models (over the Uno) these differences become even less important.

I suggest you adapt the "Alternative" FloatShield for the library using the MATLAB Function, however, save the original somewhere - let's say in `assets\src` if we ever need it in the future.

>I therefore propose the idea of recreating all FloatShield library blocks while using MATLAB Function block in them whenever possible. What do you think about that ?
You may do that if you want, but it's not all that important. I leave this up to you, but I have to note that there are more pressing and complex issues remaining unsolved here. As you proved yourself, it's not worth remaking the simple ones through custom blocks since what you saved in Actuator write, lost in Reference read. 

TL;DR:
- Use the "Alternative" block for the overall library, but save the original somewhere else just to be sure. 
- Since you've proved that the memory save depends on the case, I'd leave the other blocks alone for now. Especially since there are more pressing problems right now - do not get sidetracked!:)
- Please have a look at my requests [here](https://github.com/gergelytakacs/AutomationShield/issues/183#issuecomment-521529445) so we can "close" the Simulink portion for now and move back to system identification.

PeterChmurciak, 2019-08-19:
@gergelytakacs I have modified the LibraryBlocks  to save some memory through usage of Matlab Function blocks 5b01846.

The PID example is more or less complete cd2b9c2, with your feedback in mind:
The scope now looks:

With 45 second x-axis range:
[image]

With 60 second x-axis range:
[image]

How big time range did you have in mind ? I think 45 seconds looks good with this smaller window format as the 60 seconds one did not resize y axis on Fan power part as naturally, but that can be probably managed in settings if needed. However current section length is 60 seconds so I would not go much higher as to not show multiple "jumps" in one window.

Current results look like:

[image]

The "noise" in higher portions of tube was not caused by any of the blocks that I have changed - the change of blocks did not improve the results as I stated in https://github.com/gergelytakacs/AutomationShield/issues/183#issuecomment-520885234, the quality of the results depends on (at least for now) an unknown factor.

Here are more screenshots of Scopes for you to see [SimulinkPID 19.8.2019.zip](https://github.com/gergelytakacs/AutomationShield/files/3518021/SimulinkPID.19.8.2019.zip).
I also included the Simulink PID scheme saved in PDF and svg. Please see if that is the vector format that you talked about, and if it needs to be changed.

gergelytakacs, 2019-08-20:
Awesome. Everything is as it should be and I think the Simulink API is finished (for now). More to come when the identification is ready but that shell be easy and fast once we get a decent model.

I've inserted your new contributions into the paper, see here or on Dropbox: [IFAC_WC_2020_FloatShield.pdf](https://github.com/gergelytakacs/AutomationShield/files/3520408/IFAC_WC_2020_FloatShield.pdf) (I've not worked on the text, just put the images inside).

Please devote your time to system identification, that is #167 #166 and possibly #184. I can't be much help right now, but @erik1392 promised to help after he gets back from his vacation, and @mgulan also promised to contribute - so you can try to talk to them. Let's continue identification related stuff in those threads.

closed 2019-08-20 by gergelytakacs