gergelytakacs/AutomationShield #184: FloatShield: airspeed vs. power

gergelytakacs, 2019-07-29:
It seems in the other posts that we have found a reasonable model of the system dynamics. The model and its alternates require airspeed as an input. We have to relate airspeed to fan power:

- Any analytical ideas? At least a heuristic relation.
- How would we measure it (cheaply;)?
- Which airspeed is even considered? The one with the ball or the one without?

PeterChmurciak, 2019-07-30:
@gergelytakacs As I understand it we need two different speeds, the speed of the air without the ball and the speed of ball at any given moment. Correct me if I am wrong.

We could get the speed of air without ball by using some device that would either measure speed directly (maybe air flow sensor or some small anemometer), or device that would measure volume of air that goes out of the tube (basically CFM). Then we could relate the air speed experimentally by incrementally increasing fan power.

The speed of ball is probably the harder part, as it probably also needs to be related to fan power somehow. Its speed could be possibly calculated by using the laser sensor readings - as changeOfDistance/perChangeOfTime (dy/dt). But I am not sure as how that could be related to fan power.

gergelytakacs, 2019-07-31:
@PeterChmurciak 

Assuming the model [here](https://github.com/gergelytakacs/AutomationShield/issues/166#issuecomment-510793969) is correct, and it should be, we do not need the speed of the ball since this is an internal - state - variable. You should have worked with state variables in TARII, now this is one. 

Our main concern is v_air to which I was referring to. This, I think is the airspeed **with the ball** in the column, which is a bit tricky. According to the reference there Jernigan2009, that depends not only on the power, but also the height of the ball --- which I think is pretty reasonable.

Still, this thread is more about the hardware issue - how can we measure airspeed on the cheap... (And yes, the idea is to increment fan speed and measure it, thereby establish some sort of relation).

gergelytakacs, 2019-08-01:
@PeterChmurciak 

@erik1392 will attempt to create a measurement of airspeed vs. power vs. ball height. He will use a Pitot-tube and a custom made special ball that will be statically positioned within the tube. He is already working on it. He also proposed a CFD simulation, as a possibility.

We discussed the question of "what airspeed to measure and when". Of course, the primitive models we found assume linear flow, for which it doesn't really matter all that much. We'd rather need a ballpark number or some sort of realistic scenario. We'll see how this turns out.

gergelytakacs, 2019-08-12:
@PeterChmurciak 

@erik1392 has attempted to measure airspeed by a Pitot tube and a differential pressure sensor. I'm just reporting his findings, as he sent these to me over DM privately. I have not analyzed anything as of yet. He says that the pressure difference was too small for the given sensor so the speed readings don't make much sense. He also tested the force on the ball as a function of various speeds/heights by a force sensor and says that these look pretty good. Mind you that I have not analyzed or seen the results, just archiving it here and informing you.

[float-meranie.zip](https://github.com/gergelytakacs/AutomationShield/files/3491118/float-meranie.zip)

Here is the rig he used to measure:
[image]
[image]
[image]

@erik1392 is out for two weeks on vacation though.

gergelytakacs, 2019-09-03:
@PeterChmurciak  and @erik1392 

Please try to follow up on the airspeed issue.  Any sort of identification example is interesting and useful, I'm open to all ideas.

PeterChmurciak, 2019-09-08:
@gergelytakacs The two most common methods that come up when searching for measuring air speed are by using anemometer or pitot tube. Equipment for both is a bit pricey for our purposes...

I have tried working with results that @erik1392 managed to get with our budget-friendly equipment. There were two kinds of results differentiated by name - for example `70mm_P_F` and  `70mm_F`. I guess that `P` stands for pressure and `F` stands for force, and the experiments had to do something with it, but I am not absolutely sure about that. I assumed that they were experiments under different conditions and separated them.

The results from `_P_F` experiments plotted:

[image]

And the approximate relationship between Force and Power based on `_P_F` experiments:

[image]

The results from `_F` experiments plotted:

[image]

And the approximate relationship between Force and Power based on `_F` experiments:

[image]

Just putting it out here for you guys to see what kind of data we currently have. Not sure if it can be used.

mgulan, 2019-09-09:
@PeterChmurciak I think these data can actually be helpful (to identify the system) in a way. It would be nice to have @erik1392 justify the meaning of that notation.

erik1392, 2019-09-09:
@PeterChmurciak I used pitot tube but the sensor I used have too low resolution, therefore, it was unmeasurable at given low airspeeds(I found some not that expensive alternatives that might work), the anemometer is unusable for us. 

The p_f experiments were made with 3d printed ball with pitot tube sticking from the middle, so the airflow is quite obstructed especially behind the ball. The second experiment was just to measure force and I used only the ball on the stick so airflow is almost unaltered compared to normal operation. 

Looks like the force vs. power si quite linear and steady-state force is also independent from the ball hight, however, I think transfer functions (force-power) will differ at different heights caused by change of air volume under the ball.

gergelytakacs, 2019-09-10:
Okay so:
- @erik1392 is thinking about getting a better pressure sensor to get meaningful airflow estimates. The ones in the data are useless.
- @PeterChmurciak Nice job with the relationship. The second one, according to @erik1392 's description is relevant for the force.
- @mgulan Any way the force relationship can be useful for us?
- @PeterChmurciak Maybe just for practice you could make an alternate input method to the library, in which the user inputs "mN" instead of % power! How about that?:) This is sort of a "side quest" though.

mgulan, 2019-09-16:
@PeterChmurciak or @gergelytakacs  what is the voltage amplitude of the PWM signal?

gergelytakacs, 2019-09-16:
@mgulan 
It should be 12 V. (controlled through the MOSFET).

PeterChmurciak, 2020-02-25:
@gergelytakacs @erik1392 I have found an interesting air speed sensor, also known as hot-wire anemometer [Wind Sensor Rev. P](https://moderndevice.com/product/wind-sensor-rev-p/) or [Wind Sensor Rev. C](https://moderndevice.com/product/wind-sensor/). 

Could be potentially useful if we would want to pursue PWM to Air Speed relationship.

gergelytakacs, 2020-03-04:
@PeterChmurciak 

Awesome find! I like it. So, because of the relatively high price (compared to the rest of the hardware), I'm guessing you plan to use one to verify/identify a model. That sounds good. Where (physically) would you build this into the tube? Any extra data is useful, but since I've lost my contact with the model (read: don't remember) I cannot judge what role wind speed plays in there.

Lemme know if we should get one, and which one. (I guess the better one with temp compensation).

@erik1392 @mgulan Please have your say.

mgulan, 2020-03-04:
@gergelytakacs I concur. Interesting. As an auxiliary data source for a more in-depth understanding it definitely makes sense. Worth the money.

gergelytakacs, 2020-03-04:
@mgulan @PeterChmurciak 

There is a small issue though:
[image]

I need to think about what's the best solution to get it.

Doesn't anyone else sell it?

mgulan, 2020-03-04:
@gergelytakacs Shipping [here](https://www.digitalsmarties.net/) seems to be mere 4.50€. There are to Rev's, one sold out, one may be available.

gergelytakacs, 2020-03-04:
@mgulan 

Excellent find, thank you. Once our excellent state-sponsored paychecks come in, I'll get them. Until them, I'm more passionate about not starving to death. :)

PeterChmurciak, 2020-03-05:
@gergelytakacs 
>what role wind speed plays

Virtually in all [models that we reviewed](https://github.com/gergelytakacs/AutomationShield/issues/166) the air speed was an important factor, but we did not have a means to get it. I believe that we could use this sensor to acquire PWM(or Power) vs AirSpeed relationship, and with that we could use PWM(or Power) (which we know at all times) in the models as system input instead of air speed (which we curently do not know). 

>Where (physically) would you build this into the tube?

I think we could make hole to the side of the tube, in a shape to allow insertion of the measuring part of mentioned sensor. There could also be multiple (closable or tapeable) holes in different heights to see how it affects the measured air speed. Alternatively the sensor could be simply mounted at the end of the tube and measure speed there. 

Of course I expect the ball to be outside of the tube during measurement. With the ball moving around, I do not think this sensor could be placed within the tube.