gergelytakacs/AutomationShield #185: FloatShield: Arduino Serial Plotter Screenshot - Redo

gergelytakacs, 2019-08-01:
@PeterChmurciak 

Unfortunately the screenshots from the Arduino API do not look very good on paper. It is my fault, I have requested large, full screen copies and that was not a good idea. If I create a figure, the lines are too thin and the numbers are invisible, like this:
[image]

To remedy this, I'd like to ask you for a re-do of the screenshots. This is an illustration figure, so its role is not to convey how good the PID control is, but rather that the user can see the results live in the plotter screen. The new idea is: create a purposefully small window of roughly quarter-screen or smaller dimensions; so that the numbers and the screen elements are not too small in relation to the plot. You can hit ALT+PrintScrn to get just the window. Here is an example from another paper what I am thinking of:
[image]

Can you please do this for when you'll have the time?

PeterChmurciak, 2019-08-02:
@gergelytakacs Sure thing.

I did not know about `ALT+PrintScrn` method. It saved me a lot of time otherwise spent cropping the images like a... Like somebody who does not know how to make a screenshot properly.

I have made multiple sets of screenshots - [Arduino PID Screenshots REDO 2.8.2019.zip](https://github.com/gergelytakacs/AutomationShield/files/3462987/Arduino.PID.Screenshots.REDO.2.8.2019.zip). 

The ball will surely stabilise after some time in all cases, but there is a certain element of randomness in how the ball will go from one reference level to another. That is the reason I have made more sets, so you can pick whichever ones look best to you.

Tell me if the format is still not allright or the screenshots should look differently.

gergelytakacs, 2019-08-05:
@PeterChmurciak 

Thank you very much, this is plenty to choose from! I've already picked one I liked. I know that the ball settles, you can't just really show that in this IDE screenshot. The idea behind showing this figure is just to say that it can be viewed/used in stand-alone mode in the IDE. The chart will have no other meaning.

If you liked the `CTRL+PrtSc` trick, have a look at the free software GreenShot. You can do crop while you do the screenshot and do cool annotations easily.

closed 2019-08-05 by gergelytakacs