gergelytakacs/AutomationShield #187: Simulink links broken for Heat and Opto

gergelytakacs, 2019-08-20:
@PeterChmurciak Since you've changed the name of the individual block libraries for "Opto" and "Heat" (because of the reserved "Float" issue), the examples for Opto and Heat ceased to work. Please take a look at these examples and fix the missing blocks, possibly by replacing them with the new ones. Example:

[image]
or
[image]

PeterChmurciak, 2019-08-20:
@gergelytakacs Hmm, thats weird because I have no such problem. Possibly you have to use `installForSimulink.m` and/or `slblocks.m` functions from `/simulink` folder in order for it to appear and work normally ? Can you please try that ? Because I do not have this issue so solving it is quite troublesome for me.

gergelytakacs, 2019-08-26:
@PeterChmurciak  I can confirm the problem exists. I have removed the path definitions from the computer, re-ran the add scripts with the same results. If you open an example from Opto or Heat, you see the missing links. By pure chance, I got a new laptop where I had to do a clean install of Windows/Matlab/Git etc. The problem still persists.

If you look at the properties of the missing block, you see that it refers to the library module by the old name, e.g. Opto and Heat. It is enough to re-write the definitions to the new ones, I think:
[image]

PeterChmurciak, 2019-08-27:
@gergelytakacs You were right about the bug, I have not opened the examples before - tried it only by using the blocks in new model... Sorry about that.

I have fixed the examples in c8a56ac. Tried to run/build them to make sure that they seem to work. Only issue was with `Sensor Read` block of `OptoShield`. There was a bug in the block mask:
[image]

The `Parameter3` should have been called `Ts` for the block to work, for some reason it was like this so I have fixed that.

All Simulink examples should work now, please check if the issue has been resolved.

gergelytakacs, 2019-08-28:
#c8a56ac Fixes this.

closed 2019-08-28 by gergelytakacs

gergelytakacs, 2019-08-28:
@PeterChmurciak Thank you for fixing this issue! I have no idea what the problem was with Opto, since I used it before, but oh well. :) I can confirm that your commit fixed the issue.