gergelytakacs/AutomationShield #192: Zero and Due support in MATLAB/Simulink

gergelytakacs, 2019-10-23:
Faster dynamic systems (MagnetoShield) and more complicated control and estimation schemes are unlikely to work in MATLAB/Simulink, at least on an Uno. So it is essential to learn
- Which one of Due/Zero is supported in MATLAB / Simulink and if it works for our purposes.
- If the communication speed is improved with Due and Zero in MATLAB.
- How does the usage of Due and Zero change in Simulink for FloatShield.

gergelytakacs, 2019-10-31:
@PeterChmurciak @KuboJakub 

I have a verbal update from Kubo about his lack of success in the Zero department. Can you confirm this?

Testing other platforms (e.g. Due / Zero) could be something you do @PeterChmurciak as well.

KuboJakub, 2019-10-31:
Arduino MKR Zero is compatible with Simulink support package since 2019b
version.
In older versions it looks like it is not supported.
https://www.mathworks.com/help/supportpkg/arduino/release-notes.html

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gergelytakacs, 2019-10-31:
@KuboJakub 

Okay, but that's the MKR Zero, not the Zero:
[image]
The question was, if the Zero is supported? (Or you can make it work with e.g. the MKR Zero support)

PS: If you answer through email, please delete the reply and any other text so that the thread remains readable.

PeterChmurciak, 2019-10-31:
@gergelytakacs Based on support package pages, Arduino Zero is suported neither by [matlab](https://www.mathworks.com/hardware-support/arduino-matlab.html) nor by [simulink](https://www.mathworks.com/hardware-support/arduino-simulink.html), so using it might prove to be impossible for now. 

Arduino Due is supported by both, with some limitations concerning DAC and CAN channels (not sure if those limitations will have any consequences in our case).

Arduino MKR Zero is supported only by matlab support package without shield support, not by simulink. (Based on release notes found by @KuboJakub, they added some Simulink support to it in latest 2019b version). 

I have currently matlab 2019a installed (as we agreed on this version for the simulink examples to be saved in), and have tried to connect with Zero to Matlab and Simulink, but with no success - it does not recognize Zero at all. I find Zero and MKR Zero to be too different boards for us to even hope that we could use partial support of MKR Zero for Zero. 

Concerning Matlab/Simulink examples I would for now focus on using  Due and possibly other, but supported boards.

gergelytakacs, 2019-11-01:
@PeterChmurciak @KuboJakub 

Both of you agree that Zero is unsupported in both MATLAB and Simulink. So for now, just ignore it and attempt to make your respective shields work with the Uno/Mega (not much difference) and the Due.

> Concerning Matlab/Simulink examples I would for now focus on using Due and possibly other, but supported boards.

Exacly. However, the other (possibly new) boards must be R3 pinout compatible, so that we can use the shields physically on them. So for now, Due is fine - our main motivation is the speed increase anyways.

KuboJakub, 2019-11-02:
@PeterChmurciak 
May I ask you, VL53L0X sensor communicates also via I2C no? And when you were creating a FloatShield lib for Simulink how did you solve I2C communication? Did you make it with the S-Function block?

KuboJakub, 2019-11-04:
@gergelytakacs @PeterChmurciak 
Now I am trying to create an I2C block for writing data on DAC via S-Function Block. I used an S-Function Builder to create a wrapper and header file as @PeterChmurciak used by PololuVL530X block for raw readings. But still, there occur some errors. I declared that the function will have 1 input  uint8 and 0 outputs. I also added Wire library for I2C communication but every time by building the program a procedure is terminated with the following error:
```
Error(s) encountered while building "Write_cez_Builder":
### Failed to generate all binary outputs.
```
Here are my program Simulink program and files created by builder + required libraries
[ERROR.txt](https://github.com/gergelytakacs/AutomationShield/files/3805239/ERROR.txt)
[WriteData2.zip](https://github.com/gergelytakacs/AutomationShield/files/3805246/WriteData2.zip)

If you have an idea why is it not working you will rly help me. Minimally could you write me which libraries you included in S-Function and maybe some quick explanation of how you did it? Because it is possible I have just used the S-Function Builder in a wrong way.
Thanks.

gergelytakacs, 2019-11-05:
@KuboJakub 
> May I ask you, VL53L0X sensor communicates also via I2C no?

Yes, the sensor in the FloatShield uses I2C communication. So you will also need it for the DAC module in Simulink.

> And when you were creating a FloatShield lib for Simulink how did you solve I2C communication? Did you make it with the S-Function block?

I'd leave this consultation to @PeterChmurciak please. You may try to take up communication with him using email etc.

PeterChmurciak, 2019-11-05:
@gergelytakacs @KuboJakub Indeed I did use the S-function block to realise the I2C sensor in simulink. There might have been a better way but I did not know about it.

For some to me unknown reason, I am getting error: `fatal error: Arduino.h: No such file or directory
 #include "Arduino.h"` when trying to build your function for Due board. Interestingly enough, I do not get this error when building your function for Uno... Honestly I hate this weird simulink compiler and its nuances. Based on your report you do not get error like this ?

The best way (at least the way I did it) of debugging the building process is to read the report from diagnostic viewer thoroughly and change one thing at a time to see if the error perished.

From what I have seen in your files, you really did use the builder bit incorrectly, but it can work anyway. Some tips to solve few of your possible errors:

* There is a little mess in the files - currently there are two S-functions created, one named `I2Cwrite` and other `I2Cwrite2`, as the first does not have its required files it will not work. Make sure in S-function block to use name `I2Cwrite2` and module `'I2Cwrite2_wrapper'` to use the second one.
* I have included `#include "twi.c"` (ideally before `#include "Wire.h"`) in wrapper to solve some future errors
* The function wrapper needs to be a C++ type file despite being created as C type by builder - you have to change its suffix from `I2Cwrite2_wrapper.c` to `I2Cwrite2_wrapper.cpp`
* In the function wrapper, all the `void` function definitions need to have `extern "C"` added before them - so it looks like `extern "C" void I2Cwrite2...`
* The `PCF8591` needs to be defined somewhere before it is used (for example after `#include "Wire.h"`), definition might look like this `#define PCF8591 (0x90 >> 1)` (based on google)

This is for now all that I have managed to figure out from reading errors while building the function for Uno. I am not able to build it for Due currently, as the compiler can not find `Arduino.h` header file (by building I mean pressing `Deploy to Hardware` button in Simulink, while having Uno/Due selected in `Model Configuration Parameters` as `Hardware board`).