gergelytakacs/AutomationShield #194: FloatShield: VL53L0X pololu library update

PeterChmurciak, 2019-10-30:
@gergelytakacs The issue thread for this topic was already closed and I did not want to spam other threads with it. 

The [pololu guys](https://github.com/pololu/vl53l0x-arduino) have recently updated their library and also have pulled in our [request](https://github.com/pololu/vl53l0x-arduino/pull/36#event-2754554614) for submodule functionality. That means we can now use the "official" library as a git submodule for our VL53L0X laser sensor. It would again require some git command kung-fu to swap the used sumbodules, but it would provide an "regularly" (or at least sometimes) updated library for the sensor. Just wanted to let you know that there is this option now.

gergelytakacs, 2019-10-31:
Thank you for the heads up! This is good news. Have you tested the updated library by any chance?

And by the way, congrats to improving the Polulu library:)

PeterChmurciak, 2019-10-31:
@gergelytakacs  Well, thanks :). There have been only [minor changes](https://github.com/PeterChmurciak/vl53l0x-arduino/compare/master...pololu:master) but to make sure I have tested the IDE examples and they work as they should - so we can update it if we want to. The Matlab and Simulink library counterparts use their own library files, and would need to be updated manually, which is not worth doing right now, considering there have been only minor changes in the library files.

gergelytakacs, 2019-10-31:
Thank you for the heads up and confirming that their update works.

I will leave this issue open with the hope that I will be in the mood to do said Git kung-fu, but right now I lack the mental capacity. If you wish, you may do it yourself, in other case let's leave this issue open.

gergelytakacs, 2019-12-02:
Closed and cancelled because of the need to account for SAM boards as per the suggestion of @PeterChmurciak in #190

closed 2019-12-02 by gergelytakacs