gergelytakacs/AutomationShield #197: MagnetoShield: The next one (R3->R4)

gergelytakacs, 2019-12-09:
This issue shall contain all suggested and required changes done on the next version of the MagnetoShield.

- The current Hall sensor is a legacy product. Neither this one nor the new version suggested by Allegro Microsystems is easily available through various shops. A different hall sensor should be picked for the new version.

KuboJakub, 2020-01-27:
I tested the meassurement functions of the R3 board and looks like current measurement works quite well but the voltage measurement is bit unprecise. Measures approximately one volt lower voltage. Also the top of the graph is a bit curved...cant be the C2 capacitor the problem?
[image]

KuboJakub, 2020-01-27:
So i searched for some suitable substitution of hall sensor A1302kua. Honestly, it looks 1302 was the best one but i found A1326 available on Farnell (has bit worse sensitivity otherwise looks the same):
https://sk.farnell.com/allegro-microsystems/a1326llhlx-t/hall-effect-sensor-linear-sip/dp/2840972
And this one from TME (better sensitivity almost as 1302). I personally prefer this one:
https://www.tme.eu/sk/details/ss39et/hallove-senzory/honeywell/

The pin configuration is the same so there is no need to change PCB roads. I am just a bit confused about SS39ET magnetic range. It is written it has a typical range +-1000, on maximum there is nothing written...Do i understand it correctly, it can measure also over the typical range?

KuboJakub, 2020-01-27:
About the tube for a permanent magnet. I have not found out anything else as the old system with the rubbery ring.
There was an idea from Erik to glue the tube on the magnet and somehow create a barrier in the middle of the tube, so the magnet can not stick on the electromagnet... But now I remembered the rubbery ring solution should provide easy access to the perm. magnet in the case the magnet would switch the orientation. If we glue the tube on the magnet, it will be uncomfortable to manipulate with that in the case the switch-thing happens (it can happen by tuning the regulator).

gergelytakacs, 2020-01-28:
Hello @KuboJakub 

Sorry that I have not been to attentive today, I have been a bit distracted. I've talked to Erik about your issue with the voltage, and to be honest - I am a bit puzzled as well as of why this is happening. I suggest you do try to read the voltage on the divider using the 'scope - as Erik said. So far I have no better ideas. I will try to think about it.  Maybe we should attempt to consult someone. We shouldn't blow this problem up too much, since up to now this function did not exist at all:)

Looking at the SS39ET: What is our maximal range in Gauss? The typical range, I guess is more about linearity. It may measure over these values but: a) can it output a ratiometric voltage for it b) will it be correct...? Otherwise, as far as I can see it looks fine: 1.4 mV/G gain (instead of 1.3 mV/G) and the same electrical characteristics. I say, if the physical layout is the same, we should try it. If not, then we have a couple spare old ones, or we can salvage them from old shields. I say we try it!

About the enclosure: I don't necessarily mind the rubber ring. However, couldn't you find anything better than the tube you've used? Maybe from glass? If not, okay...

KuboJakub, 2020-01-28:
I've provided the measurement on the pin A1 vs directly on the electromagnet. The result is the same, the values on the pin closer to the top are more rounded.
[image]

But first i would like to ask (probably i already did but i forgot why is it there), do we need diode D1? Because if i undenstand it correctly capacitor C2 is there for filtration of reverse current from the coil. But if we put there a diode is it not against the purpose of the capacitor (see pic below)?

Then i have one idea, to put one more capacitor between pin where we get measurements and the ground (see pic below). What do you think?
[image]

KuboJakub, 2020-01-28:
> Looking at the SS39ET: What is our maximal range in Gauss? The typical range, I guess is more about linearity. It may measure over these values but: a) can it output a ratiometric voltage for it b) [...]
I don't know what exactly our Gauss range is, but if we assume that 1302kua gives without magnet 2.5V -> 512 unites on 10bit ADC and with the magnet at the bottom around 21 unites -> 0.1V and the resolution is 1.3mV/G, so the range could be around 1850G.

I think, if we want to know the sensor is good, we have to just try it. But so far I prefer the idea to stay with 1302kua because we know it is good. But if we really have to change a sensor than I think this could be a good compensation (can't see big difference between this and 1302)

erik1392, 2020-01-29:
> But first i would like to ask (probably i already did but i forgot why is it there), do we need diode D1? Because if i undenstand it correctly capacitor C2 is there for filtration of reverse current [...]

No, the diode is there to deal with reverse current, CAP is only a filter. 

> Then i have one idea, to put one more capacitor between pin where we get measurements and the ground (see pic below). What do you think?

No, the CAP you suggested would make the problem even worse.

gergelytakacs, 2020-01-30:
@KuboJakub 

As @erik1392 said, the role of the diode is to protect from reverse current - the cap is to smooth out transients. So I think both are needed. Also, adding a CAP at the suggested place would make things just a lot worse. 

I have been thinking about the issue, and there must be a current loop or something we don't actually see. So I'm suggesting the following test:

**Take an op-amp and configure it with a gain <1 (to reduce 12 V > cca 3 V). The input will be the 12 V signal from the magnet, as you would measure with the Analog Discovery.  The output will be directly hooked up onto the ADC of the Arduino. The OPAMP will be hooked to the 0/12V rail, of course.** See if this solves the problem, I think it will. If this works as expected, we can put a single channel opamp there instead of a voltage divider.

> I think, if we want to know the sensor is good, we have to just try it. But so far I prefer the idea to stay with 1302kua because we know it is good. But if we really have to change a sensor than I think this could be a good compensation (can't see big difference between this and 1302)

We will test both. Have a couple spare 1302's and if not, we can salvage from the old boards. If we are good, then we can try the Honeywell alternative too

KuboJakub, 2020-01-30:
Maybe it could work. I will try it as soon as i will be able to come to lab.
But still i think the devider is all right and the mistake is with the diode. I mean if the diode is against reverse current from the coil, i would place it there like this (between R7 and GND - without the diode connection, not like on the board which i burned down)? Because now the diode is placed like it let pass the reverse current...maybe i am wrong, but this makes me more sense.
[image]

gergelytakacs, 2020-02-03:
@KuboJakub @erik1392 

Sorry for the slow response time I was away for a conference.

Dunno, man, that diode is pretty standard fare thing, and I think it is placed correctly. Your suggestion does not make sense to me, as I think it would not do anything in the circuit.

But I tell you what: try to remove the diode alltoghether and see if it helps! If it does, you were correct and we will think more about this issue. And I would also love if you could try to use the opamp in the design...

KuboJakub, 2020-02-03:
Okay, I will definitely try it.  I have removed the diode and unfortunately, it did not help, so i hope the opamp solution will be good.
I have finished the library for R3/R4. Because it has quite a lot of new things (especially writing on DAC is a lot different) and from this reason, I made it like a new library. Should i upload it here like a new separate library or reworke it on one universal library for R1/R2/R4? Because it will be a bit messy i think when i put it together.

KuboJakub, 2020-02-04:
I tried the mentioned idea and after few changes it worked rly well.
Because there is no possibility to get gain<1 on a non-inverting opamp I used just unite gain with high impedance and then a voltage divider with the same ratio as on the Shield. 
[image]
When i measured voltage with Discovery, it worked perfectly, but when i plugged output from the divider to Arduino, serial communication crashed. But i think the problem was in bad connection after soldering... Tomorrow/day after i will solve it.
The wide of graphs is different because of the difference in sampling by measuring.
[image]

gergelytakacs, 2020-02-05:
@KuboJakub @erik1392 

> I have removed the diode and unfortunately, it did not help

I suspected that is the case. So I guess, put it back:)

> separate library or rework it on one universal library for R1/R2/R4

This is awesome news, thank you. Let me put you this way: my preference is to have one library. Complex and messy is not the same thing. Will it be complex? Yes, sure. Will it be messy? That is really up to you, that's what a part of programming is about, to make it tidy and readable. But before you try to integrate your new solution to the legacy one, I'd say you must thoroughly test it.k Have you done that? 

> Because there is no possibility to get gain<1 on a non-inverting opamp 

Alright, but why not use an inverting opamp with gain <1 then correct the difference in software, e.g. `voltage = - voltage`? I think using a buffer then a voltage divider introduces unnecessary uncertainty in the measurement. If you'd use an inverting opamp with software correction, you can also ditch the Zener diode... In this you still have a divider which could possibly corrupt the measurement by introducing loops that we don't see right now (because of our inexperience). So I strongly suggest the inverting configuration.

> But i think the problem was in bad connection after soldering... Tomorrow/day after i will solve it.

Most probably . Okay, interested to say what's the deal with that.

I am not sure though I understand what's on your attached graph though :(

KuboJakub, 2020-02-06:
So i tested the solution with the unity gain and voltage divider. Worked well, the issue with serial communication has been caused by I2C communication. I tried to measure the voltage on analog pins A4 and A5 but as @gergelytakacs said pins A4 and A5 are reserved for I2C on Arduino UNO. When i used another analog pin on Arduino MEGA it worked as i expected. So i think the problem with voltage-on-board measurement is solved. Now measurements  from the board provide very close values to the real voltage on the electromagnet -> board measures 10.15V and i have measured with the oscilloscope 10.30V.

Schematic and PCB are reworked already too. Please check them on the Synology drive.
I have also changed the library, so it is complex but not messy now :D minimally i hope so.

gergelytakacs, 2020-02-10:
@KuboJakub 

Thank you for the great news. The small difference, (10.15 V vs. 10.30 V is negligible), if otherwise the measurement is not skewed, etc. Good job, thank you.

I have asked @erik1392 last week to go a round of schematic checking and PCB checking. As soon as he is ready, I'll do a round and will order the updated PCB.

> I have also changed the library, so it is complex but not messy now ;D minimally i hope so.

I will take a look at that soon and will let you know where and how to begin. In the meantime there are lots of things you can work on, namely

- A fully functional Simulink API
- Compatibility (testing) with more powerful boards, e.g. Due/Zero/Metro M4
- Look into modelling issues.
- Any sort of well-worked control or estimation example on the hardware. It's educational afterall, whatever you contribute is useful.

Have a visit with us this week, if possible:)

gergelytakacs, 2020-02-11:
@erik1392 @KuboJakub 

Board renders:
[image]
[image]

gergelytakacs, 2020-02-11:
@erik1392 @KuboJakub 

PCB layout files have been accepted by manufacturer and in production already.

gergelytakacs, 2020-02-28:
@KuboJakub @erik1392 

I'll either bring it next Wednesday (unless zombie apocalypse) or you come visit me in DS:
[image]

KuboJakub, 2020-03-03:
Maybe bit out of topic but I found probably a mistake in the AutomationShield library
https://github.com/gergelytakacs/AutomationShield/blob/7fa9552ebacb5283ad1b7c0366aa7672c0fc4a73/src/AutomationShield.h#L35
Is it not incorrect in the case of use Arduino Due with 12-bit ADC?

gergelytakacs, 2020-03-04:
@gergelytakacs 

You are absolutely correct.

I'm not sure this has even been used somewhere, but it should be `#ifdef` based on architecture. Maybe ou could correct it?:)

KuboJakub, 2020-03-16:
Are yours R3 working? I have some hardware issue. When i am  trying to communicate with 12-bit DAC, somehow the communication stops. I am sure the code is correct coz we used that one already and worked. Do you have the same issue or does the shield work good?

KuboJakub, 2020-03-17:
Okay, I found the mistake but unfortunately @erik1392 it was caused by that hole you projected into the new version. Diode D4 should not be on the pin A0 and ground at the same time. Originally in my rework, D4 is between A1 and the GND but as the "big hole" was added it moved the pattern too close to the road of A0 pin and happened the patterns overlap...
[image]
[image]
[image]
I just cut the road on pin A0 so thankfully it works now but reference read from a potentiometer is cut off.
If I can suggest, maybe if some next movement with roads and patterns is made, it would be helpful if we make some detailed documentation - I mean a picture of the original position and picture of the new position and why the change was made. Because this was rly stupid overlap again that anyone can overlook.

KuboJakub, 2020-06-14:
I have found one more HW mistake we made and I would be glad to change it. We have in our circuit capacitor C2. Its purpose used to be to make voltage and current "softer" (if am i right). But we use DC whitch already has some features for making DC voltage clearer in adapter and on the board probably too. 
Actually, as i found out by identification of the model, the capacitor is creating small current kicks, which work like a disturbances (and quite big). The solution would be to add, in electric equation, a part with capacitor current-voltage dependency, or just remove the capacitor and leav the identified model as it is.
I personally would remove the capacitor because I think it is pointless in our circuit and makes just problems...probably the problem with the voltage measurement mentioned at the beginning of this issue was caused by the capacitor too. As we used Opamp the current flow could not go through resistor in voltage devider. So the current flow changed direction through the coil and caused very chaotic current behavior.

First is comparison of measured data on circuit with capacitor and identified model.
Second is the circuit without capacitor and identified model.
[image]
[image]

gergelytakacs, 2020-06-15:
@KuboJakub (FYI @erik1392 ) 

Great job!

Well, this is very interesting. Basically, I completely agree with your conclusions - if it does nothing (or makes things worse) than it should be removed. You've proved this by experiment, so that's that.

However, I am a flabbergasted of this behavior. I had the C2 capacitor in place for the R2 tests and saw no such behavior. Wasn't the C2 by any chance soldered with the wrong polarity? Didn't we/you confuse the type of the capacitor? Or maybe the one in the R2 was incorrectly soldered so we did not see these weird measurement - effects?

Conclusion: 

- Kubo: If a component is not needed (or just makes things work) feel free to remove it. In this case, please make sure to edit the circuit schematics and the PCB accordingly.
- Erik: This question bugs me a bit, so maybe, just so we can learn from our mistakes - how about soldering the C2 in place on your version and testing this issue? My personal bet is that its polarity was swapped.