gergelytakacs/AutomationShield #198: LinkShield: The next one (R1->R2)

gergelytakacs, 2020-01-10:
This issue lists all the potential ideas and changes that should be done when planning a new release:
- Mark the pins of the servo for easier soldering and assembly on the silkscreen.

martinvrican, 2020-01-23:
The results of measurement with fixed arduino to the table
[image]

All reasults and files are in following folder on drive:    ...\Research\11 Link\Measurements

gergelytakacs, 2020-01-24:
Awesome, thank you @martinvrican

gergelytakacs, 2020-12-11:
@martinvrican  @erik1392  @mgulan 

The linkshield can be tested on the B&K Electrodynamic shaker we have in the lab. Also, it would be interesting to see how the LDV can be used to calibrate the gyro/accelerometer! So for this one, we actually have great tools in the lab.