gergelytakacs/AutomationShield #206: FloatShield: The next one (R2->R3)

PeterChmurciak, 2020-02-24:
@gergelytakacs @erik1392 Personally I am very dissatisfied with how the things have ended up on the FloatShield R2. Despite our best efforts, the R2 version, specifically the [fan](https://github.com/gergelytakacs/AutomationShield/issues/170#issuecomment-576154474) that the board was built to accomodate, is not usable for our purposes. It saddens me that "my" version is not better - not really an proper upgrade, to the R1.

Therefore I propose to make another version (not really sure what to call it, whether R2 - version 2, or R3) that would bring all the compatibility changes the R2 should have brought, without using the new fan - using the original one - either canibalised or [bought from store](https://www.gme.sk/ventilator-sunon-pmd1204pqb1). Alternatively some other fan of this type (two-cable type) could be used, but based on our previous experience, I am not confident in experimenting with unknown fans. 

I am well aware that @gergelytakacs is currently preoccupied with more important things than my "hurt ego". Because of this, and possibly to avoid the school bureaucracy, I will order the PCB and required parts by myself.

All I would like to ask for, is your (gently applied) constructive criticism 🙂

PeterChmurciak, 2020-02-24:
@gergelytakacs @erik1392 Could you please look at the scheme and PCB layout to see if there is any mistake or whether you would like to see something done differently ?
(By no means do I want to hurry it. If there should be another version, it has to be a working one)

Original R1 scheme (for reference):
[image]

New R3 scheme:
[image]

And PCB layout for R3:

Top: (gray line represents fan dimensions)
[image]

Bottom:
[image]

gergelytakacs, 2020-03-04:
@PeterChmurciak 

First of all, good job, thank you. I'm reviewing this right now - based on this post and not the files.

Schematics:
- Looks good, I went through it quite carefully.
- I can't confirm the pin orientation of the MOSFET from the top of my head. Sleep deprived and on the train, it looks ok. Please double check it.
- The schematic symbol of the MOSFET is not very well readable. The reason is the small arrowhead, since that is the main point of orientation for Drain/Source.

PCB:
- Missing logos, etc.
- Looks good otherwise.
- I have not double checked the correspondence with the individual traces w/ the schematics, but if it was auto-generated it should be all-right. Double check it before we send it to manufacturing.
- It is a bit weird how this "fat" trace continues as a thin one:
[image]

@erik1392 please express your opinion as well.

gergelytakacs, 2020-03-04:
@PeterChmurciak 

Replying to your thread opener:
- Other than you, three more people (@gergelytakacs @erik1392 @mgulan ) were looking at the new version, neither of us expected these problems. We learned a lot. This is what development/science looks like. Don't beat yourself up over it.
- I can order the new PCB as soon as it is ready, don't worry about that.
- I would avoid any great changes now, since time is pressing. I may have ideas, but let's not use time for hardware revisions!

PeterChmurciak, 2020-03-05:
@gergelytakacs 
>pin orientation of the MOSFET

Orientation of mosfet is (from top pin to bottom pin on picture) `Gate` connected through resistor to pwm pin 5, `Drain` connected to the fan grounding pin and `Source` connected to the ground. I have checked and I believe it is correct.

>schematic symbol of the MOSFET

Is this symbol better ?

[image]

And PCB:
Top:
[image]

Bottom:
[image]

I have downloaded and used [Helvetica Condensed](https://www.maisfontes.com/helvetica-condensed) font on the AutomationShield logo this time, hopefully it is the correct one.

I will check everything once more.

Please take a look again and tell me if everything is all right, or you would like to see something done differently.