gergelytakacs/AutomationShield #211: AutomationShield compatibitliy with DUE - ADC resolution

KuboJakub, 2020-03-30:
Arduino DUE has 12-bit ADC so I have changed the analog reference code. But i did not know that default resolution for DUE is 10-bit, you have to change it in the code with "analogReadResolution(bits)".
My question is, are we gonna use this advantage of DUE and every library will include this code for the case of use DUE or ZERO, or do we use default settings for 10-bit resolution of the ADC for all boards?
https://github.com/gergelytakacs/AutomationShield/blob/78befb76112fa5325af6ee27cccbf2954d269beb/src/AutomationShield.h#L37
To see what I added to make 12-bit resolution in Magneto library chack the link:
https://github.com/gergelytakacs/AutomationShield/blob/7f707ad3bb54a5d9c336a07e08bf23d35887f5a9/src/MagnetoShield.cpp#L28

KuboJakub, 2020-03-30:
There is an issue that if we use 12-bit resolution for DUE, all functions for converting data from analog pins have to be defined for both boards... It means if I read analog position from a sensor and wanna convert it to percents: 
data*(100/1023) for UNO
data*(100/4095) for DUE
So library have to have a rework (maybe in .h file) where all 1023 which are used e.g. in mapFloat() function, will be replaced by 4095.... I hope, you know what i mean.

gergelytakacs, 2020-04-09:
Okay, I understand. However you need to be more specific what do you need to rework there for the rest of the team.