gergelytakacs/AutomationShield #250: MagnetoShield R4: marginal issues

KuboJakub, 2021-03-20:
- Reconsider size of levitating magnets. We have magnets with 9 and 8 mm diameter. 9mm magnet in 10mm tube could be stabilized by physical contact with sides of tube, what is not desired.
- Use of the different magnet could require  new identification, based on the weight and material of the magnet.
- We have two models of the MagnetoShield. Which one is better? Having 2 models and 2 magnets can provide diversity in working with the system. Maybe going on with both of them?

gergelytakacs, 2021-03-22:
If you have any time left and will to do it, you can do objective studies to attempt to answer these questions. Until then, this issue is to remain open.

gergelytakacs, 2022-05-19:
Closing this, anyone up for Magneto R5 shall revisit;)

closed 2022-05-19 by gergelytakacs