gergelytakacs/AutomationShield #266: Float library problems

erik1392, 2021-05-05:
Arduino DUE
- [ ]  ToF sensor ADXL345 library incomparable with Wire1 I2C port on Arduino due.

Adafrute Metro M4 Express
- [ ]  Arduino core does not have a flag for architecture SAMD51, ADC resolution is not selected correctly.

gergelytakacs, 2021-05-05:
- Due, allrighty, not really a huge problem we should work on it. The ADXL345 is not even a library, it has been designed by @martinvrican ? Can you take a look at tit?
- Metro, I'm not sure I understand you correctly. The sampling core does not have a flag or the Arduino Core?

erik1392, 2021-05-05:
issue updated.. no problem with adafruit boards specifically but with all 12bit ADC boards. 
ADXL345 used in Arduino IDE is the original Adafruit library.  Only  Matlab/Simulink has some semi-custom library.

erik1392, 2021-05-05:
OK, my mistake.. I was confused that ADXL345 is ToF sensor but no... it is an accelerometer.  So this issue is irrelevant for FloatShield.. but it looks like the Adafruit ADXL345 library in src/lib folder is not happy when you are compiling anything for Due. It looks like it is not even used in LinkShield because @martinvrican has already custom one. So I suggest removing that Lib from Automation Shield if it is not utilized.

gergelytakacs, 2021-05-05:
That's what I thought. That @martinvrican has already implemented it. In that case, @martinvrican can we remove the submodule for the original ADXL345?