gergelytakacs/AutomationShield #338: MagnetoShield 12V Rail Necessary?

topherbuckley, 2023-01-20:
We are reviewing this design for use in a course, and we noticed a discrepancy between the [Solenoid](https://www.adafruit.com/product/3872#technical-details) mentioned in the BOM claiming to be a "5V Electromagnet" yet you have designed it to use the 12V rail. Why is this? Why not use the regulated 5V rail throughout the design? Maybe this was to avoid any voltage drop issues expected on the 5V rail that might be caused by the draw from the solenoid? We were thinking to redesign using a single 5V rail but wanted to check if you had already considered this and avoided it for some reason.

gergelytakacs, 2023-09-14:
Hello @topherbuckley Apologize for the late reply, for some reason the notificaitons were not really arriving.

I believe the magnet is actually 12V! That's why the 12V rail:) @erik1392 care to comment?

gergelytakacs, 2023-09-14:
The BOM on the Wiki page does not list voltage, points to an eBay seller, where magnet specs are actually 24 V.
[image]
The BOM in the MagnetoShield conference paper lists the voltage as 12/24V.
[image]
We are not actually sure where you have found the information about the magnet using 5 V. In any case, I believe the current draw will be quite substantial anyways so USB power is likely to be not enough.

closed 2023-09-14 by gergelytakacs